coolsnowwolf/lede #12238: 云编译linux fail to build

xiaouex, 2024-06-15:
### 详细叙述

今天同步代码编译新固件出现的问题，看了下error的地方好像是跟led驱动有关的？不太懂，求指教

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

rax3000m

### 详细日志

make[2]: Leaving directory '/workdir/openwrt/scripts/config'
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt'
make[3]: Entering directory '/workdir/openwrt/target/linux'
make[4]: Entering directory '/workdir/openwrt/target/linux/mediatek'
rm -f /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/localversion
/workdir/openwrt/scripts/kconfig.pl  + /workdir/openwrt/target/linux/generic/config-6.1 /workdir/openwrt/target/linux/mediatek/filogic/config-6.1 > /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /workdir/openwrt/.config >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.target
/workdir/openwrt/scripts/package-metadata.pl kconfig /workdir/openwrt/tmp/.packageinfo /workdir/openwrt/.config 6.1 > /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.override
/workdir/openwrt/scripts/kconfig.pl 'm+' '+' /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.target /dev/null /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.override > /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set
mv /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.old
grep -v INITRAMFS /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.old > /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set
echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set
rm -rf /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/modules
cmp -s /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.prev || { cp /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config; cp /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.prev; }
export MAKEFLAGS= ; [ -d /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/user_headers ] || make -C /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jun 15 15:22:45 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.92 CC="aarch64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/user_headers headers_install
grep '=[ym]' /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.config.set | LC_ALL=C sort | /workdir/openwrt/staging_dir/host/bin/mkhash md5 > /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.vermagic
touch /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/.configured
rm -f /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/vmlinux /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92/System.map
make -C /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jun 15 15:22:45 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.92 CC="aarch64-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92'
#
# No change to .config
#
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92'
make -C /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jun 15 15:22:45 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.92 CC="aarch64-openwrt-linux-musl-gcc" Image dtbs modules
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.1.92'
  CALL    scripts/checksyscalls.sh
  CC      drivers/net/phy/mediatek-ge-soc.o
drivers/net/phy/mediatek-ge-soc.c:1187:57: warning: 'enum led_brightness' declared inside parameter list will not be visible outside of this definition or declaration
 1187 |                                          u8 index, enum led_brightness value)
      |                                                         ^~~~~~~~~~~~~~
drivers/net/phy/mediatek-ge-soc.c:1187:72: error: parameter 3 ('value') has incomplete type
 1187 |                                          u8 index, enum led_brightness value)
      |                                                    ~~~~~~~~~~~~~~~~~~~~^~~~~
drivers/net/phy/mediatek-ge-soc.c:1186:12: error: function declaration isn't a prototype [-Werror=strict-prototypes]
 1186 | static int mt798x_phy_led_brightness_set(struct phy_device *phydev,
[... cut ...]
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#2.08#0.71#2.74
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

xiaouex, 2024-06-16:
推测与13号的commit有关
[image]

aiamadeus, 2024-06-16:
现在呢，这仓库年久失修真 TM 难同步

xiaouex, 2024-06-16:
> 现在呢，这仓库年久失修真 TM 难同步

解决了，编译成功

closed 2024-06-16 by aiamadeus