coolsnowwolf/lede #12242: 编译报错

AXBNG, 2024-06-16:
### 详细叙述

最新源码编译报错，两天前正常，求修复

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

AX6

### 详细日志

2024-06-16T10:10:24.1999621Z make[6]: *** [scripts/Makefile.build:118: arch/arm64/kernel/asm-offsets.s] Error 1
2024-06-16T10:10:24.2001003Z make[5]: *** [Makefile:1285: prepare0] Error 2
2024-06-16T10:10:24.2001796Z make[5]: *** Waiting for unfinished jobs....
2024-06-16T10:10:24.2013959Z   DTC     arch/arm64/boot/dts/qcom/sdm850-lenovo-yoga-c630.dtb
2024-06-16T10:10:24.2055145Z   DTC     arch/arm64/boot/dts/qcom/sdm850-samsung-w737.dtb
2024-06-16T10:10:24.2281767Z   DTC     arch/arm64/boot/dts/qcom/sm6125-sony-xperia-seine-pdx201.dtb
2024-06-16T10:10:24.2404431Z   DTC     arch/arm64/boot/dts/qcom/sm6350-sony-xperia-lena-pdx213.dtb
2024-06-16T10:10:24.2463936Z   DTC     arch/arm64/boot/dts/qcom/sm7225-fairphone-fp4.dtb
2024-06-16T10:10:24.2560608Z   DTC     arch/arm64/boot/dts/qcom/sm8150-hdk.dtb
2024-06-16T10:10:24.2647681Z   DTC     arch/arm64/boot/dts/qcom/sm8150-microsoft-surface-duo.dtb
2024-06-16T10:10:24.2654952Z   DTC     arch/arm64/boot/dts/qcom/sm8150-mtp.dtb
2024-06-16T10:10:24.2676947Z   DTC     arch/arm64/boot/dts/qcom/sm8150-sony-xperia-kumano-bahamut.dtb
2024-06-16T10:10:24.3038882Z   DTC     arch/arm64/boot/dts/qcom/sm8150-sony-xperia-kumano-griffin.dtb
2024-06-16T10:10:24.3137447Z   DTC     arch/arm64/boot/dts/qcom/sm8250-hdk.dtb
2024-06-16T10:10:24.3153589Z   DTC     arch/arm64/boot/dts/qcom/sm8250-mtp.dtb
2024-06-16T10:10:24.3167446Z   DTC     arch/arm64/boot/dts/qcom/sm8250-sony-xperia-edo-pdx203.dtb
2024-06-16T10:10:24.3517613Z   DTC     arch/arm64/boot/dts/qcom/sm8250-sony-xperia-edo-pdx206.dtb
2024-06-16T10:10:24.3722301Z   DTC     arch/arm64/boot/dts/qcom/sm8350-hdk.dtb
2024-06-16T10:10:24.3799386Z   DTC     arch/arm64/boot/dts/qcom/sm8350-microsoft-surface-duo2.dtb
2024-06-16T10:10:24.3809246Z   DTC     arch/arm64/boot/dts/qcom/sm8350-mtp.dtb
2024-06-16T10:10:24.4033083Z   DTC     arch/arm64/boot/dts/qcom/sm8350-sony-xperia-sagami-pdx214.dtb
2024-06-16T10:10:24.4147604Z   DTC     arch/arm64/boot/dts/qcom/sm8350-sony-xperia-sagami-pdx215.dtb
2024-06-16T10:10:24.4157367Z   DTC     arch/arm64/boot/dts/qcom/sm8450-hdk.dtb
2024-06-16T10:10:24.4161689Z   DTC     arch/arm64/boot/dts/qcom/sm8450-qrd.dtb
2024-06-16T10:10:24.4345729Z   DTC     arch/arm64/boot/dts/qcom/sm8450-sony-xperia-nagara-pdx223.dtb
2024-06-16T10:10:24.4669509Z make[5]: Leaving directory '/home/runner/work/AX6/AX6/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.92'
2024-06-16T10:10:24.4672066Z make[4]: *** [Makefile:25: /home/runner/work/AX6/AX6/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.92/.modules] Error 2
2024-06-16T10:10:24.4673472Z make[4]: Leaving directory '/home/runner/work/AX6/AX6/openwrt/target/linux/qualcommax'
2024-06-16T10:10:24.4674353Z make[3]: *** [Makefile:11: compile] Error 2
2024-06-16T10:10:24.4674925Z make[3]: Leaving directory '/home/runner/work/AX6/AX6/openwrt/target/linux'
2024-06-16T10:10:24.4675425Z time: target/linux/compile#31.68#6.81#24.33
2024-06-16T10:10:24.4679007Z     ERROR: target/linux failed to build.
2024-06-16T10:10:24.4681282Z make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
2024-06-16T10:10:24.4686665Z make[2]: Leaving directory '/home/runner/work/AX6/AX6/openwrt'
2024-06-16T10:10:24.4708790Z make[1]: *** [target/Makefile:23: /home/runner/work/AX6/AX6/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
2024-06-16T10:10:24.4712904Z make[1]: Leaving directory '/home/runner/work/AX6/AX6/openwrt'
2024-06-16T10:10:24.4719621Z make: *** [/home/runner/work/AX6/AX6/openwrt/include/toplevel.mk:231: world] Error 2
2024-06-16T10:10:24.4736589Z ##[error]Process completed with exit code 2.

WYC-2020, 2024-06-17:
单线程日志

AXBNG, 2024-06-17:
> 单线程日志

刚单线程跑了一下，大佬有空看下
2024-06-17T04:07:41.8779546Z In file included from ./include/linux/workqueue.h:9,
2024-06-17T04:07:41.8780704Z                  from ./include/linux/mm_types.h:19,
2024-06-17T04:07:41.8781648Z                  from ./include/linux/mmzone.h:22,
2024-06-17T04:07:41.8782548Z                  from ./include/linux/gfp.h:7,
2024-06-17T04:07:41.8783181Z                  from ./include/linux/xarray.h:15,
2024-06-17T04:07:41.8784021Z                  from ./include/linux/radix-tree.h:21,
2024-06-17T04:07:41.8784590Z                  from ./include/linux/idr.h:15,
2024-06-17T04:07:41.8785413Z                  from ./include/linux/kernfs.h:12,
2024-06-17T04:07:41.8785972Z                  from ./include/linux/sysfs.h:16,
2024-06-17T04:07:41.8786526Z                  from ./include/linux/kobject.h:20,
2024-06-17T04:07:41.8787174Z                  from ./include/linux/of.h:17,
2024-06-17T04:07:41.8787735Z                  from ./include/linux/irqdomain.h:35,
2024-06-17T04:07:41.8788310Z                  from ./include/linux/acpi.h:13,
2024-06-17T04:07:41.8788834Z                  from ./include/acpi/apei.h:9,
2024-06-17T04:07:41.8789277Z                  from ./include/acpi/ghes.h:5,
2024-06-17T04:07:41.8789729Z                  from ./include/linux/arm_sdei.h:8,
2024-06-17T04:07:41.8790289Z                  from arch/arm64/kernel/asm-offsets.c:10:
2024-06-17T04:07:41.8790962Z ./include/linux/timer.h:23:33: error: duplicate member 'cust_data'
2024-06-17T04:07:41.8791539Z    23 |         unsigned long           cust_data;
2024-06-17T04:07:41.8791983Z       |                                 ^~~~~~~~~
2024-06-17T04:07:42.1413016Z make[6]: *** [scripts/Makefile.build:118: arch/arm64/kernel/asm-offsets.s] Error 1
2024-06-17T04:07:42.1413988Z make[5]: *** [Makefile:1285: prepare0] Error 2
2024-06-17T04:07:42.1415878Z make[5]: Leaving directory '/home/runner/work/AX6/AX6/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.92'
2024-06-17T04:07:42.1418699Z make[4]: *** [Makefile:25: /home/runner/work/AX6/AX6/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.92/.modules] Error 2
2024-06-17T04:07:42.1420613Z make[4]: Leaving directory '/home/runner/work/AX6/AX6/openwrt/target/linux/qualcommax'
2024-06-17T04:07:42.1421554Z make[3]: *** [Makefile:11: compile] Error 2
2024-06-17T04:07:42.1422205Z make[3]: Leaving directory '/home/runner/work/AX6/AX6/openwrt/target/linux'
2024-06-17T04:07:42.1422698Z time: target/linux/compile#22.68#6.24#27.73
2024-06-17T04:07:42.1425598Z     ERROR: target/linux failed to build.
2024-06-17T04:07:42.1428340Z make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
2024-06-17T04:07:42.1434556Z make[2]: Leaving directory '/home/runner/work/AX6/AX6/openwrt'
2024-06-17T04:07:42.1453676Z make[1]: *** [target/Makefile:23: /home/runner/work/AX6/AX6/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
2024-06-17T04:07:42.1458610Z make[1]: Leaving directory '/home/runner/work/AX6/AX6/openwrt'
2024-06-17T04:07:42.1469760Z make: *** [/home/runner/work/AX6/AX6/openwrt/include/toplevel.mk:231: world] Error 2
2024-06-17T04:07:42.1486783Z ##[error]Process completed with exit code 2.

closed 2024-06-17 by coolsnowwolf