coolsnowwolf/lede #12249: qmi编译通不过

ChaingTsung, 2024-06-17:
### 详细叙述

wwan包编译通不过

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64

### 详细日志

...
make[3]: Leaving directory '/home/yoki/openwrt/package/wwan/app/quectel_cm_5G'
time: package/wwan/app/quectel_cm_5G/compile#1.39#0.35#1.80
make[3]: Entering directory '/home/yoki/openwrt/package/wwan/driver/quectel_MHI'
mkdir -p /home/yoki/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/yoki/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/pcie_mhi-1.3.6/.pkgdir/kmod-pcie_mhi/. /home/yoki/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/yoki/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-pcie_mhi_installed
echo "kmod-pcie_mhi" >> /home/yoki/openwrt/staging_dir/target-x86_64_musl/pkginfo/quectel_MHI.default.install
make[3]: Leaving directory '/home/yoki/openwrt/package/wwan/driver/quectel_MHI'
time: package/wwan/driver/quectel_MHI/compile#1.41#0.32#1.66
make[3]: Entering directory '/home/yoki/openwrt/package/wwan/app/luci-app-pcimodem'
mkdir -p /home/yoki/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/yoki/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/openwrt/build_dir/target-x86_64_musl/luci-app-pcimodem/.pkgdir/luci-app-pcimodem/. /home/yoki/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/yoki/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-app-pcimodem_installed
echo "luci-app-pcimodem" >> /home/yoki/openwrt/staging_dir/target-x86_64_musl/pkginfo/luci-app-pcimodem.default.install
make[3]: Leaving directory '/home/yoki/openwrt/package/wwan/app/luci-app-pcimodem'
time: package/wwan/app/luci-app-pcimodem/compile#19.49#1.94#21.10
make[3]: Entering directory '/home/yoki/openwrt/package/wwan/driver/quectel_QMI_WWAN'
rm -f /home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/.built
touch /home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/.built_check
make -C "/home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.32" ARCH="x86" CROSS_COMPILE="x86_64-openwrt-linux-musl-" CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0=qmi_wwan_q-3.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" M="/home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0"  modules
make[4]: Entering directory '/home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.32'
  CC [M]  /home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.o
/home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.c: In function '_rmnet_vnd_get_stats64':
/home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.c:831:33: error: implicit declaration of function 'u64_stats_fetch_begin_irq'; did you mean 'u64_stats_fetch_begin'? [-Werror=implicit-function-declaration]
  831 |                         start = u64_stats_fetch_begin_irq(&stats64->syncp);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 u64_stats_fetch_begin
/home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.c:843:26: error: implicit declaration of function 'u64_stats_fetch_retry_irq'; did you mean 'u64_stats_fetch_retry'? [-Werror=implicit-function-declaration]
  843 |                 } while (u64_stats_fetch_retry_irq(&stats64->syncp, start));
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                          u64_stats_fetch_retry
cc1: all warnings being treated as errors
make[6]: *** [scripts/Makefile.build:243: /home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/qmi_wwan_q.o] Error 1
make[5]: *** [/home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.32/Makefile:1913: /home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0] Error 2
make[4]: *** [Makefile:234: __sub-make] Error 2
make[4]: Leaving directory '/home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.32'
make[3]: *** [Makefile:50: /home/yoki/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_q-3.0/.built] Error 2
make[3]: Leaving directory '/home/yoki/openwrt/package/wwan/driver/quectel_QMI_WWAN'
time: package/wwan/driver/quectel_QMI_WWAN/compile#3.29#1.12#4.53
    ERROR: package/wwan/driver/quectel_QMI_WWAN failed to build.
make[2]: *** [package/Makefile:173: package/wwan/driver/quectel_QMI_WWAN/compile] Error 1
make[2]: Leaving directory '/home/yoki/openwrt'
make[1]: *** [package/Makefile:167: /home/yoki/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/openwrt'
make: *** [/home/yoki/openwrt/include/toplevel.mk:248: world] Error 2

coolsnowwolf, 2024-06-21:
fixed

smallprogram, 2024-06-23:
这个问题又出现了

hzjnet, 2024-06-26:
+1 确实

ChaingTsung, 2024-07-01:
还是同样的问题，大佬
```
2024-06-28T09:53:57.6592110Z make[3]: Entering directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/utils/f2fs-tools'
2024-06-28T09:53:57.8010787Z echo "libf2fs" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/f2fs-tools.default.install
2024-06-28T09:53:57.8027274Z echo "mkf2fs" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/f2fs-tools.default.install
2024-06-28T09:53:57.8042346Z make[3]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/utils/f2fs-tools'
2024-06-28T09:53:57.8048359Z time: package/utils/f2fs-tools/default/compile#0.12#0.04#0.15
2024-06-28T09:53:57.8161799Z make[3]: Entering directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/utils/usbmode'
2024-06-28T09:53:57.9741325Z mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2024-06-28T09:53:57.9761641Z SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/usbmode-2022-02-24-3c8595a4/.pkgdir/usb-modeswitch/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2024-06-28T09:53:57.9804371Z touch /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.usb-modeswitch_installed
2024-06-28T09:53:57.9861461Z echo "usb-modeswitch" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/usbmode.default.install
2024-06-28T09:53:57.9875531Z make[3]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/utils/usbmode'
2024-06-28T09:53:57.9880264Z time: package/utils/usbmode/compile#0.14#0.05#0.18
2024-06-28T09:53:57.9983221Z make[3]: Entering directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/wwan/app/luci-app-pcimodem'
2024-06-28T09:53:58.2790525Z mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2024-06-28T09:53:58.2815416Z SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-pcimodem/.pkgdir/luci-app-pcimodem/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2024-06-28T09:53:58.2858739Z touch /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-app-pcimodem_installed
2024-06-28T09:53:58.2926144Z echo "luci-app-pcimodem" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/luci-app-pcimodem.default.install
2024-06-28T09:53:58.2943345Z mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2024-06-28T09:53:58.2966702Z SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-pcimodem/.pkgdir/luci-i18n-pcimodem-zh-cn/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2024-06-28T09:53:58.3000310Z touch /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-i18n-pcimodem-zh-cn_installed
2024-06-28T09:53:58.3018785Z echo "luci-i18n-pcimodem-zh-cn" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/luci-app-pcimodem.default.install
2024-06-28T09:53:58.3042124Z make[3]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/wwan/app/luci-app-pcimodem'
2024-06-28T09:53:58.3051650Z time: package/wwan/app/luci-app-pcimodem/compile#0.24#0.09#0.31
2024-06-28T09:53:58.3156013Z make[3]: Entering directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/wwan/driver/fibocom_QMI_WWAN'
2024-06-28T09:53:58.4538677Z mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2024-06-28T09:53:58.4572644Z SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/.pkgdir/kmod-qmi_wwan_f/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
[... cut ...]
2024-06-28T09:53:59.4671781Z make[1]: *** [package/Makefile:110: /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-06-28T09:53:59.4693893Z make[1]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt'
2024-06-28T09:53:59.4724561Z make: *** [/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/include/toplevel.mk:231: world] Error 2
```

closed 2024-07-09 by coolsnowwolf