coolsnowwolf/lede #12261: 编译失败

yk271, 2024-06-22:
### 详细叙述

x86_64 最新源码 6.6 内核编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2022-02-13-108a36cb'
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2022-02-13-108a36cb=fullconenat-2022-02-13-108a36cb -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include  -fPIC -D_INIT=libipt_FULLCONENAT_init -c -o libipt_FULLCONENAT.o libipt_FULLCONENAT.c;
libipt_FULLCONENAT.c:6:10: fatal error: xtables.h: No such file or directory
    6 | #include <xtables.h>
      |          ^~~~~~~~~~~
compilation terminated.
make[4]: *** [Makefile:4: libipt_FULLCONENAT.o] Error 1
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2022-02-13-108a36cb'
make[3]: *** [Makefile:63: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2022-02-13-108a36cb/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/network/services/fullconenat'
time: package/network/services/fullconenat/compile#0.31#0.10#0.40
    ERROR: package/network/services/fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/network/services/fullconenat/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

GreatMichaelLee, 2024-06-22:
+1 我的是说缺少xtables的动态库

find /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2022-02-13-108a36cb/ipkg-x86_64/iptables-mod-fullconenat -name 'CVS' -o -name '.svn' -o -name '.#' -o -name '~'| xargs -r rm -rf
Package iptables-mod-fullconenat is missing dependencies for the following libraries:
libxtables.so.12
make[3]: *** [Makefile:69: /home/builder/lede_x86/bin/packages/x86_64/base/iptables-mod-fullconenat_2022-02-13-108a36cb-10_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/network/services/fullconenat'
time: package/network/services/fullconenat/compile#0.03#0.14#0.17
ERROR: package/network/services/fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/network/services/fullconenat/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

hzjnet, 2024-06-22:
DEPENDS:=+iptables +kmod-ipt-fullconenat

smallprogram, 2024-06-22:
+

GreatMichaelLee, 2024-06-22:
> DEPENDS:=+iptables +kmod-ipt-fullconenat

where,which?^^

hzjnet, 2024-06-22:
> package/network/services/fullconenat

package/network/services/fullconenat/Makefile

closed 2024-06-22 by coolsnowwolf