coolsnowwolf/lede #12264: 高通驱动“ath10k-ct”一直编译失败

sherlsenlinmu, 2024-06-23:
### 详细叙述

高通方案驱动涉及这个“ath10k-ct”的全部编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

acrh17

### 详细日志

make[2]: Circular package/kernel/ath10k-ct/compile <- package/kernel/mac80211/compile dependency dropped.
make[3]: Entering directory '/workdir/openwrt/package/kernel/ath10k-ct'
make[3]: *** No rule to make target '/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
make[3]: Leaving directory '/workdir/openwrt/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.15#0.07#0.20
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

hzjnet, 2024-06-23:
> ### 详细叙述
> 高通方案驱动涉及这个“ath10k-ct”的全部编译失败
> 
> ### 重复 issue
> * [ ]  没有类似的 issue
> 
> ### 具体型号
> acrh17
> 
> ### 详细日志
> make[2]: Circular package/kernel/ath10k-ct/compile <- package/kernel/mac80211/compile dependency dropped. make[3]: Entering directory '/workdir/openwrt/package/kernel/ath10k-ct' make[3]: *** No rule to make target '/workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/workdir/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'. Stop. make[3]: Leaving directory '/workdir/openwrt/package/kernel/ath10k-ct' time: package/kernel/ath10k-ct/regular/compile#0.15#0.07#0.20 ERROR: package/kernel/ath10k-ct failed to build (build variant: regular). make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1 make[2]: Leaving directory '/workdir/openwrt' make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2 make[1]: Leaving directory '/workdir/openwrt' make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

先用最简配置编译

sherlsenlinmu, 2024-06-23:
> 先用最简配置编译

感谢提醒，不过都试过了，只要涉及到这个驱动最后都卡这里了

1582130940, 2024-06-23:
make package/kernel/ath10k-ct/{clean,compile} V=s -j$(nproc)
单独编译这个试一下，能过其他的就能编译，只要不清就行

hzjnet, 2024-06-25:
我的尝试N次了，结果就是用最简配置编译才能成功，前提是初始源码即：make distclean

hzjnet, 2024-06-25:
> make package/kernel/ath10k-ct/{clean,compile} V=s -j$(nproc) 单独编译这个试一下，能过其他的就能编译，只要不清就行

尝试过这个，结果是不行

xuexijiaoben, 2024-07-02:
一样的问题

closed 2024-07-10 by coolsnowwolf

SQ528, 2024-07-26:
make[2]: Circular package/kernel/ath10k-ct/compile <- package/kernel/mac80211/compile dependency dropped.
make[3]: Entering directory '/workdir/openwrt/package/kernel/ath10k-ct'
make[3]: *** No rule to make target '/workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
make[3]: Leaving directory '/workdir/openwrt/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.16#0.06#0.20
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
3星期过去了还是不行！