coolsnowwolf/lede #12267: 6.1以后的内核都会不定时重启op

zsrjim, 2024-06-23:
### 详细叙述

6.1-6.6的内核编译出来的固件都能使用，但是不固定时间会自动重启op，虚拟机正常运行。内核改成5.4使用几天都不会重启。插件都是一样的。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

 pve8.22虚拟机,Intel(R) Pentium(R) Silver N6000

### 详细日志

[    0.000000] Linux version 6.6.35 (runner@fv-az714-319) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6688-de0b0aa0b) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT_DYNAMIC Sat Jun 22 18:39:58 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=8ca04ef0-df10-1ea8-3b46-512a0ecd4102 rootwait console=tty0 console=ttyS0,115200n8 noinitrd swiotlb=65536
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003ffd9fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ffda000-0x000000003fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: using sched offset of 4507889747 cycles
[    0.000002] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000005] tsc: Detected 1113.600 MHz processor
[    0.000534] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000537] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000540] last_pfn = 0x3ffda max_arch_pfn = 0x400000000
[    0.000570] MTRR map: 4 entries (3 fixed + 1 variable; max 19), built from 8 variable MTRRs
[    0.000573] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.003883] found SMP MP-table at [mem 0x000f5b70-0x000f5b7f]
[    0.004016] ACPI: Early table checksum verification disabled
[    0.004020] ACPI: RSDP 0x00000000000F5990 000014 (v00 BOCHS )
[    0.004024] ACPI: RSDT 0x000000003FFE2F80 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004029] ACPI: FACP 0x000000003FFE2D62 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004033] ACPI: DSDT 0x000000003FFDF040 003D22 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004036] ACPI: FACS 0x000000003FFDF000 000040
[    0.004038] ACPI: APIC 0x000000003FFE2DD6 000080 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004041] ACPI: SSDT 0x000000003FFE2E56 0000CA (v01 BOCHS  VMGENID  00000001 BXPC 00000001)
[    0.004043] ACPI: HPET 0x000000003FFE2F20 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004045] ACPI: WAET 0x000000003FFE2F58 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004047] ACPI: Reserving FACP table memory at [mem 0x3ffe2d62-0x3ffe2dd5]
[    0.004048] ACPI: Reserving DSDT table memory at [mem 0x3ffdf040-0x3ffe2d61]
[    0.004049] ACPI: Reserving FACS table memory at [mem 0x3ffdf000-0x3ffdf03f]
[    0.004050] ACPI: Reserving APIC table memory at [mem 0x3ffe2dd6-0x3ffe2e55]
[    0.004051] ACPI: Reserving SSDT table memory at [mem 0x3ffe2e56-0x3ffe2f1f]
[    0.004052] ACPI: Reserving HPET table memory at [mem 0x3ffe2f20-0x3ffe2f57]
[    0.004052] ACPI: Reserving WAET table memory at [mem 0x3ffe2f58-0x3ffe2f7f]
[    0.004100] Zone ranges:
[    0.004101]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.004103]   DMA32    [mem 0x0000000001000000-0x000000003ffd9fff]
[    0.004104]   Normal   empty
[    0.004105] Movable zone start for each node
[    0.004106] Early memory node ranges
[    0.004106]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.004107]   node   0: [mem 0x0000000000100000-0x000000003ffd9fff]
[    0.004108] Initmem setup node 0 [mem 0x0000000000001000-0x000000003ffd9fff]
[    0.004113] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.004138] On node 0, zone DMA: 97 pages in unavailable ranges
[    0.008274] On node 0, zone DMA32: 38 pages in unavailable ranges
[    0.009301] ACPI: PM-Timer IO Port: 0x608
[    0.009313] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
[... 551 lines cut ...]
[   12.894962] kmodloader: done loading kernel modules from /etc/modules.d/*
[   15.388148] 8021q: adding VLAN 0 to HW filter on device eth0
[   15.389886] br-lan: port 1(eth0) entered blocking state
[   15.390578] br-lan: port 1(eth0) entered disabled state
[   15.391238] virtio_net virtio1 eth0: entered allmulticast mode
[   15.392043] virtio_net virtio1 eth0: entered promiscuous mode
[   15.394958] br-lan: port 1(eth0) entered blocking state
[   15.395639] br-lan: port 1(eth0) entered forwarding state
[   16.336921] br-lan: port 1(eth0) entered disabled state
[   16.338726] virtio_net virtio1 eth0: left allmulticast mode
[   16.339705] virtio_net virtio1 eth0: left promiscuous mode
[   16.340673] br-lan: port 1(eth0) entered disabled state
[   17.670806] 8021q: adding VLAN 0 to HW filter on device eth0
[   17.672608] br-lan: port 1(eth0) entered blocking state
[   17.673308] br-lan: port 1(eth0) entered disabled state
[   17.673964] virtio_net virtio1 eth0: entered allmulticast mode
[   17.674761] virtio_net virtio1 eth0: entered promiscuous mode
[   17.676923] br-lan: port 1(eth0) entered blocking state
[   17.677620] br-lan: port 1(eth0) entered forwarding state
[   85.221454] br-lan: port 2(tap_lu) entered blocking state
[   85.222278] br-lan: port 2(tap_lu) entered disabled state
[   85.223051] tap_lu: entered allmulticast mode
[   85.223917] tap_lu: entered promiscuous mode
[   85.224740] br-lan: port 2(tap_lu) entered blocking state
[   85.225439] br-lan: port 2(tap_lu) entered forwarding state

zsrjim, 2024-06-23:
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.102200] ACPI: Core revision 20230628
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.102815] clocksource: hpet: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604467 ns
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.103976] APIC: Switch to symmetric I/O mode setup
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.104789] x2apic enabled
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.105481] APIC: Switched APIC routing to: physical x2apic
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.106147] kvm-guest: APIC: send_IPI_mask() replaced with kvm_send_ipi_mask()
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.107011] kvm-guest: APIC: send_IPI_mask_allbutself() replaced with kvm_send_ipi_mask_allbutself()
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.108023] kvm-guest: setup PV IPIs
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.109530] ..TIMER: vector=0x30 apic1=0 pin1=2 apic2=-1 pin2=-1
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.110255] clocksource: tsc-early: mask: 0xffffffffffffffff max_cycles: 0x100d48ce7a0, max_idle_ns: 440795212195 ns
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.111397] Calibrating delay loop (skipped) preset value.. 2227.20 BogoMIPS (lpj=11136000)
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.112558] x86/cpu: User Mode Instruction Prevention (UMIP) activated
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.113419] Last level iTLB entries: 4KB 0, 2MB 0, 4MB 0
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.121403] Last level dTLB entries: 4KB 0, 2MB 0, 4MB 0, 1GB 0
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.122061] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.123040] Spectre V2 : Spectre BHI mitigation: SW BHB clearing on vm exit
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.123812] Spectre V2 : Spectre BHI mitigation: SW BHB clearing on syscall
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.124574] Spectre V2 : Mitigation: Enhanced / Automatic IBRS
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.125243] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.126154] Spectre V2 : mitigation: Enabling conditional Indirect Branch Prediction Barrier
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.127087] Speculative Store Bypass: Mitigation: Speculative Store Bypass disabled via prctl
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.128034] MMIO Stale Data: Mitigation: Clear CPU buffers
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.128662] Register File Data Sampling: Vulnerable: No microcode
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.129349] SRBDS: Unknown: Dependent on hypervisor status
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.129952] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.130849] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.131683] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'compacted' format.
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.137410] Freeing SMP alternatives memory: 40K
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.137965] pid_max: default: 32768 minimum: 301
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.138584] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.139440] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.140607] smpboot: CPU0: Intel(R) Pentium(R) Silver N6000 @ 1.10GHz (family: 0x6, model: 0x9c, stepping: 0x0)
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] RCU Tasks: Setting shift to 1 and lim to 1 rcu_task_cb_adjust=1.
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] RCU Tasks Trace: Setting shift to 1 and lim to 1 rcu_task_cb_adjust=1.
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] Performance Events: Tremont events, full-width counters, Intel PMU driver.
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] ... version:                2
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] ... bit width:              48
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] ... generic registers:      4
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] ... value mask:             0000ffffffffffff
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] ... max period:             00007fffffffffff
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] ... fixed-purpose events:   3
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] ... event mask:             000000070000000f
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] signal: max sigframe size: 1104
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] rcu: Hierarchical SRCU implementation.
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] rcu: 	Max phase no-delay instances is 1000.
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141397] smp: Bringing up secondary CPUs ...
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.141741] smpboot: x86: Booting SMP configuration:
Mon Jun 24 01:04:00 2024 kern.info kernel: [    0.142348] .... node  #0, CPUs:      #1
[... cut ...]
Mon Jun 24 01:05:26 2024 daemon.err sing-box[8007]: [31mFATAL[0m[0000] read config at ./config.json: open ./config.json: no such file or directory
Mon Jun 24 01:05:31 2024 daemon.err sing-box[8025]: [31mFATAL[0m[0000] read config at ./config.json: open ./config.json: no such file or directory
Mon Jun 24 01:05:36 2024 daemon.err sing-box[8043]: [31mFATAL[0m[0000] read config at ./config.json: open ./config.json: no such file or directory
Mon Jun 24 01:05:36 2024 daemon.info procd: Instance sing-box::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

brianzjh, 2024-06-24:
是不是硬件的问题啊！我用的是N100，就没这个问题，也是用pve

zsrjim, 2024-06-24:
> 是不是硬件的问题啊！我用的是N100，就没这个问题，也是用pve

虚拟机本身没有重启。自动重启的只有openwrt。

brianzjh, 2024-06-24:
> > 是不是硬件的问题啊！我用的是N100，就没这个问题，也是用pve
> 
> 虚拟机本身没有重启。自动重启的只有openwrt。

我的x86软路由完全没有这个问题，升级硬件吧！使劲的折腾

yhl452493373, 2024-06-24:
> 是不是硬件的问题啊！我用的是N100，就没这个问题，也是用pve

J4205同样这个问题

GreatMichaelLee, 2024-06-24:
升级N6000 CPU的微码 两个步骤 1）找到你的软路由机器的厂家找他们要包含Intel CPU微码版本高于0x2400024的BIOS(没有的话，强烈要求他们更新） 2) 虚拟主机操作系统升级微码
PVE直接参考这个：https://github.com/intel/Intel-Linux-Processor-Microcode-Data-Files/blob/main/README.md
ESXI的有点不同，参考这个：https://larkin.wang/archives/49/
ESXI的升级完了是这样：
[image]
软路由虚拟机内部也能查询同一版本
[image]

zsrjim, 2024-06-24:
> Intel CPU已经明确虚拟机崩溃是由于微码问题造成，解决此问题，需要升级N5105/6000 CPU的微码 两个方法：
> 
> 1. 找到你的软路由机器的厂家找他们要包含Intel CPU微码版本高于0x24000024（目前最新版本为0x24000026) 的BIOS(没有的话，强烈要求他们更新）升级BIOS各厂家步骤方法不同，各自找吧
> 2. 或者升级虚拟主机操作系统内置微码
> 
> * PVE直接参考这个：https://play.gamerblog.us/2023/04/18/pve%E6%9B%B4%E6%96%B0cpu%E5%BE%AE%E7%A0%81%E5%BD%BB%E5%BA%95%E8%A7%A3%E5%86%B3n5105%E7%9A%84vm%E6%AD%BB%E6%9C%BA%E9%97%AE%E9%A2%98/  PVE升级完了是这样的（我的这个PVE主机不是N系列CPU,我贴的图只是个我的机器示例，升级完了运行下面dmesg命令应该显示0x24000024/26)
>   ![image](https://private-user-images.githubusercontent.com/20067401/342406372-685a24ea-1a47-4896-b025-7814943f125d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyNzA1NzgsIm5iZiI6MTcxOTI3MDI3OCwicGF0aCI6Ii8yMDA2NzQwMS8zNDI0MDYzNzItNjg1YTI0ZWEtMWE0Ny00ODk2LWIwMjUtNzgxNDk0M2YxMjVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI0VDIzMDQzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRlNzdmYTM0ZGU0NmZkNGE0ZGU4NTcyMTcwMTgxZTBiNjdkNGU0MWMxYTVhZTgxMDQ0YmY0NGYyNzg1MmI4Y2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.eB54ZAg5ajRiF1o8jtujjmrfScXn6Wlk0iWMEHv5o_g)
> * ESXI的有点不同，参考这个：https://larkin.wang/archives/49/  ESXI的升级完了运行图里的命令，应该是图示这样(我的ESXI是N5105,对应CPU型号06-9c-00：
> 
> ![image](https://private-user-images.githubusercontent.com/20067401/342395617-7855a99b-7254-4988-a82a-b3153eb541f5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyNzA1NzgsIm5iZiI6MTcxOTI3MDI3OCwicGF0aCI6Ii8yMDA2NzQwMS8zNDIzOTU2MTctNzg1NWE5OWItNzI1NC00OTg4LWE4MmEtYjMxNTNlYjU0MWY1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI0VDIzMDQzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVlMGUzMThkMWZkMzZkMjZiZjFhNmE4YThhNzJmYjEwYzFmYzI1MDQ0NzU0MWYxYTk1NTY4OWVmYmJmZjdjNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ioCqHk_XXAxGWg12H8POf42xpcJhNA6gPVPKNVUmpeE)
> 
> * 软路由虚拟机内部也能查询同一版本
> 
> ![image](https://private-user-images.githubusercontent.com/20067401/342396229-17851c73-c790-4660-8711-e2a2ab844721.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyNzA1NzgsIm5iZiI6MTcxOTI3MDI3OCwicGF0aCI6Ii8yMDA2NzQwMS8zNDIzOTYyMjktMTc4NTFjNzMtYzc5MC00NjYwLTg3MTEtZTJhMmFiODQ0NzIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI0VDIzMDQzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBiYzQyYzI4Y2Y3ZTM3MjU4MTFhMzRhYThiYWQ4Mzg4ZmZmMmFlMzVhNWZiZDllNTQ4NTk4OWEyN2E1MWZmYjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Dh8DL20_XD3Jf15X0lx45uqhavn4Bzd8igWwLr17E1M)

[Tue Jun 25 07:18:18 2024] SRBDS: Vulnerable: No microcode
[Tue Jun 25 07:18:18 2024] microcode: Current revision: 0x24000026
[Tue Jun 25 07:18:18 2024] microcode: Updated early from: 0x24000024
好像不行PVE的版本已经是024的了、op还是会重启不定时的

GreatMichaelLee, 2024-06-24:
> [Tue Jun 25 07:18:18 2024] SRBDS: Vulnerable: No microcode [Tue Jun 25 07:18:18 2024] microcode: Current revision: 0x24000026 [Tue Jun 25 07:18:18 2024] microcode: Updated early from: 0x24000024 [...]

升级到026，我之前很早就升级了024一直都是好的，运行了好多天都没事，但最近几个linux内核版本又开始随机崩了（当然没有升024之前那么频繁），升级了026，又好了。我看你刚升了026，观察一下

zsrjim, 2024-06-25:
> 升级到026，我之前很早就升级了024一直都是好的，运行了好多天都没事，但最近几个linux内核版本又开始随机崩了（当然没有升024之前那么频繁），升级了026，又好了。我看你刚升了026，观察一下

测试了还是会自动重启，编译的新内核6.6.35的

brianzjh, 2024-06-25:
> 本已经是024的了、op还是会重启不定时的

看来是硬件和bios问题，我说我的N100就没这个问题

zsrjim, 2024-06-25:
> > 本已经是024的了、op还是会重启不定时的
> 
> 看来是硬件和bios问题，我说我的N100就没这个问题

你说是硬件问题吧用LXC的版本OP又没问题。。。。。。。

brianzjh, 2024-06-25:
> > > 本已经是024的了、op还是会重启不定时的
> > 
> > 
> > 看来是硬件和bios问题，我说我的N100就没这个问题
> 
> 你说是硬件问题吧用LXC的版本OP又没问题。。。。。。。

说实话，我看过lxc的op使用方法，觉得太麻烦了，您那里有简易的使用方法吗？？请教了

brianzjh, 2024-06-25:
> 虚拟机是正常运行的没有重启

我就是在pve上的虚拟机里用的op，网卡直通的，性能也还可以。lxc看着太复杂，就没用

zsrjim, 2024-06-25:
> > 虚拟机是正常运行的没有重启
> 
> 我就是在pve上的虚拟机里用的op，网卡直通的，性能也还可以。lxc看着太复杂，就没用

找一个带脚本的库很简单。

brianzjh, 2024-06-25:
> > > 虚拟机是正常运行的没有重启
> > 
> > 
> > 我就是在pve上的虚拟机里用的op，网卡直通的，性能也还可以。lxc看着太复杂，就没用
> 
> 找一个带脚本的库很简单。

求脚本

GreatMichaelLee, 2024-06-25:
> 终于有人发现这个问题了，我的设备是r86s，n5105，esxi8.0安装6.1内核不定时在esxi上op重启现象，重启次数多就直接挂掉，然后更新bios，刷回5.4，没有出现无故重启的现象了，不行就用esxi，省心方便

我的是N5105, esxi 8.0UC2,  软路由内核6.6.35， 微码0x24000026, 稳的一批。

GreatMichaelLee, 2024-06-25:
> 测试了还是会自动重启，编译的新内核6.6.35的

那你这个很有可能是硬件问题了。

zsrjim, 2024-06-25:
> 那你这个很有可能是硬件问题了。

用lxc版的就没问题😂😂😂😂

loydcheung, 2024-06-25:
> > 终于有人发现这个问题了，我的设备是r86s，n5105，esxi8.0安装6.1内核不定时在esxi上op重启现象，重启次数多就直接挂掉，然后更新bios，刷回5.4，没有出现无故重启的现象了，不行就用esxi，省心方便
> 
> 
> 
> 我的是N5105, esxi 8.0UC2,  软路由内核6.6.35， 微码0x24000026, 稳的一批。

我的国伟J4125物理机2022年初买的，从来没更新过bios自查了一下，微码0x1a，是不是超级老？

Wilson-coder, 2024-06-26:
我来补充一些信息，一个工控机3205U的芯片，和一个戴尔的11代笔记本，都是物理刷Openwrt，没有任何虚拟机。用6.1内核的话，一直出现随机重启的现象。改成5.15，已经稳定使用很多天了。原来一直以为是我自己选的插件可能有啥冲突，不过最近事情太多没时间去排查，看到这个帖子才知道不是我一个人的问题。

zsrjim, 2024-06-26:
> 我来补充一些信息，一个工控机3205U的芯片，和一个戴尔的11代笔记本，都是物理刷Openwrt，没有任何虚拟机。用6.1内核的话，一直出现随机重启的现象。改成5.15，已经稳定使用很多天了。原来一直以为是我自己选的插件可能有啥冲突，不过最近事情太多没时间去排查，看到这个帖子才知道不是我一个人的问题。

我的问题是不选插件直接进去配置好网络也会重启。自带的东西没动。

zuni11, 2024-06-26:
J4125，微码0x1a，这两天编译的，也会自己重启，内核6.1.95

GreatMichaelLee, 2024-06-27:
> 我的 J4125 和朋友的 3865U 从 6.1.92 开始就遇到重启的问题，重启时间不固定，均是物理机安装，目前切到 6.6 稳定了一段时间了，不确定是什么问题。 @coolsnowwolf

看样子6.1的内核问题多，建议要么就用5.x要么用6.6吧

loydcheung, 2024-06-27:
我是J4125物理机，6.1内核目前没有遇到过重启问题。但我一直在编译时有选中firmware菜单里的intel- microcode，不知道这个是不是对大家有帮助。因为微码是可以在bios也可以在操作系统层面解决的。

GreatMichaelLee, 2024-06-27:
> dmesg > /root/dmesg.log

没用的，啥都没有

bob-zebedy, 2024-06-27:
[image]

j4125 物理机直装已经连续 182 天不间断了

zuni11, 2024-06-27:
> ![Google Chrome-OpenWrt - 总览 - LuCI-20240627102601-OvDxbuvC@2x](https://private-user-images.githubusercontent.com/53208952/343606758-5353b1f8-401b-4cb8-83fd-c4b4e992a65f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk0NTgwMTgsIm5iZiI6MTcxOTQ1NzcxOCwicGF0aCI6Ii81MzIwODk1Mi8zNDM2MDY3NTgtNTM1M2IxZjgtNDAxYi00Y2I4LTgzZmQtYzRiNGU5OTJhNjVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI3VDAzMDgzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUxNTgwMGJkOTVlNjM4OGMyZTFmYmU2MDRkNWFjODgzMWFiYzBjOGI3MjcxNWNlZTc3YWIyNDhmNTJjZTVlYjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.DVdtdLQkxF9Jg4H539kA0JKF6D0Oi8_0U7Qp9eXFPDg)
> 
> j4125 物理机直装已经连续 172 天不间断了

我是从内核6.1.94开始发现重启的

GreatMichaelLee, 2024-06-28:
> ![Google Chrome-OpenWrt - 总览 - [...]
直装没有问题，是虚拟机的问题。

GreatMichaelLee, 2024-06-28:
> 我是J4125物理机，6.1内核目前没有遇到过重启问题。但我一直在编译时有选中firmware菜单里的intel- microcode，不知道这个是不是对大家有帮助。因为微码是可以在bios也可以在操作系统层面解决的。

没有用，虚拟机崩溃，微码要在host主机层面patch和运作，在虚拟机内部不起作用。

loydcheung, 2024-06-28:
> > 我是J4125物理机，6.1内核目前没有遇到过重启问题。但我一直在编译时有选中firmware菜单里的intel- microcode，不知道这个是不是对大家有帮助。因为微码是可以在bios也可以在操作系统层面解决的。
> 
> 
> 
> 没有用，虚拟机崩溃，微码要在host主机层面patch和运作，在虚拟机内部不起作用。

说的对，但我是直装的，所以证明编入微码这招应该对物理机随机重启的问题有用。

bob-zebedy, 2024-06-28:
> ![Google Chrome-OpenWrt - 总览 - [...]

soga，我说怎么我都没遇到过

dragon888888888888, 2024-06-28:
一样的，不过我的是ARM的 5.15内核不重启，但最新源码已无法编译成功了，6.6的源码编译没问题，总是会自动重启

jimgo188, 2024-06-29:
我一直以为就我有这个问题，6.1内核，那我也换5.15内核看看了

zsrjim, 2024-06-29:
> 我一直以为就我有这个问题，6.1内核，那我也换5.15内核看看了

5.4的内核也没问题，我都测试过了。就6.1，6.6的会重启，微码都刷最新了。

jimgo188, 2024-06-29:
我是X86物理安装的型号是3965U，不定期重启，有时晚上重启最讨厌，游戏魔兽直接掉线了。重排好麻烦

TheLastManT1, 2024-06-29:
e5表示没有问题，我已经运行一个月了，用的pve
[image]

jimgo188, 2024-07-01:
我也是升到6.1.95就异常重启，可能是这个内核版本的问题，现在我暂时用的5.15没这个问题了

closed 2024-07-04 by coolsnowwolf

kornkaida, 2024-11-05:
N100，为什么我的微码显示跟大家都不一样
[Tue Nov  5 21:38:54 2024] microcode: Current revision: 0x0000001a
[Tue Nov  5 21:38:54 2024] microcode: Updated early from: 0x00000010
这个是最新版么

Fyred, 2025-04-15:
关SFE换成软件加速保平安