coolsnowwolf/lede #12268: 更新到最新源码AX6（ipq807x）编译出错

mitsukileung, 2024-06-23:
### 详细叙述

更新到最新源码AX6（ipq807x）编译出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Ax6

### 详细日志

2024-06-23T16:49:02.1228271Z make[3]: Entering directory '/workdir/openwrt/package/qca/qca-nss-ecm'
2024-06-23T16:49:02.2307158Z rm -f /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-nss-ecm-2023-10-20-82b27915/.built
2024-06-23T16:49:02.2325950Z touch /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-nss-ecm-2023-10-20-82b27915/.built_check
2024-06-23T16:49:02.2353963Z make -C "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.95" ECM_FRONT_END_NSS_ENABLE=y ECM_FRONT_END_SFE_ENABLE=n ECM_NON_PORTED_SUPPORT_ENABLE=y ECM_INTERFACE_BOND_ENABLE=y ECM_INTERFACE_VLAN_ENABLE=y ECM_CLASSIFIER_MARK_ENABLE=y ECM_CLASSIFIER_DSCP_ENABLE=y ECM_CLASSIFIER_PCC_ENABLE=n ECM_BAND_STEERING_ENABLE=n ECM_IPV6_ENABLE=y ECM_INTERFACE_MACVLAN_ENABLE=y ECM_INTERFACE_MAP_T_ENABLE=y ECM_INTERFACE_IPSEC_ENABLE=y ECM_INTERFACE_PPPOE_ENABLE=y ECM_INTERFACE_PPTP_ENABLE=y ECM_INTERFACE_PPP_ENABLE=y ECM_INTERFACE_L2TPV2_ENABLE=y ECM_INTERFACE_GRE_TAP_ENABLE=y ECM_INTERFACE_GRE_TUN_ENABLE=y ECM_INTERFACE_SIT_ENABLE=y ECM_INTERFACE_TUNIPIP6_ENABLE=y CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl=target-aarch64_cortex-a53_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jun 22 18:39:58 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/symvers/qca-nss-clients.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/symvers/fullconenat.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/symvers/qca-nss-drv.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/symvers/nat46.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/symvers/mac80211.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/symvers/qca-nss-dp.symvers /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/symvers/qca-ssdk.symvers" KERNELRELEASE=6.1.95 M="/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-nss-ecm-2023-10-20-82b27915" EXTRA_CFLAGS="-I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/qca-nss-drv -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/qca-mcs -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/nat46 -Wno-error=unused-function " SoC="ipq807x"  modules
2024-06-23T16:49:02.2368141Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.95'
2024-06-23T16:49:02.4402850Z   CC [M]  /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-nss-ecm-2023-10-20-82b27915/frontends/cmn/ecm_ported_ipv4.o
2024-06-23T16:49:02.9868882Z In file included from /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-nss-ecm-2023-10-20-82b27915/frontends/cmn/ecm_ported_ipv4.c:110:
2024-06-23T16:49:02.9873404Z /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-nss-ecm-2023-10-20-82b27915/frontends/nss/ecm_nss_common.h:20:10: fatal error: nat46-core.h: No such file or directory
2024-06-23T16:49:02.9876814Z    20 | #include <nat46-core.h>
2024-06-23T16:49:02.9877576Z       |          ^~~~~~~~~~~~~~
2024-06-23T16:49:02.9878463Z compilation terminated.
2024-06-23T16:49:02.9923892Z make[5]: *** [scripts/Makefile.build:250: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-nss-ecm-2023-10-20-82b27915/frontends/cmn/ecm_ported_ipv4.o] Error 1
2024-06-23T16:49:02.9926500Z make[4]: *** [Makefile:2010: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-nss-ecm-2023-10-20-82b27915] Error 2
2024-06-23T16:49:02.9928774Z make[3]: *** [Makefile:164: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/qca-nss-ecm-2023-10-20-82b27915/.built] Error 2
2024-06-23T16:49:02.9930668Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq807x/linux-6.1.95'
2024-06-23T16:49:02.9931944Z make[3]: Leaving directory '/workdir/openwrt/package/qca/qca-nss-ecm'
2024-06-23T16:49:02.9932722Z time: package/qca/qca-nss-ecm/compile#0.78#0.10#0.87
2024-06-23T16:49:02.9935156Z     ERROR: package/qca/qca-nss-ecm failed to build.
2024-06-23T16:49:02.9937338Z make[2]: *** [package/Makefile:116: package/qca/qca-nss-ecm/compile] Error 1
2024-06-23T16:49:02.9941759Z make[2]: Leaving directory '/workdir/openwrt'
2024-06-23T16:49:02.9960271Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
2024-06-23T16:49:02.9965543Z make[1]: Leaving directory '/workdir/openwrt'
2024-06-23T16:49:02.9980927Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
2024-06-23T16:49:03.0013860Z ##[error]Process completed with exit code 2.

AXBNG, 2024-06-23:
不会吧，我早上刚自动编译完

brianzjh, 2024-06-24:
> 不会吧，我早上刚自动编译完

你应该不是红米ax6设备吧！我重新下载源和清除缓存都不行，跟他这个错误一样

AXBNG, 2024-06-24:
> > 不会吧，我早上刚自动编译完
> 
> 你应该不是红米ax6设备吧！我重新下载源和清除缓存都不行，跟他这个错误一样

你但凡进我主页看一眼也不会说这话

brianzjh, 2024-06-24:
> > > 不会吧，我早上刚自动编译完
> > 
> > 
> > 你应该不是红米ax6设备吧！我重新下载源和清除缓存都不行，跟他这个错误一样
> 
> 你但凡进我主页看一眼也不会说这话

那为啥我重新下载源码和清缓存，都不行啊！！难道还需要把系统重新做一遍啊！再说了哥们，别人没义务去关注你的信息啊！对吧

brianzjh, 2024-06-24:
> > > 不会吧，我早上刚自动编译完
> > 
> > 
> > 你应该不是红米ax6设备吧！我重新下载源和清除缓存都不行，跟他这个错误一样
> 
> 你但凡进我主页看一眼也不会说这话

我之前也是编译没有问题的，就是更新了最新的qca部分后，报的错误跟他反馈的一样！难道是巧合吗？？

AXBNG, 2024-06-24:
> > > > 不会吧，我早上刚自动编译完
> > > 
> > > 
> > > 你应该不是红米ax6设备吧！我重新下载源和清除缓存都不行，跟他这个错误一样
> > 
> > 
> > 你但凡进我主页看一眼也不会说这话
> 
> 那为啥我重新下载源码和清缓存，都不行啊！！难道还需要把系统重新做一遍啊！再说了哥们，别人没义务去关注你的信息啊！对吧

[image]

无语......这issue反馈的就是AX6，我不是我费那力气敲键盘？我说我的action通过了那源码应该没什么问题，你还扯上义务了，搞得好像我求你帮忙一样......

brianzjh, 2024-06-24:
> ![image](https://private-user-images.githubusercontent.com/89022914/342181928-bbaa01f8-5f33-45a9-8d67-20599dd67eb9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

这种无意义的回复就到此为止吧！你没求我，我也没求你，但我没义务了解你的事，明白吗？？我就是就事说事，接不接受都与我的问题无关

AXBNG, 2024-06-24:
> 这种无意义的回复就到此为止吧！你没求我，我也没求你，但我没义务了解你的事，明白吗？？我就是就事说事，接不接受都与我的问题无关

先撩者贱，眼瞎者蠢

brianzjh, 2024-06-24:
> 先撩者贱，眼瞎者蠢

自负者愚，口嗨者傻！！shut you！来而不往而礼也

brianzjh, 2024-06-24:
> 先撩者贱，眼瞎者蠢

想嘴巴不干净，老子就给你点教训！！你没资格让爷关注你，知道吗

brianzjh, 2024-06-24:
> 先撩者贱，眼瞎者蠢

再给我嘴巴不干净，我就今天在这里给你打个样，操到你脸上就可以了，不信你就试试看

AXBNG, 2024-06-24:
> 再给我嘴巴不干净，我就今天在这里给你打个样，操到你脸上就可以了，不信你就试试看

哪儿来的疯狗，先是眼瞎，再是嘴贱，无能狂怒，贻笑大方

AXBNG, 2024-06-24:
> 想嘴巴不干净，老子就给你点教训！！你没资格让爷关注你，知道吗

一口一个爷，现实中当孙子当久了是吧，这么牛逼怎么不自己做源码嘞？隔着网线给谁教训呐666

AXBNG, 2024-06-24:
> 自负者愚，口嗨者傻！！shut you！来而不往而礼也

这辈子就会这两个单词了吧，听多了就记住了还挺聪明呢，快赶上人类了，加油

brianzjh, 2024-06-24:
> 哪儿来的疯狗，先是眼瞎，再是嘴贱，无能狂怒，贻笑大方

我就是来打你这疯狗，自卑自负的狗的，你以为你什么东西啊！我非要看你的狗内容吗？？先看看自己啥货色吧

brianzjh, 2024-06-24:
> 一口一个爷，现实中当孙子当久了是吧，这么牛逼怎么不自己做源码嘞？隔着网线给谁教训呐666

至少我家狗操了你奶奶，才是你的爷！你这话说的，你那么牛逼让别人关注你这狗东西，你为啥不做源码啊？？先嘴贱者，我必操之

brianzjh, 2024-06-24:
> 这辈子就会这两个单词了吧，听多了就记住了还挺聪明呢，快赶上人类了，加油

会什么，不用你这种低级货色关心，你也没那个资格关心。看看你自己的嘴贱记录吧！现实中活的那个逼样，舔不知耻的以为别人关注你。。。。。。我为啥要关注一条狗啊！！你自己说这话，不觉得自己是个大傻逼吗

brianzjh, 2024-06-24:
> 这辈子就会这两个单词了吧，听多了就记住了还挺聪明呢，快赶上人类了，加油

自己做什么职业的？什么收入？自己心里没点逼数吗？？在这里狗叫！没事，爷就在这里打你这条狗

brianzjh, 2024-06-24:
> 这辈子就会这两个单词了吧，听多了就记住了还挺聪明呢，快赶上人类了，加油

先去回家问问你妈和你奶奶，是不是被我养的狗操了

brianzjh, 2024-06-24:
> 这辈子就会这两个单词了吧，听多了就记住了还挺聪明呢，快赶上人类了，加油

自己嘴贱，狗屁都不是还自负，还先骂人！！我有惯着你的狗嘴毛病的义务吗？？今天就让你知道，我有打你狗嘴的权力

AXBNG, 2024-06-24:
> 自己做什么职业的？什么收入？自己心里没点逼数吗？？在这里狗叫！没事，爷就在这里打你这条狗

笑了，乱伦产物是这样的，自己眼瞎可以，别人怼你就炸毛，自己在这儿高潮吧傻卵，骂人都这么低级，完全激不起我的兴趣啊网蛆，找不到亲爹也不用见着个男人就高潮吧，你满嘴喷粪的样子和您母亲潮喷的姿态倒是有几分相似，您爷爷应该很熟悉~~~

AXBNG, 2024-06-24:
> 自己嘴贱，狗屁都不是还自负，还先骂人！！我有惯着你的狗嘴毛病的义务吗？？今天就让你知道，我有打你狗嘴的权力

你有鸡毛权力啊哈哈哈哈哈哈，还倒打一耙，你丫是小日本后代吧这么会扭曲事实

AXBNG, 2024-06-24:
> 会什么，不用你这种低级货色关心，你也没那个资格关心。看看你自己的嘴贱记录吧！现实中活的那个逼样，舔不知耻的以为别人关注你。。。。。。我为啥要关注一条狗啊！！你自己说这话，不觉得自己是个大傻逼吗

不是，怎么骂来骂去句句不离狗啊，你跟狗有啥深仇大恨呐，全家被狗日了？

AXBNG, 2024-06-24:
> 先去回家问问你妈和你奶奶，是不是被我养的狗操了

你妈你奶你姥姥让我别跟你一般见识，说你是她们和狗的产物，脑子不正常

AXBNG, 2024-06-24:
> 至少我家狗操了你奶奶，才是你的爷！你这话说的，你那么牛逼让别人关注你这狗东西，你为啥不做源码啊？？先嘴贱者，我必操之

骂人还要鹦鹉学舌？？？你这水平够次的也是，回家让你妈妈教教你口活儿再来吧

AXBNG, 2024-06-24:
> 自己嘴贱，狗屁都不是还自负，还先骂人！！我有惯着你的狗嘴毛病的义务吗？？今天就让你知道，我有打你狗嘴的权力

快点打字啊傻吊，我在这儿刷新好多次了，哦对了，我可不会自称是你爷，毕竟你是小日本儿杂种，我嫌脏啊哈哈哈

AXBNG, 2024-06-24:
> 自己嘴贱，狗屁都不是还自负，还先骂人！！我有惯着你的狗嘴毛病的义务吗？？今天就让你知道，我有打你狗嘴的权力

真没意思，粪坑下边信号不好断网了？

AXBNG, 2024-06-24:
> 自己嘴贱，狗屁都不是还自负，还先骂人！！我有惯着你的狗嘴毛病的义务吗？？今天就让你知道，我有打你狗嘴的权力

战五渣真让人犯困，不会再回你了傻卵，你回复我一次家里户口本少一页，自己慢慢玩吧~~~

brianzjh, 2024-06-24:
> 笑了，乱伦产物是这样的，自己眼瞎可以，别人怼你就炸毛，自己在这儿高潮吧傻卵，骂人都这么低级，完全激不起我的兴趣啊网蛆，找不到亲爹也不用见着个男人就高潮吧，你满嘴喷粪的样子和您母亲潮喷的姿态倒是有几分相似，您爷爷应该很熟悉~~~

笑什么啊！是不是在笑我家狗操了你妈之后产出你这么个玩意啊！我再告诉你这货色一次，你不是我要关注的对像，也没那个价值让我关注！所以，我就会操你脸上！以后记住这次，再跟别人得瑟你的自卑的时候，小心再次再操脸上，这次就是例子！去问你妈妈的爽吧

brianzjh, 2024-06-24:
> > > > > > > > > > > 不会吧，我早上刚自动编译完
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > 你应该不是红米ax6设备吧！我重新下载源和清除缓存都不行，跟他这个错误一样
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > 你但凡进我主页看一眼也不会说这话
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 那为啥我重新下载源码和清缓存，都不行啊！！难道还需要把系统重新做一遍啊！再说了哥们，别人没义务去关注你的信息啊！对吧
> > > > > > > 
> > > > > > > 
> > > > > > > ![image](https://private-user-images.githubusercontent.com/89022914/342181928-bbaa01f8-5f33-45a9-8d67-20599dd67eb9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkxOTc1NDMsIm5iZiI6MTcxOTE5NzI0MywicGF0aCI6Ii84OTAyMjkxNC8zNDIxODE5MjgtYmJhYTAxZjgtNWYzMy00NWE5LThkNjctMjA1OTlkZDY3ZWI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI0VDAyNDcyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUxMGI5YTE0MzE5ZDZiNTMxZTY3ZTA3MDRjMWMyNTE4YmFmNWQxYTVjMjY4ZmJhM2RkYTQ1YjIyYjJhYzBhOWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5I8tlhKgS6P4A5PxMt67piW5l90bQlGivpddfayMKGg)
> > > > > > > 无语......这issue反馈的就是AX6，我不是我费那力气敲键盘？我说我的action通过了那源码应该没什么问题，你还扯上义务了，搞得好像我求你帮忙一样......
> > > > > > 
> > > > > > 
> > > > > > 这种无意义的回复就到此为止吧！你没求我，我也没求你，但我没义务了解你的事，明白吗？？我就是就事说事，接不接受都与我的问题无关
> > > > > 
> > > > > 
> > > > > 先撩者贱，眼瞎者蠢
> > > > 
> > > > 
> > > > 自负者愚，口嗨者傻！！shut you！来而不往而礼也
> > > 
> > > 
> > > 这辈子就会这两个单词了吧，听多了就记住了还挺聪明呢，快赶上人类了，加油
> > 
> > 
> > 自己嘴贱，狗屁都不是还自负，还先骂人！！我有惯着你的狗嘴毛病的义务吗？？今天就让你知道，我有打你狗嘴的权力
> 
> 战五渣真让人犯困，不会再回你了傻卵，你回复我一次你家里户口本就少一页，自己慢慢玩吧~~~

> 你有鸡毛权力啊哈哈哈哈哈哈，还倒打一耙，你丫是小日本后代吧这么会扭曲事实

就是有这种操你脸上的权力，你不要把你自己家的身事说出来，那么煞笔

brianzjh, 2024-06-24:
> 不是，怎么骂来骂去句句不离狗啊，你跟狗有啥深仇大恨呐，全家被狗日了？

不就是因为我家狗操你妈了，才长出你这种狗尿胎吗？？

brianzjh, 2024-06-24:
> 真没意思，粪坑下边信号不好断网了？

真不是，我在观察我家狗操你妈呢

brianzjh, 2024-06-24:
> 战五渣真让人犯困，不会再回你了傻卵，你回复我一次你家里户口本就少一页，自己慢慢玩吧~~~

那是谁啊！总以为人不遛它，它就觉得自己又行了。。。。。你的嘴巴能开光，我就能把你家户口本，就像杀乌杂一样的杀掉

brianzjh, 2024-06-24:
> 战五渣真让人犯困，不会再回你了傻卵，你回复我一次你家里户口本就少一页，自己慢慢玩吧~~~

对了，你这个AX开头，是不是就是你妈被我家狗操的时候的“哎呦喂”XXXXX啊！要不我拍个视频让你看看呗，你妈与狗

brianzjh, 2024-06-24:
> 骂人还要鹦鹉学舌？？？你这水平够次的也是，回家让你妈妈教教你口活儿再来吧

你说说，你是一个什么穷逼狗东西啊！天天在网上喷大便，要是没钱盖你的狗窝，就让爷给你放狗过去，操一下你妈不就行了吗？？爷很大方的

brianzjh, 2024-06-24:
> 快点打字啊傻吊，我在这儿刷新好多次了，哦对了，我可不会自称是你爷，毕竟你是小日本儿杂种，我嫌脏啊哈哈哈

我也等你半天了，狗东西，今天一定把你操翻

brianzjh, 2024-06-24:
> 真没意思，粪坑下边信号不好断网了？

自卑的狗东西，爷就是不需要关注你，爷没这个义务，请去你的18线的狗窝里去秀你的自卑去吧

AXBNG, 2024-06-24:
> 自卑的狗东西，爷就是不需要关注你，爷没这个义务，请去你的18线的狗窝里去秀你的自卑去吧

无能狂怒，笑死我了，谁要你这个杂种关注啊我都嫌脏

AXBNG, 2024-06-24:
> 我也等你半天了，狗东西，今天一定把你操翻

好家伙，这是天天被别人操翻不敢吱声，跑网上咬人来了，真可悲啧啧啧

AXBNG, 2024-06-24:
> 我也等你半天了，狗东西，今天一定把你操翻

你等个几把等啊，这么半天都硬不起来，你妈撅着屁股等得水都干了，小残废

brianzjh, 2024-06-24:
> 无能狂怒，笑死我了，谁要你这个杂种关注啊我都嫌脏

一条狗在一个18线的井里，充斥着自己的自卑，还企图让人类注意它！！可笑，煞笔

AXBNG, 2024-06-24:
> 你说说，你是一个什么穷逼狗东西啊！天天在网上喷大便，要是没钱盖你的狗窝，就让爷给你放狗过去，操一下你妈不就行了吗？？爷很大方的

是吗？你妈说你很小气欸，精子活跃度为 0

brianzjh, 2024-06-24:
> 你等个几把等啊，这么半天都硬不起来，你妈撅着屁股等得水都干了，小残废

我刚才在观察我家狗操你奶奶呢，不跟你说了吗？？你眼瞎啊！！我操你脸上一次，我家狗的毛就掉一根，因为它是你爷爷啊

AXBNG, 2024-06-24:
> 对了，你这个AX开头，是不是就是你妈被我家狗操的时候的“哎呦喂”XXXXX啊！要不我拍个视频让你看看呗，你妈与狗

我这名字是把你妈按在键盘上的时候乱打出来的，到底什么意思你得回家问你妈，哦不，得去凤楼问

brianzjh, 2024-06-24:
> 是吗？你妈说你很小气欸，精子活跃度为 0

有啥怀疑的呢？？很活跃的，不然也不会有你这种狗尿胎啊

brianzjh, 2024-06-24:
> 我这名字是把你妈按在键盘上的时候乱打出来的，到底什么意思你得回家问你妈，哦不，得去凤楼问

乱打的啊！怪不得呢！是不是被我家狗操的妈暗爽啊

brianzjh, 2024-06-24:
> 我这名字是把你妈按在键盘上的时候乱打出来的，到底什么意思你得回家问你妈，哦不，得去凤楼问

你这狗爪子打字真慢，让你操你脸上都没劲

AXBNG, 2024-06-24:
> 有啥怀疑的呢？？很活跃的，不然也不会有你这种狗尿胎啊

你得问你妈啊，你家的传统，不孕不育儿孙满堂哈哈哈

AXBNG, 2024-06-24:
> 乱打的啊！怪不得呢！是不是被我家狗操的妈暗爽啊

你妈说一点都不爽，但是比你的强

AXBNG, 2024-06-24:
> 你这狗爪子打字真慢，让你操你脸上都没劲

我可不操你脸上，嫌恶心

brianzjh, 2024-06-24:
> 你得问你妈啊，你家的传统，不孕不育儿孙满堂哈哈哈

别把你家狗东西的传统给别人，我要跟你说，人和狗不一样，人狗殊途

AXBNG, 2024-06-24:
> 别把你家狗东西的传统给别人，我要跟你说，人和狗不一样，人狗殊途

那你还在这儿叫唤啥啊哈哈哈

brianzjh, 2024-06-24:
> 我可不操你脸上，嫌恶心

但我得操你脸上啊！就像我家狗操你妈一样

brianzjh, 2024-06-24:
> 那你还在这儿叫唤啥啊哈哈哈

你没看我操你很快乐吗？？

AXBNG, 2024-06-24:
> 但我得操你脸上啊！就像我家狗操你妈一样

看来你家狗天天操你妈，给你看应激了都笑死

brianzjh, 2024-06-24:
> 那你还在这儿叫唤啥啊哈哈哈

18线井里的傻狗，快回家看看你妈吧！被全城的狗操着呢

AXBNG, 2024-06-24:
> 你没看我操你很快乐吗？？

你妈说你都硬不起来，操谁啊天天打嘴炮

brianzjh, 2024-06-24:
> 看来你家狗天天操你妈，给你看应激了都笑死

看来你这狗不识字啊！我家狗操你妈，认识吗？？

AXBNG, 2024-06-24:
> 18线井里的傻狗，快回家看看你妈吧！被全城的狗操着呢

话都说不利索的傻逼哈哈哈

brianzjh, 2024-06-24:
> 你妈说你都硬不起来，操谁啊天天打嘴炮

人类的话，你听不懂，就别学了！人家说的是，狗操你妈妈

[124 more replies]

closed 2024-06-24 by coolsnowwolf