coolsnowwolf/lede #12269: 执行 make menuconfig 控制台报错

bob-zebedy, 2024-06-24:
### 详细叙述

最新代码，控制台会出现以下 error
```shell
$ make menuconfig                                                                                                                                                                                                                                                          
tmp/.config-package.in:78758:error: recursive dependency detected!
tmp/.config-package.in:78758:	symbol PACKAGE_iptables is selected by PACKAGE_luci-app-ssr-plus
tmp/.config-package.in:62219:	symbol PACKAGE_luci-app-ssr-plus depends on PACKAGE_libustream-mbedtls
tmp/.config-package.in:59485:	symbol PACKAGE_libustream-mbedtls depends on PACKAGE_libustream-openssl
tmp/.config-package.in:59498:	symbol PACKAGE_libustream-openssl is selected by PACKAGE_luci-ssl-openssl
tmp/.config-package.in:60540:	symbol PACKAGE_luci-ssl-openssl depends on PACKAGE_iptables
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"
```

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

无

coolsnowwolf, 2024-06-24:
删除 feeds 文件夹，再来一次

bob-zebedy, 2024-06-24:
> 删除 feeds 文件夹，再来一次

尝试过，整个目录都删了

WYC-2020, 2024-06-25:
这是递归引用了

bob-zebedy, 2024-06-25:
> 这是递归引用了

我知道。
问题就是 9728c59e9b2a2d8d4df5b073f256430fa85b31f7 这个提交引起了这个问题，怎么解决。

JarodWan, 2024-07-06:
拿掉 luci-app-ssr-plus 就好了

kissyouhunter, 2024-07-06:
不选中 luci-app-ssr-plus  make menu的时候也有同样报错

JarodWan, 2024-07-08:
前天本地编译过一次， 没有报错，使用的是Passwall

JarodWan, 2024-07-08:
刚刚重新做了一次 没有报错， 用了最新的 luci-app-ssr-plus， 也没有问题

closed 2024-07-09 by bob-zebedy