coolsnowwolf/lede #12270: cloudflare 构建失败

caixin1218, 2024-06-24:
### 详细叙述

   <*> cloudflared. Cloudflare Tunnel client

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

SHELL= flock /ubuntu/peaceful/tmp/.root-copy.flock -c 'cp -fpR /ubuntu/peaceful/build_dir/target-x86_64_musl/arp-scan-1.9.8/.pkgdir/arp-scan-database/. /ubuntu/peaceful/staging_dir/target-x86_64_musl/root-x86/'
touch /ubuntu/peaceful/staging_dir/target-x86_64_musl/root-x86/stamp/.arp-scan-database_installed
echo "arp-scan-database" >> /ubuntu/peaceful/staging_dir/target-x86_64_musl/pkginfo/arp-scan.default.install
make[3]: Leaving directory '/ubuntu/peaceful/feeds/packages/net/arp-scan'
time: package/feeds/packages/arp-scan/compile#0.30#0.10#0.38
make[3]: Entering directory '/ubuntu/peaceful/feeds/packages/utils/bc'
mkdir -p /ubuntu/peaceful/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /ubuntu/peaceful/tmp/.root-copy.flock -c 'cp -fpR /ubuntu/peaceful/build_dir/target-x86_64_musl/bc-1.06.95/.pkgdir/bc/. /ubuntu/peaceful/staging_dir/target-x86_64_musl/root-x86/'
touch /ubuntu/peaceful/staging_dir/target-x86_64_musl/root-x86/stamp/.bc_installed
echo "bc" >> /ubuntu/peaceful/staging_dir/target-x86_64_musl/pkginfo/bc.default.install
make[3]: Leaving directory '/ubuntu/peaceful/feeds/packages/utils/bc'
time: package/feeds/packages/bc/compile#0.30#0.10#0.37
make[3]: Entering directory '/ubuntu/peaceful/feeds/packages/net/bridge-utils'
mkdir -p /ubuntu/peaceful/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /ubuntu/peaceful/tmp/.root-copy.flock -c 'cp -fpR /ubuntu/peaceful/build_dir/target-x86_64_musl/linux-x86_64/bridge-utils-1.7.1/.pkgdir/bridge/. /ubuntu/peaceful/staging_dir/target-x86_64_musl/root-x86/'
touch /ubuntu/peaceful/staging_dir/target-x86_64_musl/root-x86/stamp/.bridge_installed
echo "bridge" >> /ubuntu/peaceful/staging_dir/target-x86_64_musl/pkginfo/bridge-utils.default.install
make[3]: Leaving directory '/ubuntu/peaceful/feeds/packages/net/bridge-utils'
time: package/feeds/packages/bridge-utils/compile#0.35#0.09#0.40
make[3]: Entering directory '/ubuntu/peaceful/feeds/packages/net/cloudflared'
rm -f /ubuntu/peaceful/build_dir/target-x86_64_musl/cloudflared-2023.2.2/.built
touch /ubuntu/peaceful/build_dir/target-x86_64_musl/cloudflared-2023.2.2/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/ubuntu/peaceful/tmp/go-build" GO_MOD_CACHE_DIR="/ubuntu/peaceful/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/cloudflare/cloudflared" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/cloudflare/cloudflared/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/ubuntu/peaceful/build_dir/target-x86_64_musl/cloudflared-2023.2.2" GO_BUILD_DIR="/ubuntu/peaceful/build_dir/target-x86_64_musl/cloudflared-2023.2.2/.go_work/build" GO_BUILD_BIN_DIR="/ubuntu/peaceful/build_dir/target-x86_64_musl/cloudflared-2023.2.2/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/ubuntu/peaceful/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/ubuntu/peaceful/build_dir/target-x86_64_musl/cloudflared-2023.2.2=cloudflared-2023.2.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/ubuntu/peaceful/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/ubuntu/peaceful/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/ubuntu/peaceful/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/ubuntu/peaceful/build_dir/target-x86_64_musl/cloudflared-2023.2.2=cloudflared-2023.2.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/ubuntu/peaceful/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/ubuntu/peaceful/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/ubuntu/peaceful/build_dir/target-x86_64_musl/cloudflared-2023.2.2/.go_work/build" GOCACHE="/ubuntu/peaceful/tmp/go-build" GOMODCACHE="/ubuntu/peaceful/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1686251287' -linkmode external -extldflags '-L/ubuntu/peaceful/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/ubuntu/peaceful/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X main.Version=2023.2.2 -buildid '1686251287' -linkmode external -extldflags '-L/ubuntu/peaceful/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/ubuntu/peaceful/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets

Building targets
github.com/lucas-clemente/quic-go/internal/qtls
# github.com/lucas-clemente/quic-go/internal/qtls
../../vendor/github.com/lucas-clemente/quic-go/internal/qtls/go120.go:6:13: cannot use "The version of quic-go you're using can't be built on Go 1.20 yet. For more details, please see https://github.com/lucas-clemente/quic-go/wiki/quic-go-and-Go-versions." (untyped string constant "The version of quic-go you're using can't be built on Go 1.20 yet. F...) as int value in variable declaration

make[3]: *** [Makefile:68: /ubuntu/peaceful/build_dir/target-x86_64_musl/cloudflared-2023.2.2/.built] Error 1
make[3]: Leaving directory '/ubuntu/peaceful/feeds/packages/net/cloudflared'
time: package/feeds/packages/cloudflared/compile#2.06#0.97#1.57
    ERROR: package/feeds/packages/cloudflared failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/cloudflared/compile] Error 1
make[2]: Leaving directory '/ubuntu/peaceful'
make[1]: *** [package/Makefile:110: /ubuntu/peaceful/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/ubuntu/peaceful'
make: *** [/ubuntu/peaceful/include/toplevel.mk:231: world] Error 2

TheLastManT1, 2024-06-29:
用op官方的cf替换即可

caixin1218, 2024-06-30:
> 用op官方的cf替换即可

受教，感谢

closed 2024-07-01 by caixin1218