coolsnowwolf/lede #12272: 云编译 UU游戏加速出错

NeedforGit, 2024-06-24:
### 详细叙述

最新云编译，UU游戏加速出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

SHELL= flock /workdir/openwrt/tmp/.uugamebooster-x86_64-v4.14.4.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "uugamebooster-x86_64-v4.14.4.tar.gz" "b9ca050a6ed3658b460150b02f7772e6" "" "https://uu.gdl.netease.com/uuplugin/openwrt-x86_64/v4.14.4/uu.tar.gz?"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://uu.gdl.netease.com/uuplugin/openwrt-x86_64/v4.14.4/uu.tar.gz?/uugamebooster-x86_64-v4.14.4.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
  0     0    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 403 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/uugamebooster-x86_64-v4.14.4.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/uugamebooster-x86_64-v4.14.4.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/uugamebooster-x86_64-v4.14.4.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:86: /workdir/openwrt/dl/uugamebooster-x86_64-v4.14.4.tar.gz] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/uugamebooster'
time: package/feeds/packages/uugamebooster/compile#0.14#0.05#3.37
    ERROR: package/feeds/packages/uugamebooster failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/uugamebooster/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

yanxin152133, 2024-06-26:
网络不行吧

slippersheepig, 2024-06-27:
网易官方下载地址拦截了境外流量下载，大陆流量是可以下载成功的

closed 2024-07-08 by coolsnowwolf

slippersheepig, 2024-07-22:
> > 网易官方下载地址拦截了境外流量下载，大陆流量是可以下载成功的
> 
> 那就是说只要云编译的话，就不行了？只能国内网络本地编译？

现在没有问题了，作者已经更新commit

NeedforGit, 2024-07-23:
> > > 网易官方下载地址拦截了境外流量下载，大陆流量是可以下载成功的
> > 
> > 
> > 那就是说只要云编译的话，就不行了？只能国内网络本地编译？
> 
> 现在没有问题了，作者已经更新commit

还是不行啊，我看版本确实换了，但云编译还是出错：

+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/uugamebooster-x86_64-v6.3.10.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:86: /workdir/openwrt/dl/uugamebooster-x86_64-v6.3.10.tar.gz] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/uugamebooster'
time: package/feeds/packages/uugamebooster/compile#0.16#0.08#7.24
    ERROR: package/feeds/packages/uugamebooster failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/uugamebooster/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.