coolsnowwolf/lede #12273: unbuntu 24.04 node 编译不过去,22.02可以编译

guhill1, 2024-06-24:
### 详细叙述

unbuntu 24.04 node 编译不过去,python2.7依赖安装不了,unbuntu 22.02可以编译
看看能否解决?

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

/build_dir/hostpkg/node-v16.19.1/out/Release/.deps//home/lede/openwrtLean/build_dir/hostpkg/node-v16.19.1/out/Release/obj.target/v8_libbase/deps/v8/src/base/cpu.o.d.raw -I/home/lede/openwrtLean/staging_dir/host/include -I/home/lede/openwrtLean/staging_dir/hostpkg/include -I/home/lede/openwrtLean/staging_dir/target-x86_64_musl/host/include  -c
In file included from ../deps/v8/src/base/macros.h:12,
                 from ../deps/v8/src/base/cpu.h:17,
                 from ../deps/v8/src/base/cpu.cc:5:
../deps/v8/src/base/logging.h: In function 'auto v8::base::detail::GetUnderlyingEnumTypeForPrinting(T)':
../deps/v8/src/base/logging.h:154:26: error: 'uint16_t' was not declared in this scope
  154 |                          uint16_t> >;
      |                          ^~~~~~~~
../deps/v8/src/base/logging.h:17:1: note: 'uint16_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   16 | #include "src/base/template-utils.h"
  +++ |+#include <cstdint>
   17 | 
../deps/v8/src/base/logging.h:154:34: error: template argument 3 is invalid
  154 |                          uint16_t> >;
      |                                  ^
../deps/v8/src/base/logging.h:154:36: error: template argument 3 is invalid
  154 |                          uint16_t> >;
      |                                    ^
../deps/v8/src/base/logging.h:154:37: error: expected identifier before ';' token
  154 |                          uint16_t> >;
      |                                     ^
../deps/v8/src/base/logging.h:155:22: error: 'int_t' does not name a type; did you mean 'wint_t'?
  155 |   return static_cast<int_t>(static_cast<underlying_t>(val));
      |                      ^~~~~
      |                      wint_t
../deps/v8/src/base/macros.h: At global scope:
../deps/v8/src/base/macros.h:314:8: error: 'uint64_t' does not name a type
  314 | inline uint64_t make_uint64(uint32_t high, uint32_t low) {
      |        ^~~~~~~~
../deps/v8/src/base/macros.h:14:1: note: 'uint64_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   13 | #include "src/base/platform/wrappers.h"
  +++ |+#include <cstdint>
   14 | 
../deps/v8/src/base/macros.h:320:25: error: 'intptr_t' has not been declared
  320 | inline T RoundDown(T x, intptr_t m) {
      |                         ^~~~~~~~
../deps/v8/src/base/macros.h:326:11: error: 'intptr_t' has not been declared
  326 | template <intptr_t m, typename T>
      |           ^~~~~~~~
../deps/v8/src/base/macros.h: In function 'constexpr T RoundDown(T)':
../deps/v8/src/base/macros.h:330:17: error: 'm' was not declared in this scope
  330 |   STATIC_ASSERT(m != 0 && ((m & (m - 1)) == 0));
      |                 ^
../deps/v8/src/base/macros.h:182:43: note: in definition of macro 'STATIC_ASSERT'
  182 | #define STATIC_ASSERT(test) static_assert(test, #test)
      |                                           ^~~~
../deps/v8/src/base/macros.h: At global scope:
../deps/v8/src/base/macros.h:336:23: error: 'intptr_t' has not been declared
  336 | inline T RoundUp(T x, intptr_t m) {
      |                       ^~~~~~~~
../deps/v8/src/base/macros.h:340:11: error: 'intptr_t' has not been declared
  340 | template <intptr_t m, typename T>
      |           ^~~~~~~~
../deps/v8/src/base/macros.h: In function 'constexpr T RoundUp(T)':
../deps/v8/src/base/macros.h:343:20: error: 'm' was not declared in this scope
  343 |   return RoundDown<m, T>(static_cast<T>(x + (m - 1)));
      |                    ^
../deps/v8/src/base/macros.h: In function 'void* AlignedAddress(void*, size_t)':
../deps/v8/src/base/macros.h:354:51: error: 'uintptr_t' does not name a type
  354 |   return reinterpret_cast<void*>(reinterpret_cast<uintptr_t>(address) &
      |                                                   ^~~~~~~~~
../deps/v8/src/base/macros.h:354:51: note: 'uintptr_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
../deps/v8/src/base/macros.h:355:47: error: 'uintptr_t' does not name a type
  355 |                                  ~static_cast<uintptr_t>(alignment - 1));
      |                                               ^~~~~~~~~
../deps/v8/src/base/macros.h:355:47: note: 'uintptr_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
make[5]: *** [tools/v8_gypfiles/v8_libbase.target.mk:179: /home/lede/openwrtLean/build_dir/hostpkg/node-v16.19.1/out/Release/obj.target/v8_libbase/deps/v8/src/base/cpu.o] Error 1
rm 16d478c1cffc90ec8edcb20b5a487f4d908455d7.intermediate
make[4]: *** [Makefile:113: node] Error 2
make[4]: Leaving directory '/home/lede/openwrtLean/build_dir/hostpkg/node-v16.19.1'
make[3]: *** [Makefile:233: /home/lede/openwrtLean/build_dir/hostpkg/node-v16.19.1/.built] Error 2
make[3]: Leaving directory '/home/lede/openwrtLean/feeds/packages/lang/node'
time: package/feeds/packages/node/host-compile#0.60#0.62#1.19
    ERROR: package/feeds/packages/node [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/node/host/compile] Error 1
make[2]: Leaving directory '/home/lede/openwrtLean'
make[1]: *** [package/Makefile:110: /home/lede/openwrtLean/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lede/openwrtLean'
make: *** [/home/lede/openwrtLean/include/toplevel.mk:231: world] Error 2

GreatMichaelLee, 2024-06-24:
我的也是24.04，没问题

[image]
[image]

guhill1, 2024-06-24:
> 我的也是24.04，没问题，你用的node 16 都EOL了，选18或者20
> 
> ![image](https://private-user-images.githubusercontent.com/20067401/342389763-3cf117a1-e7a1-4448-b1ad-bbfb14db67ed.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyNTE3MzEsIm5iZiI6MTcxOTI1MTQzMSwicGF0aCI6Ii8yMDA2NzQwMS8zNDIzODk3NjMtM2NmMTE3YTEtZTdhMS00NDQ4LWIxYWQtYmJmYjE0ZGI2N2VkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI0VDE3NTAzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVlMTllODhlNWJiMmIwZWY0Yzg0MGJjYmE2Y2M4MWNkOGJkZDBmOGE3ZjllYzliMTQ4ZTMzZjUzOGI3MDg2NDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lboYJmPBuKSWHaCWg34AX-7u-aj0ml5R80xLaMGx0V4) ![image](https://private-user-images.githubusercontent.com/20067401/342389877-bdbf07b1-3656-495f-960f-f72b15ec4ad2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyNTE3MzEsIm5iZiI6MTcxOTI1MTQzMSwicGF0aCI6Ii8yMDA2NzQwMS8zNDIzODk4NzctYmRiZjA3YjEtMzY1Ni00OTVmLTk2MGYtZjcyYjE1ZWM0YWQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI0VDE3NTAzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIxZTRmYTgyODJjZDk1ZWM2ODkzMjI2MzY0M2VmZDcyMGNmZjgwMWQ1ZmIwNWNhY2UyZDBmNjM2ZDRlYzQ2ZGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.wTeNhmOZ654DgUZ9FLEKwhTXHFrn1i527SX0oU_5OAI)

好的,非常感谢,我试试

guhill1, 2024-06-24:
> language\node.js把node版本选高点试试

好的,非常感谢,我试试

closed 2024-06-24 by guhill1