coolsnowwolf/lede #12274: softetherVPN系统占用异常

psonys, 2024-06-24:
### 详细叙述

[image]
[image]
自从softetherVPN更新后，不定时的softether会导致CPU占用异常，远程连接失效。
系统使用6.6内核。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

畅网1235U

### 详细日志

编译正常

DragonOneY, 2024-06-25:
这是softethervpn的问题，回滚到5.01.9674-1就好了。
cd lede
git checkout master
git pull
git checkout -b rollback-softethervpn-5.01.9674-1
cd feeds/packages/net/softethervpn5
git checkout 7d89c09430930e3bf9b487333f0a18dd117f46af

psonys, 2024-06-25:
Debian:~/lede/feeds/packages/net/softethervpn5$ git checkout 7d89c09430930e3bf9b487333f0a18dd117f46af
fatal: 引用不是一个树：7d89c09430930e3bf9b487333f0a18dd117f46af

625045508, 2024-07-04:
这鬼问题，煎熬我好几天，最后也是降版本编译好了

closed 2024-07-07 by coolsnowwolf