coolsnowwolf/lede #12275: 编译quectel_QMI_WWAN 报错

smallprogram, 2024-06-24:
### 详细叙述

本地编译及云编译quectel_QMI_WWAN 报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86,ARM

### 详细日志

```log
Applying ./patches/175-reduce-dynamic-syms.patch using plaintext: 
Setting license tag...
rm -f /home/zhusir/lean_X86.config/build_dir/target-x86_64_musl/openssl-3.0.14/.configured_*
patching file tc/Makefile

Applying ./patches/180-drop_FAILED_POLICY.patch using plaintext: 
rm -f /home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/stamp/.openssl_installed
gcc -shared -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -fno-semantic-interposition -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -O2 -I/home/zhusir/lean_X86.config/staging_dir/host/include -I/home/zhusir/lean_X86.config/staging_dir/hostpkg/include -I/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/include -I/home/zhusir/lean_X86.config/staging_dir/host/include -I/home/zhusir/lean_X86.config/staging_dir/hostpkg/include -I/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/include build/temp.linux-x86_64-3.10/home/zhusir/lean_X86.config/build_dir/hostpkg/Python-3.10.7/Modules/cjkcodecs/_codecs_hk.o -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -L/usr/lib/x86_64-linux-gnu -L/usr/local/lib -o build/lib.linux-x86_64-3.10/_codecs_hk.cpython-310.so
yes
(cd /home/zhusir/lean_X86.config/build_dir/target-x86_64_musl/openssl-3.0.14; ./Configure linux-x86_64-openwrt --prefix=/usr --libdir=lib --openssldir=/etc/ssl --cross-compile-prefix="x86_64-openwrt-linux-musl-" -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/zhusir/lean_X86.config/build_dir/target-x86_64_musl/openssl-3.0.14=openssl-3.0.14 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -O3 -DPIC -fpic -I/home/zhusir/lean_X86.config/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/zhusir/lean_X86.config/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/zhusir/lean_X86.config/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -L/home/zhusir/lean_X86.config/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/zhusir/lean_X86.config/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,--gc-sections -znow -zrelro shared no-tests no-blake2 no-async no-ec2m no-aria no-sm2 no-sm3 no-sm4 no-camellia no-idea no-seed no-mdc2 no-whirlpool no-rfc3779 disable-dynamic-engine enable-devcryptoeng no-padlockeng no-dtls no-comp no-nextprotoneg && { [ -f /home/zhusir/lean_X86.config/build_dir/target-x86_64_musl/openssl-3.0.14/.configured_880b4dda6884ae76f88cc86ffad357b6_01041e4c82e8d3cb13f49cbd28d782fb ] || make clean; } )
gcc -shared -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -fno-semantic-interposition -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -O2 -I/home/zhusir/lean_X86.config/staging_dir/host/include -I/home/zhusir/lean_X86.config/staging_dir/hostpkg/include -I/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/include -I/home/zhusir/lean_X86.config/staging_dir/host/include -I/home/zhusir/lean_X86.config/staging_dir/hostpkg/include -I/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/include build/temp.linux-x86_64-3.10/home/zhusir/lean_X86.config/build_dir/hostpkg/Python-3.10.7/Modules/cjkcodecs/_codecs_jp.o -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -L/usr/lib/x86_64-linux-gnu -L/usr/local/lib -o build/lib.linux-x86_64-3.10/_codecs_jp.cpython-310.so
gcc -shared -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -fno-semantic-interposition -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -O2 -I/home/zhusir/lean_X86.config/staging_dir/host/include -I/home/zhusir/lean_X86.config/staging_dir/hostpkg/include -I/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/include -I/home/zhusir/lean_X86.config/staging_dir/host/include -I/home/zhusir/lean_X86.config/staging_dir/hostpkg/include -I/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/include build/temp.linux-x86_64-3.10/home/zhusir/lean_X86.config/build_dir/hostpkg/Python-3.10.7/Modules/_asynciomodule.o -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -L/usr/lib/x86_64-linux-gnu -L/usr/local/lib -o build/lib.linux-x86_64-3.10/_asyncio.cpython-310.so
gcc -shared -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -fno-semantic-interposition -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -Wl,-rpath,/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -Wl,--no-as-needed -lrt -O2 -I/home/zhusir/lean_X86.config/staging_dir/host/include -I/home/zhusir/lean_X86.config/staging_dir/hostpkg/include -I/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/include -I/home/zhusir/lean_X86.config/staging_dir/host/include -I/home/zhusir/lean_X86.config/staging_dir/hostpkg/include -I/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/include build/temp.linux-x86_64-3.10/home/zhusir/lean_X86.config/build_dir/hostpkg/Python-3.10.7/Modules/sha512module.o -L/home/zhusir/lean_X86.config/staging_dir/host/lib -L/home/zhusir/lean_X86.config/staging_dir/hostpkg/lib -L/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/host/lib -L/usr/lib/x86_64-linux-gnu -L/usr/local/lib -o build/lib.linux-x86_64-3.10/_sha512.cpython-310.so
patching file ip/rtm_map.c
[... cut ...] 6.config/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/zhusir/lean_X86.config/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/zhusir/lean_X86.config/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include -DHAVE_UBUS -DHAVE_POLL_H   -DNO_DHCP6  -DNO_AUTH  -DHAVE_NFTSET       -DVERSION='"2.88"'          -I/home/zhusir/lean_X86.config/staging_dir/target-x86_64_musl/usr/include    -c forward.c	
    ERROR: package/wwan/driver/quectel_QMI_WWAN failed to build.
```

hzjnet, 2024-06-24:
我还以为只有5.10内核才有这问题喃？！

WYC-2020, 2024-06-25:
这个我记得我发过一个

hzjnet, 2024-06-26:
> 这个我记得我发过一个

这个吗：https://github.com/coolsnowwolf/lede/issues/11374

mumuou, 2024-06-26:
quectel_QMI_WWAN: add linux 6.6 kernel support 这个更新导致的，回滚这个更新就好了。

hzjnet, 2024-06-28:
> quectel_QMI_WWAN: add linux 6.6 kernel support 这个更新导致的，回滚这个更新就好了。

正解 必须赞一个 ；同时WYC-2020的872-export-some-functions-of-the-sched-module.patch补丁也要到位才能正常编译quectel_srpd_pcie；

感谢2位

ChaingTsung, 2024-07-01:
```
2024-06-28T09:53:57.6592110Z make[3]: Entering directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/utils/f2fs-tools'
2024-06-28T09:53:57.8010787Z echo "libf2fs" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/f2fs-tools.default.install
2024-06-28T09:53:57.8027274Z echo "mkf2fs" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/f2fs-tools.default.install
2024-06-28T09:53:57.8042346Z make[3]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/utils/f2fs-tools'
2024-06-28T09:53:57.8048359Z time: package/utils/f2fs-tools/default/compile#0.12#0.04#0.15
2024-06-28T09:53:57.8161799Z make[3]: Entering directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/utils/usbmode'
2024-06-28T09:53:57.9741325Z mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2024-06-28T09:53:57.9761641Z SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/usbmode-2022-02-24-3c8595a4/.pkgdir/usb-modeswitch/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2024-06-28T09:53:57.9804371Z touch /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.usb-modeswitch_installed
2024-06-28T09:53:57.9861461Z echo "usb-modeswitch" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/usbmode.default.install
2024-06-28T09:53:57.9875531Z make[3]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/utils/usbmode'
2024-06-28T09:53:57.9880264Z time: package/utils/usbmode/compile#0.14#0.05#0.18
2024-06-28T09:53:57.9983221Z make[3]: Entering directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/wwan/app/luci-app-pcimodem'
2024-06-28T09:53:58.2790525Z mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2024-06-28T09:53:58.2815416Z SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-pcimodem/.pkgdir/luci-app-pcimodem/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2024-06-28T09:53:58.2858739Z touch /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-app-pcimodem_installed
2024-06-28T09:53:58.2926144Z echo "luci-app-pcimodem" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/luci-app-pcimodem.default.install
2024-06-28T09:53:58.2943345Z mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2024-06-28T09:53:58.2966702Z SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/luci-app-pcimodem/.pkgdir/luci-i18n-pcimodem-zh-cn/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2024-06-28T09:53:58.3000310Z touch /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-i18n-pcimodem-zh-cn_installed
2024-06-28T09:53:58.3018785Z echo "luci-i18n-pcimodem-zh-cn" >> /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/luci-app-pcimodem.default.install
2024-06-28T09:53:58.3042124Z make[3]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/wwan/app/luci-app-pcimodem'
2024-06-28T09:53:58.3051650Z time: package/wwan/app/luci-app-pcimodem/compile#0.24#0.09#0.31
2024-06-28T09:53:58.3156013Z make[3]: Entering directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/package/wwan/driver/fibocom_QMI_WWAN'
2024-06-28T09:53:58.4538677Z mkdir -p /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2024-06-28T09:53:58.4572644Z SHELL= flock /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/.pkgdir/kmod-qmi_wwan_f/. /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
[... cut ...]
2024-06-28T09:53:59.4671781Z make[1]: *** [package/Makefile:110: /home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-06-28T09:53:59.4693893Z make[1]: Leaving directory '/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt'
2024-06-28T09:53:59.4724561Z make: *** [/home/yoki/actions-runner/_work/Actions-OpenWrt/Actions-OpenWrt/workdir/openwrt/include/toplevel.mk:231: world] Error 2
```
同样的问题

closed 2024-07-04 by coolsnowwolf