coolsnowwolf/lede #12276: 6.1和6.6内核都缺少依赖

Turing-China, 2024-06-25:
### 详细叙述

WARNING: Makefile 'package/feeds/telephony/asterisk/Makefile' has a dependency on 'libjwt', which does not exist
#
# No change to .config
#
WARNING: Makefile 'package/feeds/telephony/asterisk/Makefile' has a dependency on 'libjwt', which does not exist
WARNING: Makefile 'package/feeds/telephony/asterisk/Makefile' has a dependency on 'libjwt', which does not exist
make[2]: Entering directory '/Ledemaster/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/Ledemaster/openwrt/scripts/config'
time: target/linux/prereq#0.08#0.06#0.13
 make[1] tools/flock/compile
 make[2] -C tools/flock compile
 make[1] tools/download
 make[2] -C tools/automake download
 make[2] -C tools/autoconf-archive download
 make[2] -C tools/autoconf download
 make[2] -C tools/bison download
 make[2] -C tools/cmake download
 make[2] -C tools/bc download
 make[2] -C tools/cpio download
 make[2] -C tools/dosfstools download
 make[2] -C tools/e2fsprogs download

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

[image]
[image]
[image]

wlshdjj, 2024-06-25:
asterisk这东西你用不着的话管你都不用管

tonyzzt001, 2024-06-25:
同一问题。

closed 2024-06-25 by coolsnowwolf