coolsnowwolf/lede #12281: 亚瑟 ax1800 pro 编译固件无法保存配置，切只能刷recovery固件

chris7395328, 2024-06-26:
### 详细叙述

之前都是用qwrt，分区表和uboot也是大雕的，看到开源了想编译试试，uboot那里刷factory固件是无法开机，recovery可以开机但是无法保存配置，不知道是不是分区表的问题？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

jdc ax1800 pro

### 详细日志

无

coolsnowwolf, 2024-06-27:
你估计用的是第三方的 uboot ，分区太小了

正确的 uboot:

https://mbd.pub/o/bread/ZJeXk59v

而且本来 uboot 就是刷 recovery 的

chris7395328, 2024-06-27:
> 你估计用的是第三方的 uboot ，分区太小了
> 
> 正确的 uboot:
> 
> https://mbd.pub/o/bread/ZJeXk59v
> 
> 而且本来 uboot 就是刷 recovery 的

谢谢，那我重新刷一次uboot.之前的uboot也是你面包多的那个，可能不是最新的

szhpf, 2024-06-27:
这个版本，重启一次，配置就无法保存成功，R2S软路由，测试好多次，都无法成功保存配置

szhpf, 2024-06-27:
> 这个版本，重启一次，配置就无法保存成功，R2S软路由，测试好多次，都无法成功保存配置

重编译好多次了，每次问题都是一样，无法保存配置，而切安装插架也不行

closed 2024-06-27 by chris7395328

chris7395328, 2024-06-30:
> 你估计用的是第三方的 uboot ，分区太小了
> 
> 正确的 uboot:
> 
> https://mbd.pub/o/bread/ZJeXk59v
> 
> 而且本来 uboot 就是刷 recovery 的

大雕，换了uboot刷编译固件可以保存设置正常使用，但是升级sysupgrade选择保留配置升级是无效诶，还是初始化，只能再导入一次配置。还有挂载点那里的默认设置是挂载什么分区的？我那显示了不存在id