coolsnowwolf/lede #12283: k2t报这个错误，请大神指点

xuexijiaoben, 2024-06-28:
### 详细叙述

编译k2t，报这个 cryptodev-linux  错误，
求大神指点

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

k2t

### 详细日志

make[3]: Entering directory '/workdir/openwrt/package/kernel/cryptodev-linux'
rm -f /workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13/.built
touch /workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13/.built_check
make -C /workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-mips_24kc_musl=target-mips_24kc_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-mips_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Jun 27 07:57:09 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/fullconenat.symvers /workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/symvers/gpio-button-hotplug.symvers" KERNELRELEASE=5.10.220 KERNEL_DIR="/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.10.220"
make[4]: Entering directory '/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13'
make -C /workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.10.220 M=/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13 ARCH=mips CROSS_COMPILE=mips-openwrt-linux-musl- modules
make[5]: Entering directory '/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.10.220'
  CC [M]  /workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13/ioctl.o
/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13/ioctl.c: In function 'cryptodev_ioctl':
/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13/ioctl.c:937:4: error: implicit declaration of function 'ksys_close'; did you mean 'sys_close'? [-Werror=implicit-function-declaration]
    ksys_close(fd);
    ^~~~~~~~~~
    sys_close
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:286: /workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13/ioctl.o] Error 1
make[5]: *** [Makefile:1832: /workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.10.220'
make[4]: *** [Makefile:30: build] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13'
make[3]: *** [Makefile:61: /workdir/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/cryptodev-linux-cryptodev-linux-1.13/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/kernel/cryptodev-linux'
time: package/kernel/cryptodev-linux/compile#0.95#0.28#1.12
    ERROR: package/kernel/cryptodev-linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/cryptodev-linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

dragon888888888888, 2024-06-28:
make[4]: Entering directory '/home/abc/le/target/linux/rockchip'
rm -f /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/localversion
/home/abc/le/scripts/kconfig.pl  + /home/abc/le/target/linux/generic/config-5.15 /home/abc/le/target/linux/rockchip/armv8/config-5.15 > /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/abc/le/.config >> /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.target
/home/abc/le/scripts/package-metadata.pl kconfig /home/abc/le/tmp/.packageinfo /home/abc/le/.config 5.15 > /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.override
/home/abc/le/scripts/kconfig.pl 'm+' '+' /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.target /dev/null /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.override > /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set
mv /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.old
grep -v INITRAMFS /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.old > /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set
echo "# CONFIG_INITRAMFS_PRESERVE_MTIME is not set" >> /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set
rm -rf /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/modules
cmp -s /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.prev || { cp /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config; cp /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.prev; }
export MAKEFLAGS= ; [ -d /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/user_headers ] || make -C /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161 KCFLAGS="-fmacro-prefix-map=/home/abc/le/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/abc/le/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Jun 27 07:57:09 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/abc/le/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.161 CC="aarch64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/user_headers headers_install
grep '=[ym]' /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.config.set | LC_ALL=C sort | /home/abc/le/staging_dir/host/bin/mkhash md5 > /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.vermagic
touch /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.configured
rm -f /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/vmlinux /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/System.map
make -C /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161 KCFLAGS="-fmacro-prefix-map=/home/abc/le/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/abc/le/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Jun 27 07:57:09 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/abc/le/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.161 CC="aarch64-openwrt-linux-musl-gcc" olddefconfig
[... cut ...]
make[4]: *** [Makefile:26: /home/abc/le/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.161/.modules] Error 2
make[4]: Leaving directory '/home/abc/le/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/abc/le/target/linux'
time: target/linux/compile#1.93#0.63#2.25
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/abc/le'
make[1]: *** [target/Makefile:23: /home/abc/le/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
5.1内核无法编译成功而已  换6.6正常

hzjnet, 2024-06-29:
> ### 详细叙述
> 
> 编译k2t，报这个 cryptodev-linux 错误， 求大神指点
> ### 重复 issue
> 
>     * [x]  没有类似的 issue

+1 
5.10.220确实和ryptodev-linux不兼容，需要补丁

closed 2024-07-01 by coolsnowwolf