coolsnowwolf/lede #12287: 【已升级g++】failed to parse bitcode for LTO module

Deerczl, 2024-06-30:
### 详细叙述

参考[issue11800](https://github.com/coolsnowwolf/lede/issues/11800)和[issue12013](https://github.com/coolsnowwolf/lede/issues/12013)，已经将gcc和g++升级到13.1版本，但编译过程依旧出现报错，
error: failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM18.1.2-rust-1.78.0-stable' Reader: 'LLVM 18.1.2-rust-1.78.0-stable')
请问应该如何解决呢，还是gcc/g++的版本问题吗？先行致谢~

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

OPENWRT

### 详细日志

    Finished release [optimized] target(s) in 0.20s
Building stage1 library artifacts (x86_64-unknown-linux-gnu)
   Compiling core v0.0.0 (/home/reval/lede/build_dir/target-aarch64_generic_musl/host/rustc-1.78.0-src/library/core)
   Compiling rustc-std-workspace-core v1.99.0 (/home/reval/lede/build_dir/target-aarch64_generic_musl/host/rustc-1.78.0-src/library/rustc-std-workspace-core)
   Compiling compiler_builtins v0.1.108
   Compiling libc v0.2.153
   Compiling alloc v0.0.0 (/home/reval/lede/build_dir/target-aarch64_generic_musl/host/rustc-1.78.0-src/library/alloc)
   Compiling cfg-if v1.0.0
   Compiling unwind v0.0.0 (/home/reval/lede/build_dir/target-aarch64_generic_musl/host/rustc-1.78.0-src/library/unwind)
   Compiling adler v1.0.2
   Compiling memchr v2.5.0
error: failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM18.1.2-rust-1.78.0-stable' Reader: 'LLVM 18.1.2-rust-1.78.0-stable')

error: could not compile `core` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: could not compile `memchr` (lib) due to 1 previous error
error: could not compile `alloc` (lib) due to 1 previous error
Build completed unsuccessfully in 0:00:31
Makefile:109: recipe for target '/home/reval/lede/build_dir/target-aarch64_generic_musl/host/rustc-1.78.0-src/.built' failed
make[3]: *** [/home/reval/lede/build_dir/target-aarch64_generic_musl/host/rustc-1.78.0-src/.built] Error 1
make[3]: Leaving directory '/home/reval/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#31.70#2.41#31.85
    ERROR: package/feeds/packages/rust [host] failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/rust/host/compile' failed
make[2]: *** [package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/home/reval/lede'
package/Makefile:110: recipe for target '/home/reval/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/reval/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/reval/lede'
/home/reval/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
reval@DESKTOP-0G9RUP8:~/lede$ gcc --version
gcc (Ubuntu 13.1.0-8ubuntu1~18.04) 13.1.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

reval@DESKTOP-0G9RUP8:~/lede$ g++ --version
g++ (Ubuntu 13.1.0-8ubuntu1~18.04) 13.1.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

yanxin152133, 2024-07-01:
看看内存
https://github.com/coolsnowwolf/lede/discussions/12251

Deerczl, 2024-07-01:
谢谢~我检查一下。

Deerczl, 2024-07-01:
[image]
同时用free -h和任务管理器查看内存占用情况，似乎内存还有富余。是WSL的问题吗？

guiys, 2024-07-02:
同样的错误。提供我的信息：
编译openwrt 4.14 x86+fw876插件过程中遇到

Deerczl, 2024-07-03:
已成功编译，确实是内存相关的问题。把解决的步骤记录一下，不一定对。
1. 在Windows的C:/Users/(用户名)/  目录下新建一文本文档，文件名为.wslconfig。内容如下。
[wsl2]
memory={SET}GB
【注】{SET}为所设置的WSL可使用的物理内存大小，不建议与实际物理内存大小相同，会崩。
2. WSL内执行make distclean，清除之前编译的内容，排除干扰
3. 在WSL和Windows内适当增大虚拟内存（swap）
4. 重新配置.config内容，尝试编译。
实际看到的内存占用去到20G，WSL单cache就有10G了。最好留有余量。
[image]
[image]

closed 2024-07-03 by Deerczl