coolsnowwolf/lede #12292: 软路由X86_64,ubuntu 换各种版本版本，编译lede5.10.220总是出错

Cakewalk598, 2024-07-01:
### 详细叙述

cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:286: /home/cwpa/lede/build_dir/target-x86_6                                                                                                                                                             4_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.13/ioctl.o] Error 1
make[5]: *** [Makefile:1832: /home/cwpa/lede/build_dir/target-x86_64_musl/linux-                                                                                                                                                             x86_64/cryptodev-linux-cryptodev-linux-1.13] Error 2
make[5]: Leaving directory '/home/cwpa/lede/build_dir/target-x86_64_musl/linux-x                                                                                                                                                             86_64/linux-5.10.220'
make[4]: *** [Makefile:30: build] Error 2
make[4]: Leaving directory '/home/cwpa/lede/build_dir/target-x86_64_musl/linux-x                                                                                                                                                             86_64/cryptodev-linux-cryptodev-linux-1.13'
make[3]: *** [Makefile:61: /home/cwpa/lede/build_dir/target-x86_64_musl/linux-x8                                                                                                                                                             6_64/cryptodev-linux-cryptodev-linux-1.13/.built] Error 2
make[3]: Leaving directory '/home/cwpa/lede/package/kernel/cryptodev-linux'
time: package/kernel/cryptodev-linux/compile#1.64#0.48#2.13
    ERROR: package/kernel/cryptodev-linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/cryptodev-linux/compile] Erro                                                                                                                                                             r 1
make[2]: Leaving directory '/home/cwpa/lede'
make[1]: *** [package/Makefile:110: /home/cwpa/lede/staging_dir/target-x86_64_mu                                                                                                                                                             sl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/cwpa/lede'
make: *** [/home/cwpa/lede/include/toplevel.mk:231: world] Error 2
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

软路由N5101

### 详细日志

cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:286: /home/cwpa/lede/build_dir/target-x86_6                                                                                                                                                             4_musl/linux-x86_64/cryptodev-linux-cryptodev-linux-1.13/ioctl.o] Error 1
make[5]: *** [Makefile:1832: /home/cwpa/lede/build_dir/target-x86_64_musl/linux-                                                                                                                                                             x86_64/cryptodev-linux-cryptodev-linux-1.13] Error 2
make[5]: Leaving directory '/home/cwpa/lede/build_dir/target-x86_64_musl/linux-x                                                                                                                                                             86_64/linux-5.10.220'
make[4]: *** [Makefile:30: build] Error 2
make[4]: Leaving directory '/home/cwpa/lede/build_dir/target-x86_64_musl/linux-x                                                                                                                                                             86_64/cryptodev-linux-cryptodev-linux-1.13'
make[3]: *** [Makefile:61: /home/cwpa/lede/build_dir/target-x86_64_musl/linux-x8                                                                                                                                                             6_64/cryptodev-linux-cryptodev-linux-1.13/.built] Error 2
make[3]: Leaving directory '/home/cwpa/lede/package/kernel/cryptodev-linux'
time: package/kernel/cryptodev-linux/compile#1.64#0.48#2.13
    ERROR: package/kernel/cryptodev-linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/cryptodev-linux/compile] Erro                                                                                                                                                             r 1
make[2]: Leaving directory '/home/cwpa/lede'
make[1]: *** [package/Makefile:110: /home/cwpa/lede/staging_dir/target-x86_64_mu                                                                                                                                                             sl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/cwpa/lede'
make: *** [/home/cwpa/lede/include/toplevel.mk:231: world] Error 2

Cakewalk598, 2024-07-01:
软路由N5105,ubuntu22.04.4

wuzhuang123, 2024-07-01:
#12283 前边有提过了，都是这个ksys_close报错，220的一个组件不兼容

Cakewalk598, 2024-07-01:
> #12283 前边有提过了，都是这个ksys_close报错，220的一个组件不兼容

难怪什么版本的Ubuntu都报错，我编译5.4看看，谢谢你的解答。

coolsnowwolf, 2024-07-01:
已修复

closed 2024-07-02 by coolsnowwolf