coolsnowwolf/lede #12293: 6.1和6.6内核频繁崩溃

Turing-China, 2024-07-02:
### 详细叙述

内核版本: 6.6.35;1
6.1的内核也一样
编译插件:
luci-app-argon-config
luci-app-firewall
luci-app-ssr-plus
luci-app-vlmcsd
luci-app-zerotier
luci-theme-argon
open-vm-tools

去掉了IPV6和samba

使用ssr崩溃最为严重 基本不可用 分钟级崩溃 编译openclash好像不会崩溃,passwall 崩溃没有ssr 崩溃频繁

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

[image]

jwkinko, 2024-07-02:
6.1 6.6都试过会重启，5.15的时候用ssrp也会不定时重启，但是passwall不会，不知道6.1 6.6重启跟passwall有没有关系

Turing-China, 2024-07-02:
[    0.000000] Linux version 6.6.35 (runner@fv-az1499-971) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6722-d7126b691) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT_DYNAMIC Sun Jun 30 20:48:30 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=41f11515-e63a-76b5-eddd-3ea2bf5a7f02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd swiotlb=65536
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000001000-0x0000000000001fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000002000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000e04bfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000e04c000-0x000000000e04cfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000e04d000-0x000000000fb6efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000fb6f000-0x000000000fbdefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000fbdf000-0x000000000fbfafff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000000fbfb000-0x000000000fbfefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000fbff000-0x00000000bfffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000023fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] e820: update [mem 0x0d0c5018-0x0d0cd057] usable ==> usable
[    0.000000] e820: update [mem 0x0d0c5018-0x0d0cd057] usable ==> usable
[    0.000000] e820: update [mem 0x0db2b018-0x0db2d057] usable ==> usable
[    0.000000] e820: update [mem 0x0db2b018-0x0db2d057] usable ==> usable
[    0.000000] e820: update [mem 0x0db29018-0x0db2a857] usable ==> usable
[    0.000000] e820: update [mem 0x0db29018-0x0db2a857] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000000fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000000001000-0x0000000000001fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000002000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000c0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000000d0c5017] usable
[    0.000000] reserve setup_data: [mem 0x000000000d0c5018-0x000000000d0cd057] usable
[    0.000000] reserve setup_data: [mem 0x000000000d0cd058-0x000000000db29017] usable
[    0.000000] reserve setup_data: [mem 0x000000000db29018-0x000000000db2a857] usable
[    0.000000] reserve setup_data: [mem 0x000000000db2a858-0x000000000db2b017] usable
[    0.000000] reserve setup_data: [mem 0x000000000db2b018-0x000000000db2d057] usable
[    0.000000] reserve setup_data: [mem 0x000000000db2d058-0x000000000e04bfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000e04c000-0x000000000e04cfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000000e04d000-0x000000000fb6efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000fb6f000-0x000000000fbdefff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000fbdf000-0x000000000fbfafff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000000fbfb000-0x000000000fbfefff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000000fbff000-0x00000000bfffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000ffc00000-0x00000000ffc29fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x000000023fffffff] usable
[    0.000000] efi: EFI v2.7 by VMware, Inc.
[    0.000000] efi: SMBIOS=0xfbfb000 MEMATTR=0xf8c0018 ACPI 2.0=0xfbf1000 RNG=0xfbf0018 
[    0.000000] random: crng init done
[    0.000000] efi: Not removing mem34: MMIO range=[0xffc00000-0xffc29fff] (168KB) from e820 map
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware20,1/440BX Desktop Reference Platform, BIOS VMW201.00V.21805430.B64.2305221830 05/22/2023
[    0.000000] vmware: hypercall mode: 0x02
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 2095.077 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000000] vmware: using clock offset of 1897138222 ns
[    0.000016] tsc: Detected 2095.077 MHz processor
[    0.052788] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.052791] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.052799] last_pfn = 0x240000 max_arch_pfn = 0x400000000
[    0.052822] MTRR map: 5 entries (2 fixed + 3 variable; max 18), built from 8 variable MTRRs
[    0.052825] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.052878] e820: update [mem 0xc0000000-0xffffffff] usable ==> reserved
[    0.052881] last_pfn = 0xc0000 max_arch_pfn = 0x400000000
[    0.062541] Using GB pages for direct mapping
[    0.062972] Secure boot disabled
[    0.062974] ACPI: Early table checksum verification disabled
[    0.062976] ACPI: RSDP 0x000000000FBF1000 000024 (v02 VMWARE)
[    0.062980] ACPI: XSDT 0x000000000FBF1064 00005C (v01 INTEL  440BX    06040000 VMW  01324272)
[    0.062984] ACPI: SRAT 0x000000000FBF10C0 000128 (v03 VMWARE EFISRAT  06040001 VMW  000007CE)
[    0.062988] ACPI: FACP 0x000000000FBFABA4 0000F4 (v04 INTEL  440BX    06040000 PTL  000F4240)
[... cut ...]
[   17.053402] vmxnet3 0000:02:01.0 eth0: intr type 3, mode 0, 9 vectors allocated
[   17.054499] vmxnet3 0000:02:01.0 eth0: NIC Link is Up 10000 Mbps
[   17.055270] 8021q: adding VLAN 0 to HW filter on device eth0
[   20.010467] tg3 0000:03:00.0 eth1: Link is up at 1000 Mbps, full duplex
[   20.013266] tg3 0000:03:00.0 eth1: Flow control is off for TX and off for RX
[   20.013786] tg3 0000:03:00.0 eth1: EEE is disabled
[   20.014184] br-lan: port 1(eth1) entered blocking state
[   20.014577] br-lan: port 1(eth1) entered forwarding state

jwkinko, 2024-07-03:
> 用新的 commit 

n60 新提交编译的启动出现些： user.err kernel: 不知道有没有影响

Tue Jul  2 23:32:09 2024 user.info kernel: [   12.645817] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   12.652653] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.info kernel: [   12.659375] block: attempting to load /tmp/overlay/etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   12.665667] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.info kernel: [   12.672378] block: attempting to load /etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   12.681219] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.err kernel: [   12.687950] block: no usable configuration
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   12.717597] UBIFS (ubi0:4): default file-system created
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   12.724966] UBIFS (ubi0:4): Mounting in unauthenticated mode
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   12.730687] UBIFS (ubi0:4): background thread "ubifs_bgt0_4" started, PID 496
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   12.941460] UBIFS (ubi0:4): UBIFS: mounted UBI device 0, volume 4, name "rootfs_data"
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   12.949279] UBIFS (ubi0:4): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   12.959170] UBIFS (ubi0:4): FS size: 56377344 bytes (53 MiB, 444 LEBs), max 454 LEBs, journal size 2793472 bytes (2 MiB, 22 LEBs)
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   12.970793] UBIFS (ubi0:4): reserved for root: 2662841 bytes (2600 KiB)
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   12.977389] UBIFS (ubi0:4): media format: w5/r0 (latest is w5/r0), UUID 05BFF148-3764-4EBB-B86B-A321AB35E0B8, small LPT model
Tue Jul  2 23:32:09 2024 user.info kernel: [   12.991025] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   12.998073] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.004787] block: attempting to load /tmp/ubifs_cfg/etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.011259] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.017974] block: attempting to load /etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.023222] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.029935] block: no usable configuration
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   13.041432] UBIFS (ubi0:4): un-mount UBI device 0
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   13.046140] UBIFS (ubi0:4): background thread "ubifs_bgt0_4" stops
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   13.058484] UBIFS (ubi0:4): Mounting in unauthenticated mode
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   13.064196] UBIFS (ubi0:4): background thread "ubifs_bgt0_4" started, PID 499
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   13.274163] UBIFS (ubi0:4): UBIFS: mounted UBI device 0, volume 4, name "rootfs_data"
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   13.281986] UBIFS (ubi0:4): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   13.291876] UBIFS (ubi0:4): FS size: 56377344 bytes (53 MiB, 444 LEBs), max 454 LEBs, journal size 2793472 bytes (2 MiB, 22 LEBs)
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   13.303501] UBIFS (ubi0:4): reserved for root: 2662841 bytes (2600 KiB)
Tue Jul  2 23:32:09 2024 kern.notice kernel: [   13.310097] UBIFS (ubi0:4): media format: w5/r0 (latest is w5/r0), UUID 05BFF148-3764-4EBB-B86B-A321AB35E0B8, small LPT model
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.561971] block: attempting to load /tmp/overlay/upper/etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.568815] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.575533] block: attempting to load /tmp/overlay/etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.581831] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.588544] block: attempting to load /etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.593790] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.600504] block: no usable configuration
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.606240] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.613225] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.619946] block: attempting to load /tmp/ubifs_cfg/etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.626409] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.633120] block: attempting to load /etc/config/fstab
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.638370] block: unable to load configuration (fstab: Entry not found)
Tue Jul  2 23:32:09 2024 user.err kernel: [   13.645085] block: no usable configuration
[... cut ...]
Tue Jul  2 23:32:09 2024 kern.info kernel: [   13.861040] mt7530-mdio mdio-bus:1f lan1: Link is Down
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.869199] procd: - early -
Tue Jul  2 23:32:09 2024 user.info kernel: [   13.872127] procd: - watchdog -
Tue Jul  2 23:32:09 2024 user.info kernel: [   14.607566] procd: - watchdog -
Tue Jul  2 23:32:09 2024 user.info kernel: [   14.752451] procd: - ubus -
Tue Jul  2 23:32:09 2024 user.info kernel: [   15.116638] procd: - init -
Tue Jul  2 23:32:09 2024 user.info kernel: [   16.218392] kmodloader: loading kernel modules from /etc/modules.d/*

Turing-China, 2024-07-03:
> 用新的 commit

编译完成运行了4小时 ,目前正常没崩溃了

jwkinko, 2024-07-03:
> > 用新的 commit
> 
> 编译完成运行了4小时 ,目前正常没崩溃了

还是会重启

Turing-China, 2024-07-03:
> > > 用新的 commit
> > 
> > 
> > 编译完成运行了4小时 ,目前正常没崩溃了
> 
> 还是会重启

我这里目前正常了 可能你的情况跟我不一样吧 贴下日志 让大哥看看吧

jwkinko, 2024-07-03:
> > > > 用新的 commit
> > > 
> > > 
> > > 编译完成运行了4小时 ,目前正常没崩溃了
> > 
> > 还是会重启
> 
> 我这里目前正常了 可能你的情况跟我不一样吧 贴下日志 让大哥看看吧

好像稳定了，十多个小时没重启了，记得编译的这个5d1aadf436ed324d371765a965b2ae4de65ce135提交的的固件，准备用最新提交试试。

zsrjim, 2024-07-04:
> > > > > 用新的 commit
> > > > 
> > > > 
> > > > 编译完成运行了4小时 ,目前正常没崩溃了
> > > 
> > > 
> > > 还是会重启
> > 
> > 
> > 我这里目前正常了 可能你的情况跟我不一样吧 贴下日志 让大哥看看吧
> 
> 好像稳定了，十多个小时没重启了，记得编译的这个5d1aadf436ed324d371765a965b2ae4de65ce135提交的的固件，准备用最新提交试试。

编译的啥。提交的什么新固件

closed 2024-07-04 by coolsnowwolf