coolsnowwolf/lede #12294: Debian 11 本地编译一直报错 error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.

iHub-2020, 2024-07-02:
### 详细叙述

之前一直在自己的本地Debain11虚拟机编译没有任何问题，然而最近这两天在编译的过程中一直卡在git clone https://github.com/coolsnowwolf/lede这一步过去不，重新安装系统后仍然无法通过，一直显示：
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
梯子肯定是开的全局，在网上查询了一些办法尝试了无数次仍然无法下载，不知道原因是啥，请大神指点迷津，谢谢

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R2S

### 详细日志

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Jul  1 02:45:29 2024 from 192.168.109.1
openwrt@ryan:~$ git clone https://github.com/coolsnowwolf/lede
Cloning into 'lede'...
remote: Enumerating objects: 139911, done.
remote: Counting objects: 100% (30802/30802), done.
remote: Compressing objects: 100% (7410/7410), done.
error: RPC failed; curl 56 GnuTLS recv error (-9): Error decoding the received TLS packet.
error: 7370 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: index-pack failed

jwkinko, 2024-07-02:
换梯子

iHub-2020, 2024-07-02:
手头上仅有的几个梯子都过了，没用啊，会不会是一直折腾把github的流量用光了的缘故？

hzjnet, 2024-07-03:
> 手头上仅有的几个梯子都过了，没用啊，会不会是一直折腾把github的流量用光了的缘故？

除了梯子也可以换git的环境变量，例如设置成：
url.https://githubfast.com.insteadof=https://github.com

VIKINGYFY, 2024-07-03:
光有梯子没用，还要看质量和工具

aaronchin, 2024-07-04:
git clone https://mirror.ghproxy.com/https://github.com/coolsnowwolf/lede
用git加速拉

iHub-2020, 2024-07-07:
换了自己的小鸡VPS线路可以了，最近一直用某大佬的CF-pages部署的ws节点，可能跟这个有关。
感谢几位大佬的指点，谢谢！

closed 2024-07-07 by coolsnowwolf