coolsnowwolf/lede #12296: openwrt最新编译，几次都不成功报错

grz161, 2024-07-02:
### 详细叙述

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency iptables-mod-conntrack-extra for firewall
 * pkg_hash_check_unresolved: cannot find dependency iptables-mod-fullconenat for firewall
 * pkg_hash_fetch_best_installation_candidate: Packages for firewall found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 *      iptables-mod-conntrack-extra
 *      iptables-mod-fullconenat
 * opkg_install_cmd: Cannot install package default-settings.
 * check_conflicts_for: The following packages conflict with firewall:
 * check_conflicts_for:         firewall4 *
 * opkg_install_cmd: Cannot install package firewall.
 * resolve_conffiles: Existing conffile /home/grz/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat is different from the conffile in the new package. The new conffile will be placed a                         t /home/grz/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat-opkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/grz/lede'
make[1]: *** [package/Makefile:111: /home/grz/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/grz/lede'
make: *** [/home/grz/lede/include/toplevel.mk:231: world] Error 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86/64

### 详细日志

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency iptables-mod-conntrack-extra for firewall
 * pkg_hash_check_unresolved: cannot find dependency iptables-mod-fullconenat for firewall
 * pkg_hash_fetch_best_installation_candidate: Packages for firewall found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 *      iptables-mod-conntrack-extra
 *      iptables-mod-fullconenat
 * opkg_install_cmd: Cannot install package default-settings.
 * check_conflicts_for: The following packages conflict with firewall:
 * check_conflicts_for:         firewall4 *
 * opkg_install_cmd: Cannot install package firewall.
 * resolve_conffiles: Existing conffile /home/grz/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat is different from the conffile in the new package. The new conffile will be placed a                         t /home/grz/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat-opkg.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/grz/lede'
make[1]: *** [package/Makefile:111: /home/grz/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/grz/lede'
make: *** [/home/grz/lede/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2024-07-02:
rm -rf bin

grz161, 2024-07-03:
谢谢大佬指教

closed 2024-07-03 by coolsnowwolf