coolsnowwolf/lede #12304: 2024年7月源码编译r7800出现交换芯片驱动错误，6月版本源码正常

hzjnet, 2024-07-05:
### 详细叙述

drivers/net/phy/ar8216.c: In function 'ar8xxx_phy_config_init' 错误，内核由5.10.220更换成5.10.219错误依旧。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

网件R7800

### 详细日志

drivers/net/phy/ar8216.c: In function 'ar8xxx_phy_config_init':
drivers/net/phy/ar8216.c:2464:22: error: 'struct net_device' has no member named 'extra_priv_flags'; did you mean 'priv_flags'?
 2464 |                 dev->extra_priv_flags |= IFF_NO_IP_ALIGN;
      |                      ^~~~~~~~~~~~~~~~
      |                      priv_flags
drivers/net/phy/ar8216.c: In function 'ar8xxx_phy_detach':
drivers/net/phy/ar8216.c:2703:14: error: 'struct net_device' has no member named 'extra_priv_flags'; did you mean 'priv_flags'?
 2703 |         dev->extra_priv_flags &= ~IFF_NO_IP_ALIGN;
      |              ^~~~~~~~~~~~~~~~
      |              priv_flags
scripts/Makefile.build:286: recipe for target 'drivers/net/phy/ar8216.o' failed
make[8]: *** [drivers/net/phy/ar8216.o] Error 1
scripts/Makefile.build:503: recipe for target 'drivers/net/phy' failed
make[7]: *** [drivers/net/phy] Error 2
scripts/Makefile.build:503: recipe for target 'drivers/net' failed
make[6]: *** [drivers/net] Error 2
Makefile:1832: recipe for target 'drivers' failed
make[5]: *** [drivers] Error 2

mmc1987, 2024-07-05:
编译k2p 也遇到问题了   
CC      drivers/net/phy/ar8216.o
drivers/net/phy/ar8216.c: In function 'ar8229_hw_init':
drivers/net/phy/ar8216.c:902:2: error: too many arguments to function 'of_get_phy_mode'
  of_get_phy_mode(priv->pdev->of_node, &phy_if_mode);
  ^~~~~~~~~~~~~~~
In file included from drivers/net/phy/ar8216.c:28:
./include/linux/of_net.h:13:12: note: declared here
 extern int of_get_phy_mode(struct device_node *np);
            ^~~~~~~~~~~~~~~
drivers/net/phy/ar8216.c: In function 'ar8xxx_phy_config_init':
drivers/net/phy/ar8216.c:2464:8: error: 'struct net_device' has no member named 'extra_priv_flags'; did you mean 'priv_flags'?
   dev->extra_priv_flags |= IFF_NO_IP_ALIGN;
        ^~~~~~~~~~~~~~~~
        priv_flags
drivers/net/phy/ar8216.c: In function 'ar8xxx_phy_detach':
drivers/net/phy/ar8216.c:2703:7: error: 'struct net_device' has no member named 'extra_priv_flags'; did you mean 'priv_flags'?
  dev->extra_priv_flags &= ~IFF_NO_IP_ALIGN;
       ^~~~~~~~~~~~~~~~
       priv_flags
make[8]: *** [scripts/Makefile.build:262: drivers/net/phy/ar8216.o] Error 1
make[7]: *** [scripts/Makefile.build:497: drivers/net/phy] Error 2
make[6]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make[5]: *** [Makefile:1750: drivers] Error 2
make[5]: Leaving directory '/home/mmc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.278'
make[4]: *** [Makefile:25: /home/mmc/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.278/.modules] Error 2
make[4]: Leaving directory '/home/mmc/lede/target/linux/ramips'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/mmc/lede/target/linux'
time: target/linux/compile#251.35#38.67#280.25
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/mmc/lede'
make[1]: *** [target/Makefile:23: /home/mmc/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/mmc/lede'
make: *** [/home/mmc/lede/include/toplevel.mk:231: world] Error 2

noviachen, 2024-07-06:
+1，小米路由器3也是这个错误

`make[5]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.278'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      drivers/net/phy/ar8216.o
drivers/net/phy/ar8216.c: In function 'ar8229_hw_init':
drivers/net/phy/ar8216.c:902:2: error: too many arguments to function 'of_get_phy_mode'
  of_get_phy_mode(priv->pdev->of_node, &phy_if_mode);
  ^~~~~~~~~~~~~~~
In file included from drivers/net/phy/ar8216.c:28:
./include/linux/of_net.h:13:12: note: declared here
 extern int of_get_phy_mode(struct device_node *np);
            ^~~~~~~~~~~~~~~
drivers/net/phy/ar8216.c: In function 'ar8xxx_phy_config_init':
drivers/net/phy/ar8216.c:2464:8: error: 'struct net_device' has no member named 'extra_priv_flags'; did you mean 'priv_flags'?
   dev->extra_priv_flags |= IFF_NO_IP_ALIGN;
        ^~~~~~~~~~~~~~~~
        priv_flags
drivers/net/phy/ar8216.c: In function 'ar8xxx_phy_detach':
drivers/net/phy/ar8216.c:2703:7: error: 'struct net_device' has no member named 'extra_priv_flags'; did you mean 'priv_flags'?
  dev->extra_priv_flags &= ~IFF_NO_IP_ALIGN;
       ^~~~~~~~~~~~~~~~
       priv_flags
make[8]: *** [scripts/Makefile.build:262: drivers/net/phy/ar8216.o] Error 1
make[7]: *** [scripts/Makefile.build:497: drivers/net/phy] Error 2
make[6]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make[5]: *** [Makefile:1750: drivers] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.278'
make[4]: *** [Makefile:25: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.278/.modules] Error 2
make[3]: *** [Makefile:11: compile] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#4.09#1.18#5.09
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2`

luxiaok, 2024-07-07:
+1, xiaomi-miwifi-mini

coolsnowwolf, 2024-07-07:
fixed

closed 2024-07-07 by coolsnowwolf