coolsnowwolf/lede #12319: 红米AX6000 编译后没有无线信号

Turing-China, 2024-07-08:
### 详细叙述

红米AX6000 编译后没有无线信号；其他功能正常；使用最新6.6内核编译;编译时间7月8日
不知有没有大哥遇到过;是我编译什么地方没选对吗；我看无线驱动也有

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

红米AX6000

### 详细日志

[image]
[image]
[image]
[image]
[image]

coolsnowwolf, 2024-07-09:
终端不连上去有什么信号？

closed 2024-07-09 by coolsnowwolf

Turing-China, 2024-07-09:
> 终端不连上去有什么信号？

终端看不到信号呀，搜索不到路由器的WIFI