coolsnowwolf/lede #12320: JDCloud AX1800 pro买了uboot，刷机后没有WIFI

whlsxl, 2024-07-10:
### 详细叙述

[image]

用LEAN的 uboot，刷recovery后，没有Wi-Fi的选项。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

JDCloud AX1800 pro

### 详细日志

功能性问题。

aiamadeus, 2024-07-10:
刷了所谓大分区就是这样的
自己手动 `wifi config && wifi` 或者重启就有了

xfmly, 2024-07-11:
非常感谢暗云大佬，完美解决问题。

whlsxl, 2024-07-11:
> 非常感谢暗云大佬，完美解决问题。

请问您操作的步骤是什么？执行完`wifi config && wifi`后重启，还是没有Wi-Fi选项。

xfmly, 2024-07-11:
执行完就有了

xfmly, 2024-07-11:
重启后wifi又没有了,执行完wifi config && wifi后wifi恢复openwrt，还是安心用qwrt吧。

whlsxl, 2024-07-11:
> 重启后wifi又没有了,执行完wifi config && wifi后wifi恢复openwrt，还是安心用qwrt吧。

执行之后还是没有，但是如果只是重启之后恢复，可以加到开机启动里。

jeasonliang, 2024-07-11:
我的是刷入后配置好所有后  一断电重新启动就恢复了,啥配置都没了!

closed 2024-07-14 by aiamadeus

whlsxl, 2024-07-15:
@aiamadeus 编译了最新代码，这个问题还是存在啊？

aiamadeus, 2024-08-08:
[image]
实测没有问题，还是你用的 uboot 或者分区表有毒