coolsnowwolf/lede #12323: 编译5.4内核错误

audyfan, 2024-07-11:
### 详细叙述

使用最新源码，编译mediatek H3C NX30 pro 5.4内核失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

H3C NX30 PRO

### 详细日志

enWrt-libtool-patched-2.2: link: aarch64-openwrt-linux-musl-gcc -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wstack-usage=262144 -DELF_LIST_OPTIONS=true -DELF_SHLIB_LIST_OPTIONS=true -DELF_PLT_UNWIND_LIST_OPTIONS=false -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40=binutils-2.40 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -DPIC -fPIC -Wno-unused-value -znow -zrelro -o ld-new ldgram.o ldlex-wrapper.o lexsup.o ldlang.o mri.o ldctor.o ldmain.o ldwrite.o ldexp.o ldemul.o ldver.o ldmisc.o ldfile.o ldcref.o plugin.o ldbuildid.o eaarch64linux.o eaarch64elf.o eaarch64elf32.o eaarch64elf32b.o eaarch64elfb.o earmelf.o earmelfb.o eaarch64linuxb.o eaarch64linux32.o eaarch64linux32b.o earmelfb_linux_eabi.o earmelf_linux_eabi.o ldelf.o ldelfgen.o  -L/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/lib -L/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib ../bfd/.libs/libbfd.so -L/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/bfd/../libiberty/pic ../libctf/.libs/libctf.so -L/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/libctf/../libiberty/pic /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/bfd/.libs/libbfd.so /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/libsframe/.libs/libsframe.so -liberty ../libiberty/libiberty.a -lz -Wl,-rpath -Wl,/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/bfd/.libs -Wl,-rpath -Wl,/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/libctf/.libs -Wl,-rpath -Wl,/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/libsframe/.libs
make[8]: Leaving directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ld'
make[7]: Leaving directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ld'
make[6]: Leaving directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ld'
make[5]: Leaving directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40'
make[4]: Leaving directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40'
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts   -fmacro-prefix-map=/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40=binutils-2.40 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC -Wno-unused-value  -I/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include -I/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify -I/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include   " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts   -fmacro-prefix-map=/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40=binutils-2.40 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC -Wno-unused-value  -I/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include -I/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify -I/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include   " LDFLAGS="-L/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/lib -L/home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib   -znow -zrelro  " make -C /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts   -fmacro-prefix-map=/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40=binutils-2.40 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC -Wno-unused-value" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" DESTDIR="/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install"  install;
make[4]: Entering directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40'
make[5]: Entering directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40'
/bin/bash ./mkinstalldirs /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr
mkdir -p -- /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr
make[6]: Entering directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/libsframe'
make[7]: Entering directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/libsframe'
 /usr/bin/mkdir -p '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/lib'
 /bin/bash ./libtool   --mode=install /home/audy/lede/staging_dir/host/bin/install -c   libsframe.la '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/lib'
OpenWrt-libtool-patched-2.2: install: /home/audy/lede/staging_dir/host/bin/install -c .libs/libsframe.so.0.0.0 /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/lib/libsframe.so.0.0.0
OpenWrt-libtool-patched-2.2: install: (cd /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/lib && { ln -s -f libsframe.so.0.0.0 libsframe.so.0 || { rm -f libsframe.so.0 && ln -s libsframe.so.0.0.0 libsframe.so.0; }; })
OpenWrt-libtool-patched-2.2: install: (cd /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/lib && { ln -s -f libsframe.so.0.0.0 libsframe.so || { rm -f libsframe.so && ln -s libsframe.so.0.0.0 libsframe.so; }; })
OpenWrt-libtool-patched-2.2: install: /home/audy/lede/staging_dir/host/bin/install -c .libs/libsframe.lai /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/lib/libsframe.la
OpenWrt-libtool-patched-2.2: install: /home/audy/lede/staging_dir/host/bin/install -c .libs/libsframe.a /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/lib/libsframe.a
OpenWrt-libtool-patched-2.2: install: chmod 644 /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/lib/libsframe.a
OpenWrt-libtool-patched-2.2: install: aarch64-openwrt-linux-musl-gcc-ranlib --plugin /home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/libexec/gcc/aarch64-openwrt-linux-musl/11.3.0/liblto_plugin.so --plugin /home/audy/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/libexec/gcc/aarch64-openwrt-linux-musl/11.3.0/liblto_plugin.so /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/lib/libsframe.a
OpenWrt-libtool-patched-2.2: install: warning: remember to run `libtool --finish /usr/lib'
 /usr/bin/mkdir -p '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/include'
 /home/audy/lede/staging_dir/host/bin/install -c -m 644 ./../include/sframe.h ./../include/sframe-api.h '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/include'
 /usr/bin/mkdir -p '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/info'
 /home/audy/lede/staging_dir/host/bin/install -c -m 644 ./doc/sframe-spec.info '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/binutils-2.40/ipkg-install/usr/info'
[... cut ...]
touch /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.binutils_installed
make[3]: Leaving directory '/home/audy/lede/package/devel/binutils'
time: package/devel/binutils/compile#180.39#25.78#106.36
make[2]: Leaving directory '/home/audy/lede'
make[1]: *** [package/Makefile:110: /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/audy/lede'
make: *** [/home/audy/lede/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2024-07-11:
这玩意主线啥时候支持 5.4 内核了？

coolsnowwolf, 2024-07-11:
[image]
最低不是 5.15 ？

audyfan, 2024-07-11:
> ![image](https://private-user-images.githubusercontent.com/31687149/347705982-f127bd32-296b-4675-978d-c98c807e8997.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

终于见到大佬了。请教大佬，那如果是要编译5.4内核的话，要怎么做啊？

audyfan, 2024-07-11:
编译小学生，用gpt给的步骤试了不行

closed 2024-07-11 by audyfan