coolsnowwolf/lede #12324: 关于ikev2协议支持问题

lexin8, 2024-07-11:
### 详细叙述

安卓13系统大部分已经不支持L2TP老协议了，只有IKEv2 ，
[image]

目前查到资料解决办法有：
1.安装strongswan，没有luci界面配置不友好。[`官方源有](https://github.com/openwrt/luci/tree/master/applications/luci-app-strongswan-swanctl)，目前LEDE没有这个luci

2.docker搭建IKEv2 [服务器](https://github.com/hwdsl2/docker-ipsec-vpn-server/blob/master/README-zh.md)

3.等待SoftEtherVPN支持 [https://github.com/SoftEtherVPN/SoftEtherVPN/issues/13](url)

4.等待大神指点。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86-64

### 详细日志

Completely failed log ！

coolsnowwolf, 2024-07-11:
切换 23.05 luci 即可用

lexin8, 2024-07-12:
> 切换 23.05 luci 即可用

[image]

Jiang00, 2024-11-06:
> 切换 23.05 luci 即可用

这个是把luci换成https://github.com/openwrt/luci/tree/openwrt-23.05这个吗,不太懂啥意思

Jiang00, 2024-11-10:
> 切换 23.05 luci 即可用
https://github.com/coolsnowwolf/luci/tree/openwrt-23.05/applications
这个里面没有找到
luci-app-strongswan-swanctl

sbcarp, 2024-11-29:
同上，目前feed中用的是 src-git luci https://github.com/coolsnowwolf/luci.git;openwrt-23.05
但不知道安装哪个app

sbcarp, 2024-11-29:
我找到luci-app-strongswan-swanctl了

siasjack, 2025-07-25:
> 我找到luci-app-strongswan-swanctl了

哥们   是基于哪个版本的openwrt      是否可以分享下连接

coolsnowwolf, 2026-09-05:
当前 LuCI feed 已包含 `applications/luci-app-strongswan-swanctl`，可用于 strongSwan/IKEv2 配置；同时存在 `luci-app-ipsec-vpnd`。原 issue 所需的 LuCI 支持已经提供，无需修改 LEDE 源码。

closed 2026-09-05 by coolsnowwolf