coolsnowwolf/lede #12329: esxi直通8125b 网卡不定时断网

Lasx, 2024-07-13:
### 详细叙述

esxi直通8125b 网卡不定时断网

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

esxi8 rtl8125b

### 详细日志

esxi6.7  下6.1.95内核报错会导致虚拟机重启，且波及宿主机，日志不能保存。切换到esxi 8下得到断网日志。
错误日志如下：
`Jul 13 10:18:20 OpenWrt kernel: ------------[ cut here ]------------
Jul 13 10:18:20 OpenWrt kernel: NETDEV WATCHDOG: eth1 (r8125): transmit queue 1 timed out
Jul 13 10:18:20 OpenWrt kernel: WARNING: CPU: 0 PID: 0 at dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel: Modules linked in: shortcut_fe_cm xt_connlimit pppoe ppp_async nf_conncount iptable_nat i915 amdgpu xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FULLCONENAT xt_CT video pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mlx5_core ipt_REJECT bnx2x ax88179_178a aqc111 xt_time xt_tcpudp xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_LOG wmi via_velocity usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm tcp_bbr slhc rtl8150 r8168 r8152 r8126 r8125 pcnet32
Jul 13 10:18:20 OpenWrt kernel:  nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mlxfw mlx4_en mlx4_core macvlan libcrc32c iptable_raw iptable_mangle iptable_filter ip_tables igc iavf i40e gpu_sched forcedeth e1000e drm_display_helper drm_buddy crc_ccitt bnx2 asn1_decoder alx 8139too 8139cp ntfs3 evdev i2c_dev cryptodev xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ixgbe igbvf e1000 amd_xgbe mdio_devres mdio netlink_diag udp_diag tcp_diag raw_diag inet_diag tun nls_utf8 ena shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_ssse3 sha1_generic authencesn authenc arc4 nls_iso8859_1 nls_cp437 uas sdhci_pltfm
Jul 13 10:18:20 OpenWrt kernel:  igb vfat fat exfat button_hotplug tg3 mii libphy
Jul 13 10:18:20 OpenWrt kernel: CPU: 0 PID: 0 Comm: swapper/0 Not tainted 6.1.95 #0
Jul 13 10:18:20 OpenWrt kernel: Hardware name: VMware, Inc. VMware20,1/440BX Desktop Reference Platform, BIOS VMW201.00V.23553139.B64.2403260940 03/26/2024
Jul 13 10:18:20 OpenWrt kernel: RIP: 0010:dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel: Code: 30 e9 51 ff ff ff 4c 89 ef c6 05 72 c6 b5 00 01 e8 46 08 fc ff 44 89 f1 4c 89 ee 48 c7 c7 70 04 0a 82 48 89 c2 e8 01 af 90 ff <0f> 0b e9 72 ff ff ff 66 2e 0f 1f 84 00 00 00 00 00 48 89 37 c7 47
Jul 13 10:18:20 OpenWrt kernel: RSP: 0018:ffffc90000003e98 EFLAGS: 00010292
Jul 13 10:18:20 OpenWrt kernel: RAX: 0000000000000039 RBX: ffff888007300438 RCX: 0000000000000027
Jul 13 10:18:20 OpenWrt kernel: RDX: ffff88803ec1d4c8 RSI: 0000000000000001 RDI: ffff88803ec1d4c0
Jul 13 10:18:20 OpenWrt kernel: RBP: ffffc90000003eb8 R08: 0000000000000000 R09: 00000000ffffefff
Jul 13 10:18:20 OpenWrt kernel: R10: ffffc90000003d18 R11: ffffffff822b2ee8 R12: ffff88800730039c
Jul 13 10:18:20 OpenWrt kernel: R13: ffff888007300000 R14: 0000000000000001 R15: ffffffff817b5940
Jul 13 10:18:20 OpenWrt kernel: FS:  0000000000000000(0000) GS:ffff88803ec00000(0000) knlGS:0000000000000000
Jul 13 10:18:20 OpenWrt kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jul 13 10:18:20 OpenWrt kernel: CR2: 00007fe3492d317c CR3: 000000000220e004 CR4: 00000000001706f0
Jul 13 10:18:20 OpenWrt kernel: Call Trace:
Jul 13 10:18:20 OpenWrt kernel:  <IRQ>
Jul 13 10:18:20 OpenWrt kernel:  ? show_regs.part.0+0x1e/0x20
Jul 13 10:18:20 OpenWrt kernel:  ? show_regs.cold+0x8/0xd
Jul 13 10:18:20 OpenWrt kernel:  ? __warn+0x6e/0xc0
Jul 13 10:18:20 OpenWrt kernel:  ? dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel:  ? report_bug+0xed/0x140
Jul 13 10:18:20 OpenWrt kernel:  ? handle_bug+0x46/0x80
Jul 13 10:18:20 OpenWrt kernel:  ? exc_invalid_op+0x19/0x70
Jul 13 10:18:20 OpenWrt kernel:  ? asm_exc_invalid_op+0x1b/0x20
Jul 13 10:18:20 OpenWrt kernel:  ? pfifo_fast_init+0x120/0x120
Jul 13 10:18:20 OpenWrt kernel:  ? dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel:  ? dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel:  ? pfifo_fast_init+0x120/0x120
Jul 13 10:18:20 OpenWrt kernel:  call_timer_fn.constprop.0+0x13/0x80
Jul 13 10:18:20 OpenWrt kernel:  run_timer_softirq+0x271/0x2c0
Jul 13 10:18:20 OpenWrt kernel:  ? lapic_next_deadline+0x27/0x30
Jul 13 10:18:20 OpenWrt kernel:  ? clockevents_program_event+0x8c/0xe0
Jul 13 10:18:20 OpenWrt kernel:  handle_softirqs+0xc1/0x1a0
Jul 13 10:18:20 OpenWrt kernel:  __irq_exit_rcu+0x5b/0x80
Jul 13 10:18:20 OpenWrt kernel:  irq_exit_rcu+0x9/0x10
Jul 13 10:18:20 OpenWrt kernel:  sysvec_apic_timer_interrupt+0x74/0x80
Jul 13 10:18:20 OpenWrt kernel:  </IRQ>
Jul 13 10:18:20 OpenWrt kernel:  <TASK>
Jul 13 10:18:20 OpenWrt kernel:  asm_sysvec_apic_timer_interrupt+0x1b/0x20
Jul 13 10:18:20 OpenWrt kernel: RIP: 0010:native_safe_halt+0x13/0x20
Jul 13 10:18:20 OpenWrt kernel: Code: 07 0f 00 2d df 4c 47 00 f4 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 8b 05 fa 1a ad 00 85 c0 7e 07 0f 00 2d bf 4c 47 00 fb f4 <c3> cc cc cc cc cc cc cc cc cc cc cc cc 55 48 89 e5 53 e8 a6 80 ff
Jul 13 10:18:20 OpenWrt kernel: RSP: 0018:ffffffff82203da8 EFLAGS: 00000246
Jul 13 10:18:20 OpenWrt kernel: RAX: 0000000000000000 RBX: ffff888003331800 RCX: 4000000000000000
Jul 13 10:18:20 OpenWrt kernel: RDX: ffff88803ec00000 RSI: ffff888003257000 RDI: 0000000000000001
Jul 13 10:18:20 OpenWrt kernel: RBP: ffffffff82203db8 R08: ffff888003331800 R09: 0000000000000162
Jul 13 10:18:20 OpenWrt kernel: R10: 0000000000000002 R11: 0000000000000001 R12: ffff888003257064
Jul 13 10:18:20 OpenWrt kernel: R13: ffff888003257064 R14: ffffffff822d6da0 R15: ffffffff822d6d88
Jul 13 10:18:20 OpenWrt kernel:  ? acpi_idle_do_entry+0x68/0x90
Jul 13 10:18:20 OpenWrt kernel:  acpi_idle_enter+0x8a/0xd0
Jul 13 10:18:20 OpenWrt kernel:  cpuidle_enter_state+0x7e/0x2c0
Jul 13 10:18:20 OpenWrt kernel:  cpuidle_enter+0x2f/0x40
Jul 13 10:18:20 OpenWrt kernel:  call_cpuidle+0x1e/0x40
Jul 13 10:18:20 OpenWrt kernel:  do_idle+0x155/0x170
Jul 13 10:18:20 OpenWrt kernel:  cpu_startup_entry+0x28/0x30
Jul 13 10:18:20 OpenWrt kernel:  rest_init+0xc9/0xd0
Jul 13 10:18:20 OpenWrt kernel:  arch_call_rest_init+0x9/0xa
Jul 13 10:18:20 OpenWrt kernel:  start_kernel+0x67c/0x68e
Jul 13 10:18:20 OpenWrt kernel:  x86_64_start_reservations+0x29/0x2b
Jul 13 10:18:20 OpenWrt kernel:  x86_64_start_kernel+0x79/0x7f
Jul 13 10:18:20 OpenWrt kernel:  secondary_startup_64_no_verify+0xce/0xdb
Jul 13 10:18:20 OpenWrt kernel:  </TASK>
Jul 13 18:18:20 OpenWrt kernel: [52493.970345] ---[ end trace 0000000000000000 ]---
Jul 13 18:18:20 OpenWrt kernel: [52493.971121] r8125 0000:0b:00.0 eth1: Transmit timeout reset Device!
Jul 13 10:18:20 OpenWrt kernel: r8125 0000:0b:00.0 eth1: Device reseting!
Jul 13 18:18:20 OpenWrt netifd: Network device 'eth1' link is down
Jul 13 10:18:20 OpenWrt kernel: br-lan: port 2(eth1) entered disabled state
Jul 13 10:18:20 OpenWrt kernel: r8125: eth1: link down
Jul 13 18:18:22 OpenWrt netifd: Network device 'eth1' link is up
Jul 13 18:18:22 OpenWrt kernel: [52496.256158] r8125: eth1: link up
`

Lasx, 2024-07-13:
我去日志怎么成这样了，重新贴一个
Jul 13 10:18:20 OpenWrt kernel: ------------[ cut here ]------------
Jul 13 10:18:20 OpenWrt kernel: NETDEV WATCHDOG: eth1 (r8125): transmit queue 1 timed out
Jul 13 10:18:20 OpenWrt kernel: WARNING: CPU: 0 PID: 0 at dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel: Modules linked in: shortcut_fe_cm xt_connlimit pppoe ppp_async nf_conncount iptable_nat i915 amdgpu xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FULLCONENAT xt_CT video pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mlx5_core ipt_REJECT bnx2x ax88179_178a aqc111 xt_time xt_tcpudp xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_LOG wmi via_velocity usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm tcp_bbr slhc rtl8150 r8168 r8152 r8126 r8125 pcnet32
Jul 13 10:18:20 OpenWrt kernel:  nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mlxfw mlx4_en mlx4_core macvlan libcrc32c iptable_raw iptable_mangle iptable_filter ip_tables igc iavf i40e gpu_sched forcedeth e1000e drm_display_helper drm_buddy crc_ccitt bnx2 asn1_decoder alx 8139too 8139cp ntfs3 evdev i2c_dev cryptodev xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ixgbe igbvf e1000 amd_xgbe mdio_devres mdio netlink_diag udp_diag tcp_diag raw_diag inet_diag tun nls_utf8 ena shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_ssse3 sha1_generic authencesn authenc arc4 nls_iso8859_1 nls_cp437 uas sdhci_pltfm
Jul 13 10:18:20 OpenWrt kernel:  igb vfat fat exfat button_hotplug tg3 mii libphy
Jul 13 10:18:20 OpenWrt kernel: CPU: 0 PID: 0 Comm: swapper/0 Not tainted 6.1.95 #0
Jul 13 10:18:20 OpenWrt kernel: Hardware name: VMware, Inc. VMware20,1/440BX Desktop Reference Platform, BIOS VMW201.00V.23553139.B64.2403260940 03/26/2024
Jul 13 10:18:20 OpenWrt kernel: RIP: 0010:dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel: Code: 30 e9 51 ff ff ff 4c 89 ef c6 05 72 c6 b5 00 01 e8 46 08 fc ff 44 89 f1 4c 89 ee 48 c7 c7 70 04 0a 82 48 89 c2 e8 01 af 90 ff <0f> 0b e9 72 ff ff ff 66 2e 0f 1f 84 00 00 00 00 00 48 89 37 c7 47
Jul 13 10:18:20 OpenWrt kernel: RSP: 0018:ffffc90000003e98 EFLAGS: 00010292
Jul 13 10:18:20 OpenWrt kernel: RAX: 0000000000000039 RBX: ffff888007300438 RCX: 0000000000000027
Jul 13 10:18:20 OpenWrt kernel: RDX: ffff88803ec1d4c8 RSI: 0000000000000001 RDI: ffff88803ec1d4c0
Jul 13 10:18:20 OpenWrt kernel: RBP: ffffc90000003eb8 R08: 0000000000000000 R09: 00000000ffffefff
Jul 13 10:18:20 OpenWrt kernel: R10: ffffc90000003d18 R11: ffffffff822b2ee8 R12: ffff88800730039c
Jul 13 10:18:20 OpenWrt kernel: R13: ffff888007300000 R14: 0000000000000001 R15: ffffffff817b5940
Jul 13 10:18:20 OpenWrt kernel: FS:  0000000000000000(0000) GS:ffff88803ec00000(0000) knlGS:0000000000000000
Jul 13 10:18:20 OpenWrt kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jul 13 10:18:20 OpenWrt kernel: CR2: 00007fe3492d317c CR3: 000000000220e004 CR4: 00000000001706f0
Jul 13 10:18:20 OpenWrt kernel: Call Trace:
Jul 13 10:18:20 OpenWrt kernel:  <IRQ>
Jul 13 10:18:20 OpenWrt kernel:  ? show_regs.part.0+0x1e/0x20
Jul 13 10:18:20 OpenWrt kernel:  ? show_regs.cold+0x8/0xd
Jul 13 10:18:20 OpenWrt kernel:  ? __warn+0x6e/0xc0
Jul 13 10:18:20 OpenWrt kernel:  ? dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel:  ? report_bug+0xed/0x140
Jul 13 10:18:20 OpenWrt kernel:  ? handle_bug+0x46/0x80
Jul 13 10:18:20 OpenWrt kernel:  ? exc_invalid_op+0x19/0x70
Jul 13 10:18:20 OpenWrt kernel:  ? asm_exc_invalid_op+0x1b/0x20
Jul 13 10:18:20 OpenWrt kernel:  ? pfifo_fast_init+0x120/0x120
Jul 13 10:18:20 OpenWrt kernel:  ? dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel:  ? dev_watchdog+0x1bf/0x1d0
Jul 13 10:18:20 OpenWrt kernel:  ? pfifo_fast_init+0x120/0x120
Jul 13 10:18:20 OpenWrt kernel:  call_timer_fn.constprop.0+0x13/0x80
Jul 13 10:18:20 OpenWrt kernel:  run_timer_softirq+0x271/0x2c0
Jul 13 10:18:20 OpenWrt kernel:  ? lapic_next_deadline+0x27/0x30
Jul 13 10:18:20 OpenWrt kernel:  ? clockevents_program_event+0x8c/0xe0
Jul 13 10:18:20 OpenWrt kernel:  handle_softirqs+0xc1/0x1a0
Jul 13 10:18:20 OpenWrt kernel:  __irq_exit_rcu+0x5b/0x80
Jul 13 10:18:20 OpenWrt kernel:  irq_exit_rcu+0x9/0x10
Jul 13 10:18:20 OpenWrt kernel:  sysvec_apic_timer_interrupt+0x74/0x80
Jul 13 10:18:20 OpenWrt kernel:  </IRQ>
Jul 13 10:18:20 OpenWrt kernel:  <TASK>
Jul 13 10:18:20 OpenWrt kernel:  asm_sysvec_apic_timer_interrupt+0x1b/0x20
Jul 13 10:18:20 OpenWrt kernel: RIP: 0010:native_safe_halt+0x13/0x20
[... cut ...]
Jul 13 18:18:20 OpenWrt kernel: [52493.971121] r8125 0000:0b:00.0 eth1: Transmit timeout reset Device!
Jul 13 10:18:20 OpenWrt kernel: r8125 0000:0b:00.0 eth1: Device reseting!
Jul 13 18:18:20 OpenWrt netifd: Network device 'eth1' link is down
Jul 13 10:18:20 OpenWrt kernel: br-lan: port 2(eth1) entered disabled state
Jul 13 10:18:20 OpenWrt kernel: r8125: eth1: link down
Jul 13 18:18:22 OpenWrt netifd: Network device 'eth1' link is up
Jul 13 18:18:22 OpenWrt kernel: [52496.256158] r8125: eth1: link up

aiamadeus, 2024-07-14:
关掉 r8125 的 rss 呢，或者换 r8169

Lasx, 2024-07-14:
> 关掉 r8125 的 rss 呢，或者换 r8169

硬件上应该是关不了了，编译的时候不编译这个吗？换不了其它网卡，pcie 拆分的双2.5G加SSD。

aiamadeus, 2024-07-14:
> 硬件上应该是关不了了，编译的时候不编译这个吗？

https://github.com/coolsnowwolf/lede/blob/master/package/lean/r8125/patches/010-config.patch#L17
改回 `ENABLE_RSS_SUPPORT = n`
> 换不了其它网卡，pcie 拆分的双2.5G加SSD。

r8169 也能驱动 r8125，选上  r8169 去掉 r8125 就行了

Lasx, 2024-07-14:
> > 硬件上应该是关不了了，编译的时候不编译这个吗？
> 
> https://github.com/coolsnowwolf/lede/blob/master/package/lean/r8125/patches/010-config.patch#L17 改回 `ENABLE_RSS_SUPPORT = n`
> 
> > 换不了其它网卡，pcie 拆分的双2.5G加SSD。
> 
> r8169 也能驱动 r8125，选上 r8169 去掉 r8125 就行了

这个关闭rss我是知道的，8169这个稳定吗？

Lasx, 2024-08-02:
新鲜的日志，有解吗？
[67431.831923] ------------[ cut here ]------------
[67431.834531] NETDEV WATCHDOG: eth1 (r8125): transmit queue 1 timed out
[67431.837273] WARNING: CPU: 1 PID: 0 at dev_watchdog+0x29b/0x2a0
[67431.839994] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FULLCONENAT xt_CT pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mlx5_core ipt_REJECT bnx2x ax88179_178a aqc111 xt_time xt_tcpudp xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_LOG via_velocity usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm tcp_bbr slhc rtl8150 r8168 r8152 r8126 r8125 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4
[67431.840129]  nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mlxfw mlx4_en mlx4_core macvlan libcrc32c iptable_raw iptable_mangle iptable_filter ip_tables igc iavf i40e forcedeth e1000e crc_ccitt bnx2 asn1_decoder alx 8139too 8139cp ntfs3 evdev i2c_dev cryptodev xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ixgbe igbvf e1000 amd_xgbe mdio_devres mdio netlink_diag udp_diag tcp_diag raw_diag inet_diag tun nls_utf8 ena shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_ssse3 sha512_generic sha3_generic sha1_ssse3 sha1_generic jitterentropy_rng drbg hmac authencesn authenc arc4 nls_iso8859_1 nls_cp437 uas sdhci_pltfm
[67431.868725]  igb vfat fat exfat button_hotplug tg3 mii libphy
[67431.910802] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.15.162 #0
[67431.915655] Hardware name: VMware, Inc. VMware20,1/440BX Desktop Reference Platform, BIOS VMW201.00V.23553139.B64.2403260940 03/26/2024
[67431.925406] RIP: 0010:dev_watchdog+0x29b/0x2a0
[67431.930234] Code: ff ff 48 8b 5d d0 c6 05 2d 33 b9 00 01 48 89 df e8 ba 30 fb ff 44 89 e9 48 89 de 48 c7 c7 20 c5 06 82 48 89 c2 e8 39 a8 14 00 <0f> 0b eb af 90 55 48 89 e5 41 57 49 89 d7 41 56 4d 89 c6 41 55 49
[67431.940586] RSP: 0018:ffffc900000c4e88 EFLAGS: 00010292
[67431.945659] RAX: 0000000000000039 RBX: ffff888006140000 RCX: 0000000000000000
[67431.950825] RDX: ffff88803ed1e120 RSI: ffff88803ed1c580 RDI: ffff88803ed1c580
[67431.956015] RBP: ffffc900000c4ec0 R08: 0000000000000000 R09: ffffc900000c4c98
[67431.961209] R10: ffffc900000c4c90 R11: ffffffff822b20c8 R12: ffff888006140478
[67431.966365] R13: 0000000000000001 R14: ffff888005600c40 R15: 0000000000000001
[67431.971577] FS:  0000000000000000(0000) GS:ffff88803ed00000(0000) knlGS:0000000000000000
[67431.976913] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[67431.982121] CR2: 00007f4b86884748 CR3: 00000000052e0003 CR4: 00000000001706e0
[67431.987405] Call Trace:
[67431.992445]  <IRQ>
[67431.997351]  ? show_regs.part.0+0x1e/0x21
[67432.002406]  ? show_regs.cold+0x8/0xd
[67432.007409]  ? __warn+0x76/0xf0
[67432.012583]  ? dev_watchdog+0x29b/0x2a0
[67432.017509]  ? report_bug+0x84/0xa0
[67432.022451]  ? handle_bug+0x39/0x90
[67432.027311]  ? exc_invalid_op+0x19/0x70
[67432.032270]  ? asm_exc_invalid_op+0x1b/0x20
[67432.037267]  ? dev_watchdog+0x29b/0x2a0
[67432.042206]  ? dev_watchdog+0x29b/0x2a0
[67432.047122]  ? pfifo_fast_enqueue+0x150/0x150
[67432.052042]  call_timer_fn.constprop.0+0x13/0x80
[67432.056933]  run_timer_softirq+0x289/0x2e0
[67432.061766]  ? lapic_next_deadline+0x27/0x30
[67432.066516]  ? clockevents_program_event+0x8d/0xf0
[67432.071216]  ? tick_program_event+0x3e/0x70
[67432.075737]  handle_softirqs+0xbf/0x1b0
[67432.080125]  irq_exit_rcu+0x5d/0x90
[67432.084414]  sysvec_apic_timer_interrupt+0x75/0x80
[67432.088639]  </IRQ>
[67432.092616]  <TASK>
[67432.096480]  asm_sysvec_apic_timer_interrupt+0x1b/0x20
[67432.100312] RIP: 0010:native_safe_halt+0x13/0x20
[67432.104101] Code: 07 0f 00 2d ff 82 4f 00 f4 c3 cc 66 66 2e 0f 1f 84 00 00 00 00 00 90 8b 05 12 c3 b6 00 85 c0 7e 07 0f 00 2d df 82 4f 00 fb f4 <c3> cc cc cc cc cc cc cc cc cc cc cc cc 55 48 89 e5 53 e8 b6 72 80
[67432.112045] RSP: 0018:ffffc9000008fe38 EFLAGS: 00000246
[67432.115712] RAX: 0000000000000000 RBX: ffff888002963000 RCX: 0000000000000020
[67432.119374] RDX: ffff88803ed00000 RSI: ffff8880030d0800 RDI: ffff8880030d0864
[67432.122969] RBP: ffffc9000008fe40 R08: ffff888002963000 R09: 000000000000019f
[67432.126420] R10: 0000000000000000 R11: 000000000000016b R12: 0000000000000001
[67432.129797] R13: ffff8880030d0864 R14: ffffffff822d3f00 R15: ffffffff822d3ee8
[67432.133175]  ? acpi_idle_do_entry+0x4b/0x60
[67432.136534]  acpi_idle_enter+0x8a/0xd0
[67432.139572]  cpuidle_enter_state+0x7e/0x2c0
[67432.142626]  cpuidle_enter+0x2f/0x40
[67432.145551]  call_cpuidle+0x1e/0x40
[67432.148379]  do_idle+0x159/0x180
[67432.151119]  cpu_startup_entry+0x1b/0x20
[... cut ...]
[67434.385671] br-lan: port 2(eth1) entered blocking state
[67434.388379] br-lan: port 2(eth1) entered forwarding state
[67965.846391] r8125 0000:0b:00.0 eth1: Transmit timeout reset Device!
[67965.870372] r8125 0000:0b:00.0 eth1: Device reseting!
[67965.897685] br-lan: port 2(eth1) entered disabled state
[67965.901472] r8125: eth1: link down
[67968.073257] r8125: eth1: link up
[67968.076309] br-lan: port 2(eth1) entered blocking state
[67968.079177] br-lan: port 2(eth1) entered forwarding state

coolsnowwolf, 2026-09-05:
日志显示的是 VMware ESXi 直通环境下 r8125 的 `NETDEV WATCHDOG`/queue 1 超时，驱动随后已经执行 reset；没有提供芯片状态、RSS 配置、IOMMU/PCIe 错误或裸机对照，无法证明是 LEDE 通用回归。当前 master 的 r8125 已提供 regular（默认关闭 RSS）和独立 RSS 变体，并固定 `CONFIG_ASPM=n`；评论也给出了关闭 RSS 或改用主线 r8169 的可行绕过。该问题属于特定 ESXi 直通/驱动组合，暂无可安全通用补丁，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf