coolsnowwolf/lede #12330: redmi-router-ax6000无WIFI信号问题

Turing-China, 2024-07-14:
### 详细叙述

使用最新源码编译
6.6.37内核
终端找不到路由器的WIFI信号;手机电脑均找不到
路由器后台可以看到无线选项和网卡均正常
[image]
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

redmi-router-ax6000

### 详细日志

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.37 (runner@fv-az1114-140) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6747-7ecb40bb3) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sat Jul 13 15:02:08 2024
[    0.000000] Machine model: Xiaomi Redmi Router AX6000
[    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
[    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
[    0.000000] OF: reserved mem: 0x000000004fc00000..0x000000004fcfffff (1024 KiB) nomap non-reusable wmcpu-reserved@4fc00000
[    0.000000] OF: reserved mem: 0x000000004fd00000..0x000000004fd3ffff (256 KiB) nomap non-reusable wo-emi@4fd00000
[    0.000000] OF: reserved mem: 0x000000004fd40000..0x000000004fd7ffff (256 KiB) nomap non-reusable wo-emi@4fd40000
[    0.000000] OF: reserved mem: 0x000000004fd80000..0x000000004ffbffff (2304 KiB) nomap non-reusable wo-data@4fd80000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000004fbfffff]
[    0.000000]   node   0: [mem 0x000000004fc00000-0x000000004ffbffff]
[    0.000000]   node   0: [mem 0x000000004ffc0000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s34920 r8192 d30616 u73728
[    0.000000] pcpu-alloc: s34920 r8192 d30616 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyS0,115200n1 loglevel=8 firmware=1 uart_en=1
[    0.000000] Unknown kernel command line parameters "firmware=1 uart_en=1", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 0MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: SWIOTLB bounce buffer size roundup to 1MB
[    0.000000] software IO TLB: mapped [mem 0x000000005fcc7000-0x000000005fdc7000] (1MB)
[    0.000000] Memory: 495880K/524288K available (8640K kernel code, 916K rwdata, 2516K rodata, 384K init, 291K bss, 28408K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 640 SPIs implemented
[... 245 lines cut ...]
[   30.966414] mt7530-mdio mdio-bus:1f wan: Link is Down
[   30.973368] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[   32.427146] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[   32.435895] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[   32.436477] mt7530-mdio mdio-bus:1f lan2: configuring for phy/gmii link mode
[   32.455639] br-lan: port 1(lan2) entered blocking state
[   32.460865] br-lan: port 1(lan2) entered disabled state
[   32.466086] mt7530-mdio mdio-bus:1f lan2: entered allmulticast mode
[   32.472350] mtk_soc_eth 15100000.ethernet eth0: entered allmulticast mode
[   32.481622] mt7530-mdio mdio-bus:1f lan2: entered promiscuous mode
[   32.493382] mt7530-mdio mdio-bus:1f lan3: configuring for phy/gmii link mode
[   32.504372] br-lan: port 2(lan3) entered blocking state
[   32.509637] br-lan: port 2(lan3) entered disabled state
[   32.514888] mt7530-mdio mdio-bus:1f lan3: entered allmulticast mode
[   32.522581] mt7530-mdio mdio-bus:1f lan3: entered promiscuous mode
[   32.531911] mt7530-mdio mdio-bus:1f lan4: configuring for phy/gmii link mode
[   32.542831] br-lan: port 3(lan4) entered blocking state
[   32.548043] br-lan: port 3(lan4) entered disabled state
[   32.553279] mt7530-mdio mdio-bus:1f lan4: entered allmulticast mode
[   32.561036] mt7530-mdio mdio-bus:1f lan4: entered promiscuous mode
[   32.570664] mt7530-mdio mdio-bus:1f wan: configuring for phy/gmii link mode
[   34.293490] mt7530-mdio mdio-bus:1f wan: Link is Up - 100Mbps/Full - flow control off
[   34.943168] mt7530-mdio mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[   34.951113] br-lan: port 1(lan2) entered blocking state
[   34.956323] br-lan: port 1(lan2) entered forwarding state

hcym, 2024-07-14:
用hanw大佬的可以了，很稳定，温度也不高，油管三十万轻松自如

Turing-China, 2024-07-15:
> 用hanw大佬的可以了，很稳定，温度也不高，油管三十万轻松自如

你用这个L大的源码编译也是没有WIFI信号？

coolsnowwolf, 2026-09-05:
已核对。日志显示 mt798x-wmac 两个 PHY 均成功初始化并注册 LED，未出现 mt76/固件加载失败；当前 Redmi Router AX6000 DTS 与上游设备定义一致。评论中报告者已确认改用 hanw 版本后 Wi‑Fi 稳定，但未提供可复现的 UCI 无线配置、hostapd 日志或当前源码差异，因此无法定位为 LEDE 源码缺陷。请在现行源码、默认配置和完整 hostapd 日志下重新报告；本条关闭为 not planned。

closed 2026-09-05 by coolsnowwolf