coolsnowwolf/lede #12331: 6.1后多个版本异常重启频繁，尤其使用是腾讯会议时

clydee, 2024-07-15:
### 详细叙述

家里两个软路由，分别是N6005和N100，使用PVE运行，内核6.1后多个版本异常重启频繁，尤其使用是腾讯会议时，尝试重新编译过多个版本都是如此，已打微码补丁

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

N100，N6005

### 详细日志

[    0.000000] Linux version 6.1.97 (clyde@ClydeDebian) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6747-7ecb40bb3) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT_DYNAMIC Sat Jul 13 15:02:08 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=f0a29de5-9c6e-568c-ac03-a1141ad7d402 rootwait console=tty0 console=ttyS0,115200n8 noinitrd swiotlb=65536
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffd9fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffda000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.16.3-0-ga6ed6b701f0a-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000001] kvm-clock: using sched offset of 1127313450848177 cycles
[    0.000002] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000004] tsc: Detected 806.400 MHz processor
[    0.000462] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000464] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000467] last_pfn = 0x7ffda max_arch_pfn = 0x400000000
[    0.000508] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[    0.004079] found SMP MP-table at [mem 0x000f5450-0x000f545f]
[    0.004091] Using GB pages for direct mapping
[    0.004210] ACPI: Early table checksum verification disabled
[    0.004213] ACPI: RSDP 0x00000000000F5270 000014 (v00 BOCHS )
[    0.004216] ACPI: RSDT 0x000000007FFE2F2B 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004220] ACPI: FACP 0x000000007FFE2D15 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004224] ACPI: DSDT 0x000000007FFDF040 003CD5 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004226] ACPI: FACS 0x000000007FFDF000 000040
[    0.004228] ACPI: APIC 0x000000007FFE2D89 000078 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004229] ACPI: SSDT 0x000000007FFE2E01 0000CA (v01 BOCHS  VMGENID  00000001 BXPC 00000001)
[    0.004231] ACPI: HPET 0x000000007FFE2ECB 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004233] ACPI: WAET 0x000000007FFE2F03 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004234] ACPI: Reserving FACP table memory at [mem 0x7ffe2d15-0x7ffe2d88]
[    0.004235] ACPI: Reserving DSDT table memory at [mem 0x7ffdf040-0x7ffe2d14]
[    0.004236] ACPI: Reserving FACS table memory at [mem 0x7ffdf000-0x7ffdf03f]
[    0.004237] ACPI: Reserving APIC table memory at [mem 0x7ffe2d89-0x7ffe2e00]
[    0.004237] ACPI: Reserving SSDT table memory at [mem 0x7ffe2e01-0x7ffe2eca]
[    0.004238] ACPI: Reserving HPET table memory at [mem 0x7ffe2ecb-0x7ffe2f02]
[    0.004238] ACPI: Reserving WAET table memory at [mem 0x7ffe2f03-0x7ffe2f2a]
[    0.004271] Zone ranges:
[    0.004272]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.004273]   DMA32    [mem 0x0000000001000000-0x000000007ffd9fff]
[    0.004275]   Normal   empty
[    0.004275] Movable zone start for each node
[    0.004276] Early memory node ranges
[    0.004276]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.004277]   node   0: [mem 0x0000000000100000-0x000000007ffd9fff]
[    0.004278] Initmem setup node 0 [mem 0x0000000000001000-0x000000007ffd9fff]
[    0.004285] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.004310] On node 0, zone DMA: 97 pages in unavailable ranges
[    0.007727] On node 0, zone DMA32: 38 pages in unavailable ranges
[    0.008812] ACPI: PM-Timer IO Port: 0x608
[    0.008821] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
[    0.008859] IOAPIC[0]: apic_id 0, version 17, address 0xfec00000, GSI 0-23
[... 545 lines cut ...]
[    7.145482] usbcore: registered new interface driver ums-isd200
[    7.147061] usbcore: registered new interface driver ums-jumpshot
[    7.148611] usbcore: registered new interface driver ums-karma
[    7.150261] usbcore: registered new interface driver ums-sddr09
[    7.151848] usbcore: registered new interface driver ums-sddr55
[    7.153753] usbcore: registered new interface driver ums-usbat
[    7.158867] xt_time: kernel timezone is -0000
[    7.162472] usbcore: registered new interface driver aqc111
[    7.164034] usbcore: registered new interface driver ax88179_178a
[    7.173026] PPP generic driver version 2.4.2
[    7.174500] PPP MPPE Compression module registered
[    7.175912] NET: Registered PF_PPPOX protocol family
[    7.177888] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[    7.178813] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[    7.292959] [drm] amdgpu kernel modesetting enabled.
[    7.315356] kmodloader: done loading kernel modules from /etc/modules.d/*
[    9.567594] 8021q: adding VLAN 0 to HW filter on device eth0
[    9.568386] br-lan: port 1(eth0) entered blocking state
[    9.569050] br-lan: port 1(eth0) entered disabled state
[    9.569798] device eth0 entered promiscuous mode
[    9.570660] br-lan: port 1(eth0) entered blocking state
[    9.571405] br-lan: port 1(eth0) entered forwarding state
[   12.442507] igc 0000:00:10.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   13.042261] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   23.332293] pppoe-wan: renamed from ppp0

clydee, 2024-07-15:
另补上N6005的日志：
[    0.000000] Linux version 6.1.97 (clyde@ClydeDebian) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6747-7ecb40bb3) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT_DYNAMIC Sat Jul 13 15:02:08 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=f0a29de5-9c6e-568c-ac03-a1141ad7d402 rootwait console=tty0 console=ttyS0,115200n8 noinitrd swiotlb=65536
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003ffd9fff] usable
[    0.000000] BIOS-e820: [mem 0x000000003ffda000-0x000000003fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.16.3-0-ga6ed6b701f0a-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000001] kvm-clock: using sched offset of 1172346145184134 cycles
[    0.000003] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000006] tsc: Detected 1996.800 MHz processor
[    0.000556] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000559] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000563] last_pfn = 0x3ffda max_arch_pfn = 0x400000000
[    0.000606] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[    0.003121] found SMP MP-table at [mem 0x000f5440-0x000f544f]
[    0.003258] ACPI: Early table checksum verification disabled
[    0.003262] ACPI: RSDP 0x00000000000F5260 000014 (v00 BOCHS )
[    0.003265] ACPI: RSDT 0x000000003FFE2354 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003269] ACPI: FACP 0x000000003FFE213E 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003273] ACPI: DSDT 0x000000003FFDF040 0030FE (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003276] ACPI: FACS 0x000000003FFDF000 000040
[    0.003278] ACPI: APIC 0x000000003FFE21B2 000078 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003280] ACPI: SSDT 0x000000003FFE222A 0000CA (v01 BOCHS  VMGENID  00000001 BXPC 00000001)
[    0.003282] ACPI: HPET 0x000000003FFE22F4 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003284] ACPI: WAET 0x000000003FFE232C 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003285] ACPI: Reserving FACP table memory at [mem 0x3ffe213e-0x3ffe21b1]
[    0.003287] ACPI: Reserving DSDT table memory at [mem 0x3ffdf040-0x3ffe213d]
[    0.003288] ACPI: Reserving FACS table memory at [mem 0x3ffdf000-0x3ffdf03f]
[    0.003288] ACPI: Reserving APIC table memory at [mem 0x3ffe21b2-0x3ffe2229]
[    0.003289] ACPI: Reserving SSDT table memory at [mem 0x3ffe222a-0x3ffe22f3]
[    0.003290] ACPI: Reserving HPET table memory at [mem 0x3ffe22f4-0x3ffe232b]
[    0.003290] ACPI: Reserving WAET table memory at [mem 0x3ffe232c-0x3ffe2353]
[    0.003325] Zone ranges:
[    0.003326]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.003328]   DMA32    [mem 0x0000000001000000-0x000000003ffd9fff]
[    0.003329]   Normal   empty
[    0.003330] Movable zone start for each node
[    0.003331] Early memory node ranges
[    0.003331]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.003332]   node   0: [mem 0x0000000000100000-0x000000003ffd9fff]
[    0.003333] Initmem setup node 0 [mem 0x0000000000001000-0x000000003ffd9fff]
[    0.003337] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.003364] On node 0, zone DMA: 97 pages in unavailable ranges
[    0.005198] On node 0, zone DMA32: 38 pages in unavailable ranges
[    0.006143] ACPI: PM-Timer IO Port: 0x608
[    0.006152] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
[    0.006192] IOAPIC[0]: apic_id 0, version 17, address 0xfec00000, GSI 0-23
[    0.006194] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.006196] ACPI: INT_SRC_OVR (bus 0 bus_irq 5 global_irq 5 high level)
[    0.006197] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.006198] ACPI: INT_SRC_OVR (bus 0 bus_irq 10 global_irq 10 high level)
[    0.006199] ACPI: INT_SRC_OVR (bus 0 bus_irq 11 global_irq 11 high level)
[    0.006201] ACPI: Using ACPI (MADT) for SMP configuration information
[    0.006202] ACPI: HPET id: 0x8086a201 base: 0xfed00000
[    0.006206] TSC deadline timer available
[    0.006207] smpboot: Allowing 1 CPUs, 0 hotplug CPUs
[    0.006230] [mem 0x40000000-0xfeffbfff] available for PCI devices
[    0.006232] Booting paravirtualized kernel on KVM
[    0.006234] clocksource: refined-jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.006238] setup_percpu: NR_CPUS:512 nr_cpumask_bits:1 nr_cpu_ids:1 nr_node_ids:1
[    0.006481] percpu: Embedded 47 pages/cpu s151784 r8192 d32536 u2097152
[... 498 lines cut ...]
[    7.094939] usbcore: registered new interface driver aqc111
[    7.096897] usbcore: registered new interface driver ax88179_178a
[    7.107133] PPP generic driver version 2.4.2
[    7.108956] PPP MPPE Compression module registered
[... cut ...]
[    9.336066] device eth2 entered promiscuous mode
[    9.384549] br-lan: port 3(eth3) entered blocking state
[    9.385257] br-lan: port 3(eth3) entered disabled state
[    9.385896] device eth3 entered promiscuous mode
[   10.359807] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   12.336651] igc 0000:00:10.0 eth1: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   12.934445] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   23.154367] pppoe-wan: renamed from ppp0

yanxin152133, 2024-07-15:
https://github.com/coolsnowwolf/lede/issues/8781

clydee, 2024-07-16:
N100也存在同样问题，不止N6005

> #8781

clydee, 2024-07-16:
> 日志中并没有有效信息，建议尝试： 在 OpenWRT 中禁用 kernel Panic 后的自动重启
> 
> ```
> sysctl -w kernel.panic=0
> ```
> 
> 发生异常时，进 PVE 截图虚拟机显示画面，应该包含 kernel Panic 日志

尝试了下，系统直接卡死，PVE中op虚机没任何反应了，没显示panic日志，也没法进行任何交互和反应，点击重启/关机也没任何反应，只能登录到pve中kill掉对应kvm进程

clydee, 2024-07-17:
再发一个日志以便排查

[    0.000000] Linux version 6.1.98 (clyde@ClydeDebian) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6747-7ecb40bb3) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT_DYNAMIC Mon Jul 15 14:06:06 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=f0a29de5-9c6e-568c-ac03-a1141ad7d402 rootwait console=tty0 console=ttyS0,115200n8 noinitrd swiotlb=65536 crashkernel=128M
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bffd9fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bffda000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.16.3-0-ga6ed6b701f0a-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000001] kvm-clock: using sched offset of 711564700764 cycles
[    0.000003] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000006] tsc: Detected 1996.800 MHz processor
[    0.000548] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000550] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000554] last_pfn = 0x140000 max_arch_pfn = 0x400000000
[    0.000596] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- UC  
[    0.000602] last_pfn = 0xbffda max_arch_pfn = 0x400000000
[    0.003144] found SMP MP-table at [mem 0x000f5440-0x000f544f]
[    0.003347] ACPI: Early table checksum verification disabled
[    0.003351] ACPI: RSDP 0x00000000000F5240 000014 (v00 BOCHS )
[    0.003354] ACPI: RSDT 0x00000000BFFE23AF 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003359] ACPI: FACP 0x00000000BFFE2191 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003363] ACPI: DSDT 0x00000000BFFDF040 003151 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003365] ACPI: FACS 0x00000000BFFDF000 000040
[    0.003367] ACPI: APIC 0x00000000BFFE2205 000080 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003369] ACPI: SSDT 0x00000000BFFE2285 0000CA (v01 BOCHS  VMGENID  00000001 BXPC 00000001)
[    0.003371] ACPI: HPET 0x00000000BFFE234F 000038 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003373] ACPI: WAET 0x00000000BFFE2387 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.003375] ACPI: Reserving FACP table memory at [mem 0xbffe2191-0xbffe2204]
[    0.003376] ACPI: Reserving DSDT table memory at [mem 0xbffdf040-0xbffe2190]
[    0.003377] ACPI: Reserving FACS table memory at [mem 0xbffdf000-0xbffdf03f]
[    0.003378] ACPI: Reserving APIC table memory at [mem 0xbffe2205-0xbffe2284]
[    0.003378] ACPI: Reserving SSDT table memory at [mem 0xbffe2285-0xbffe234e]
[    0.003379] ACPI: Reserving HPET table memory at [mem 0xbffe234f-0xbffe2386]
[    0.003380] ACPI: Reserving WAET table memory at [mem 0xbffe2387-0xbffe23ae]
[    0.003412] Reserving 128MB of memory at 2928MB for crashkernel (System RAM: 4095MB)
[    0.003420] Zone ranges:
[    0.003421]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.003422]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.003424]   Normal   [mem 0x0000000100000000-0x000000013fffffff]
[    0.003425] Movable zone start for each node
[    0.003425] Early memory node ranges
[    0.003426]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.003427]   node   0: [mem 0x0000000000100000-0x00000000bffd9fff]
[    0.003428]   node   0: [mem 0x0000000100000000-0x000000013fffffff]
[    0.003429] Initmem setup node 0 [mem 0x0000000000001000-0x000000013fffffff]
[    0.003436] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.003462] On node 0, zone DMA: 97 pages in unavailable ranges
[    0.011409] On node 0, zone Normal: 38 pages in unavailable ranges
[    0.012342] ACPI: PM-Timer IO Port: 0x608
[    0.012352] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
[    0.012391] IOAPIC[0]: apic_id 0, version 17, address 0xfec00000, GSI 0-23
[    0.012394] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[    0.012395] ACPI: INT_SRC_OVR (bus 0 bus_irq 5 global_irq 5 high level)
[    0.012397] ACPI: INT_SRC_OVR (bus 0 bus_irq 9 global_irq 9 high level)
[    0.012397] ACPI: INT_SRC_OVR (bus 0 bus_irq 10 global_irq 10 high level)
[    0.012398] ACPI: INT_SRC_OVR (bus 0 bus_irq 11 global_irq 11 high level)
[    0.012401] ACPI: Using ACPI (MADT) for SMP configuration information
[    0.012402] ACPI: HPET id: 0x8086a201 base: 0xfed00000
[    0.012406] smpboot: Allowing 2 CPUs, 0 hotplug CPUs
[    0.012424] [mem 0xc0000000-0xfeffbfff] available for PCI devices
[... 592 lines cut ...]
[  205.137416]  asm_sysvec_call_function_single+0x1b/0x20
[  205.137931] RIP: 0010:native_safe_halt+0x13/0x20
[... cut ...] net ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ixgbe igbvf e1000 amd_xgbe mdio_devres ip6_udp_tunnel udp_tunnel mdio tun nls_utf8 ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic authencesn authenc arc4 nls_iso8859_1 nls_cp437 uas sdhci_pltfm igb vfat fat exfat
[  205.156956]  button_hotplug tg3 mii libphy
[  205.165317] CR2: 00000000000000b2

aiamadeus, 2024-07-17:
这个是上游的问题吧。。
这样子，删掉
target/linux/generic/hack-6.1/600-net-enable-fraglist-GRO-by-default.patch
target/linux/generic/hack-6.6/600-net-enable-fraglist-GRO-by-default.patch
target/linux/generic/pending-6.1/680-net-add-TCP-fraglist-GRO-support.patch
target/linux/generic/pending-6.6/680-net-add-TCP-fraglist-GRO-support.patch

clydee, 2024-07-19:
> 这个是上游的问题吧。。 这样子，删掉下面这些补丁再编译看下 target/linux/generic/hack-6.1/600-net-enable-fraglist-GRO-by-default.patch target/linux/generic/hack-6.6/600-net-enable-fraglist-GRO-by-default.patch target/linux/generic/pending-6.1/680-net-add-TCP-fraglist-GRO-support.patch target/linux/generic/pending-6.6/680-net-add-TCP-fraglist-GRO-support.patch

感谢感谢，删除后重新编译确实可以解决问题，不知道什么时候可以在op上正常解决

aiamadeus, 2024-07-19:
> 我看官方说修复了这个问题，是不是补丁没同步过来？我对内核这块不是很了解。

不是一个问题，6.1.92 和 6.6.31 已经合并上面的补丁了

clydee, 2024-07-21:
> Netcore N60 (512MB内存)，试了刷6.1.98内核还是会重启。

删除aiamadeus提到的那四个文件编译后的情况嘛

clydee, 2024-07-22:
> > > Netcore N60 (512MB内存)，试了刷6.1.98内核还是会重启。
> > 
> > 
> > 删除上面提到的那四个文件编译后的情况嘛
> 
> 那几个文件是干什么用的，删掉会不会造成别的影响，我试试

看名字是增加GRO支持的，当前我试用能解决问题，还需要大佬看后续版本如何操作 @coolsnowwolf

Wilson-coder, 2024-07-22:
> 这个是上游的问题吧。。 这样子，删掉下面这些补丁再编译看下 target/linux/generic/hack-6.1/600-net-enable-fraglist-GRO-by-default.patch target/linux/generic/hack-6.6/600-net-enable-fraglist-GRO-by-default.patch [...]

删除这几个补丁后，重新编译6.1.98，已经用了两天半没再重启了。之前的编译基本坚持不到24小时必定自动重启。

clydee, 2024-07-22:
> 删除这几个补丁后，重新编译6.1.98，已经用了两天半没再重启了。之前的编译基本坚持不到24小时必定自动重启。

我也是6.1.98，之前同样的源码最频繁2分钟重启一次，现在稳定运行3d10h

xixijojo, 2024-07-27:
6.1.100目前运行两天了，没有自动重启

closed 2024-07-29 by aiamadeus

gg99yl, 2025-03-10:
> 删除这几个补丁后，重新编译6.1.98，已经用了两天半没再重启了。之前的编译基本坚持不到24小时必定自动重启。

请问具体怎样删除，重新编译呀？

xianren78, 2025-03-20:
我在PVE8.3上运行官方op（自己打了其他补丁）时遇到微信视频或者tailscale时，遇到这个问题，经排查与 
hack-6.12/600-net-enable-fraglist-GRO-by-default.patch 这个补丁有关，删除即可已解决。

commit deb218ea (2024-07-29): generic: 6.1/6.6: remove gro improvement patch

Avoid unstable situations, closed: #12331