coolsnowwolf/lede #12334: ax6s wifi只有2.5G 有线访问不了路由器界面

lztxi, 2024-07-15:
### 详细叙述

有线可以上网，获取的ip段是光猫的，路由没有分配ip
wifi 只有2.5G，没有5G

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Redmi Router AX6S

### 详细日志

Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.787010] 0x000000000000-0x000000080000 : "Preloader"
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.793129] 0x000000080000-0x0000000c0000 : "ATF"
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.798343] 0x0000000c0000-0x000000140000 : "u-boot"
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.804113] 0x000000140000-0x000000180000 : "u-boot-env"
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.809942] 0x000000180000-0x0000001c0000 : "bdata"
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.815368] 0x0000001c0000-0x000000240000 : "factory"
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.821293] 0x000000240000-0x000000280000 : "crash"
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.826700] 0x000000280000-0x0000002c0000 : "crash_log"
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.832534] 0x0000002c0000-0x0000006c0000 : "kernel"
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    0.842222] 0x0000006c0000-0x0000075c0000 : "ubi"
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.103894] mtk_soc_eth 1b100000.ethernet eth0: mediatek frame engine at 0xffffffc009260000, irq 123
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.113719] i2c_dev: i2c /dev entries driver
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.118514] mtk-wdt 10212000.watchdog: Watchdog enabled (timeout=31 sec, nowayout=0)
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.128286] NET: Registered PF_INET6 protocol family
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.134433] Segment Routing with IPv6
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.138112] In-situ OAM (IOAM) with IPv6
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.142102] NET: Registered PF_PACKET protocol family
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.147277] 8021q: 802.1Q VLAN Support v1.8
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.152867] pstore: Using crash dump compression: deflate
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.171431] mtk-pcie 1a143000.pcie: host bridge /pcie@1a143000 ranges:
Sun Jul 14 23:15:25 2024 kern.debug kernel: [    1.177987] mtk-pcie 1a143000.pcie: Parsing ranges property...
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.183884] mtk-pcie 1a143000.pcie:      MEM 0x0020000000..0x0027ffffff -> 0x0020000000
Sun Jul 14 23:15:25 2024 kern.err kernel: [    1.191919] mtk-pcie 1a143000.pcie: missing hifsys node
Sun Jul 14 23:15:25 2024 kern.warn kernel: [    1.197144] mtk-pcie: probe of 1a143000.pcie failed with error -22
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    1.203813] mtk_hsdma 1b007000.dma-controller: MediaTek HSDMA driver registered
Sun Jul 14 23:15:25 2024 kern.err kernel: [    2.216802] mt7530-mdio mdio-bus:00: reset timeout
Sun Jul 14 23:15:25 2024 kern.warn kernel: [    2.221625] mt7530-mdio: probe of mdio-bus:00 failed with error -110
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.228348] UBI: auto-attach mtd9
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.231678] ubi0: default fastmap pool size: 40
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.236201] ubi0: default fastmap WL pool size: 20
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.240989] ubi0: attaching mtd9
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.511298] ubi0: scanning is finished
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.519457] ubi0: attached mtd9 (name "ubi", size 111 MiB)
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.524962] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.531838] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.538619] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.545581] ubi0: good PEBs: 888, bad PEBs: 0, corrupted PEBs: 0
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.551584] ubi0: user volume: 2, internal volumes: 1, max. volumes count: 128
] ubi0: max/mean erase counter: 66/46, WL threshold: 4096, image sequence number: 1714977313
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.568099] ubi0: available PEBs: 0, total reserved PEBs: 888, PEBs reserved for bad PEB handling: 19
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.577320] ubi0: background thread "ubi_bgt0d" started, PID 387
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    2.584368] block ubiblock0_0: created from ubi0:0(rootfs)
Sun Jul 14 23:15:25 2024 kern.notice kernel: [    2.589855] ubiblock: device ubiblock0_0 (rootfs) set to be root filesystem
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    2.596911] clk: Disabling unused clocks
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    2.605046] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    2.612412] Freeing unused kernel memory: 448K
Sun Jul 14 23:15:25 2024 [kern.info](https://kern.info/) kernel: [    2.616902] Run /sbin/init as init process
Sun Jul 14 23:15:25 2024 kern.debug kernel: [    2.620997]   with arguments:
Sun Jul 14 23:15:25 2024 kern.debug kernel: [    2.623955]     /sbin/init
Sun Jul 14 23:15:25 2024 kern.debug kernel: [    2.626651]   with environment:
Sun Jul 14 23:15:25 2024 kern.debug kernel: [    2.629781]     HOME=/
Sun Jul 14 23:15:25 2024 kern.debug kernel: [    2.632136]     TERM=linux
Sun Jul 14 23:15:25 2024 kern.debug kernel: [    2.634832]     firmware=0
Sun Jul 14 23:15:25 2024 kern.debug kernel: [    2.637529]     uart_en=1
Sun Jul 14 23:15:25 2024 [user.info](https://user.info/) kernel: [    3.166916] init: Console is alive
[... 117 lines cut ...]
Sun Jul 14 23:15:35 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e8:f7:91:11:86:14 auth_alg=open
Sun Jul 14 23:15:35 2024 [daemon.info](https://daemon.info/) hostapd: wlan0: STA e8:f7:91:11:86:14 RADIUS: starting accounting session 9CA7D84DF05E1DB9
Sun Jul 14 23:15:41 2024 daemon.notice procd: /etc/rc.d/S99openclash: OpenClash Already Stop!
Sun Jul 14 23:15:41 2024 cron.err crond[4817]: crond (busybox 1.36.0) started, log level 5
Sun Jul 14 23:15:41 2024 [daemon.info](https://daemon.info/) procd: - init complete -
Sun Jul 14 23:15:41 2024 [daemon.info](https://daemon.info/) urandom_seed[4867]: Seed saved (/etc/urandom.seed)
Sun Jul 14 23:20:10 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
Sun Jul 14 23:20:10 2024 daemon.err uhttpd[1746]: awk: cmd. line:1: Unexpected end of string
Sun Jul 14 23:20:13 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
Sun Jul 14 23:20:13 2024 daemon.err uhttpd[1746]: awk: cmd. line:1: Unexpected end of string
Sun Jul 14 23:20:15 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
Sun Jul 14 23:20:15 2024 daemon.err uhttpd[1746]: awk: cmd. line:1: Unexpected end of string
Sun Jul 14 23:20:18 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
Sun Jul 14 23:20:18 2024 daemon.err uhttpd[1746]: awk: cmd. line:1: Unexpected end of string
Sun Jul 14 23:20:20 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
Sun Jul 14 23:20:20 2024 daemon.err uhttpd[1746]: awk: cmd. line:1: Unexpected end of string
Sun Jul 14 23:20:23 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
Sun Jul 14 23:20:23 2024 daemon.err uhttpd[1746]: awk: cmd. line:1: Unexpected end of string
Sun Jul 14 23:20:25 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
Sun Jul 14 23:20:25 2024 daemon.err uhttpd[1746]: awk: cmd. line:1: Unexpected end of string
Sun Jul 14 23:20:28 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
Sun Jul 14 23:20:28 2024 daemon.err uhttpd[1746]: awk: cmd. line:1: Unexpected end of string
Sun Jul 14 23:20:30 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp': No such file or directory
Sun Jul 14 23:20:30 2024 daemon.err uhttpd[1746]: awk: cmd. line:1: Unexpected end of string
Sun Jul 14 23:20:33 2024 daemon.err uhttpd[1746]: cat: can't open '/sys/class/thermal/thermal_zone0/temp':

kylongteng, 2024-08-03:
你的问题我也遇到了，我在https://github.com/coolsnowwolf/lede/issues/12305时遇到的是wan口有问题，后来发现lan口也有问题，就是无法识别网卡。前两天我又编译了一次，问题多出了一个和你一样的，wifi只有2.4g，5g的无线接口都没了。提了issue但估计没看就close了

453974158, 2024-10-14:
我也遇到了一样的问题。

coolsnowwolf, 2026-09-05:
已核对该日志：AX6S 的两个关键故障分别是 PCIe 报 `missing hifsys node` 导致 5 GHz 无线 PCIe 设备未探测，以及 MT7531 使用错误 MDIO 地址导致交换机探测失败，所以会同时表现为只有 2.4 GHz、网口无法访问路由器管理地址。当前 master 已包含 commit `e66721c4e321e1a71702bbe1afe10b322fa97b0f`（`mediatek: fix Redmi AX6S hardware initialization`）：将交换机地址修正为 `<31>`，并补齐两路无线 nvmem 校准区域，配合现有 PCIe hifsys 支持。该 issue 的修复已在当前代码中，关闭。

closed 2026-09-05 by coolsnowwolf