coolsnowwolf/lede #12339: 云编译固件问题，luci分支openwrt23.05

lanruihua026, 2024-07-16:
### 详细叙述

编译固件时发现lean源码luci主分支版本过老，遂切换到luci的openwrt23.05分支，编译过程中即使去掉luci-app-vsftpd这个插件，仍然报这个错误，请教一下各位大佬应如何解决，十分感谢！
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

FastRhino R68s

### 详细日志

Collected errors:
Configuring uhttpd-mod-ubus.
 * check_data_file_clashes: Package vsftpd wants to install file /workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip/etc/init.d/vsftpd
	But that file is already provided by package  * vsftpd-alt
 * check_data_file_clashes: Package vsftpd wants to install file /workdir/openwrt/build_dir/target-aarch64_generic_musl/root-rockchip/usr/sbin/vsftpd
	But that file is already provided by package  * vsftpd-alt
 * opkg_install_cmd: Cannot install package vsftpd.

yanxin152133, 2024-07-17:
vsftpd-alt  vsftpd冲突？

lanruihua026, 2024-07-17:
> vsftpd-alt vsftpd冲突？

[image]
昨天重新把配置文件删除掉，只选了几个插件编译就通过了，感觉是某些插件会存在连带冲突，但是没办法直接从日志上看出来，只能一个个试

lanruihua026, 2024-07-17:
[image]

kxdn3, 2024-07-20:
请教云编译时候怎么切换到23.05的？

lanruihua026, 2024-07-20:
> 请教云编译时候怎么切换到23.05的？

[image]
Lede的feed default做如下修改，然后连同.config文件一同上传到GitHub上，再用action进行云编译

lanruihua026, 2024-07-20:
https://github.com/P3TERX/Actions-OpenWrt
我用的这个仓库来进行编译

closed 2024-07-21 by lanruihua026