coolsnowwolf/lede #12343: error: ISO C90 forbids mixed declarations and code [-Werror=declaration-after-statement]    u8 qid = tid_to_ac[tid];

dann2333, 2024-07-19:
### 详细叙述

小米路由器3G-MT7621A无法编译，已git pull过最新版本源代码，似乎是最近更新才出的问题。麻烦帮忙看一下，谢谢！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

小米路由器3G-MT7621A

### 详细日志

```
/home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c/mt7603/dma.c: In function 'mt7603_rx_loopback_skb':
/home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c/mt7603/dma.c:60:3: error: ISO C90 forbids mixed declarations and code [-Werror=declaration-after-statement]
   u8 qid = tid_to_ac[tid];
   ^~
cc1: all warnings being treated as errors
make[6]: *** [scripts/Makefile.build:286: /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c/mt7603/dma.o] Error 1
make[5]: *** [scripts/Makefile.build:503: /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c/mt7603] Error 2
make[5]: *** Waiting for unfinished jobs....
  CC [M]  /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c/mt76x2/pci_mcu.o
  CC [M]  /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c/mt76x2/pci_phy.o
  LD [M]  /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c/mt76x2/mt76x2-common.o
  LD [M]  /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c/mt76x2/mt76x2e.o
make[4]: *** [Makefile:1832: /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.220'
make[3]: *** [Makefile:651: /home/ubuntu/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-04-04-8f301a5c/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#59.09#11.32#38.99
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [package/Makefile:110: /home/ubuntu/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:231: world] Error 2

```

dann2333, 2024-07-21:
https://github.com/openwrt/mt76/commit/513c131c6309712a51502870b041f45b4bd6a6d4

aiamadeus, 2024-07-22:
```
--- a/package/kernel/mt76/Makefile
+++ b/package/kernel/mt76/Makefile
@@ -14,7 +14,7 @@ PKG_SOURCE_DATE:=2022-12-22
 PKG_SOURCE_VERSION:=5b509e80384ab019ac11aa90c81ec0dbb5b0d7f2
 PKG_MIRROR_HASH:=6fc25df4d28becd010ff4971b23731c08b53e69381a9e4c868091899712f78a9
 PATCH_DIR:=./patches-5.4
-else ifndef ($(filter on,$(CONFIG_LINUX_6_1) $(CONFIG_LINUX_6_6)),)
+else ifndef ($(filter y,$(CONFIG_LINUX_6_1) $(CONFIG_LINUX_6_6)),)
 PKG_SOURCE_DATE:=2024-04-04
 PKG_SOURCE_VERSION:=8f301a5c5fe3d998b50666bde1d35d412833be89
 PKG_MIRROR_HASH:=49291145ccceeac2964c9a43a46bed88298e5c1da63c6f3c2c1f6b3c1902a61e
```
改成这样试一下，那么新的 5.10 编不过的

dann2333, 2024-07-22:
> ```
> --- a/package/kernel/mt76/Makefile
> +++ b/package/kernel/mt76/Makefile
> @@ -14,7 +14,7 @@ PKG_SOURCE_DATE:=2022-12-22
>  PKG_SOURCE_VERSION:=5b509e80384ab019ac11aa90c81ec0dbb5b0d7f2 [...]

手写了个patch，但是检查不通过，就没提pr，大佬看看改改合并一下？
来自 https://github.com/openwrt/mt76/commit/513c131c6309712a51502870b041f45b4bd6a6d4
`--- a/mt7925/main.c
+++ b/mt7925/main.c
@@ -260,7 +260,8 @@ __mt7925_set_stream_he_eht_caps(struct m
 		n++;
 	}
 
-	_ieee80211_set_sband_iftype_data(sband, data, n);
+	sband->iftype_data = data;
+	sband->n_iftype_data = n;
 }
 
 void mt7925_set_stream_he_eht_caps(struct mt792x_phy *phy)
--- a/mt7925/mcu.c
+++ b/mt7925/mcu.c
@@ -2209,7 +2209,7 @@ mt7925_mcu_bss_mld_tlv(struct sk_buff *s
 		       struct ieee80211_sta *sta)
 {
 	struct mt792x_vif *mvif = (struct mt792x_vif *)vif->drv_priv;
-	bool is_mld = ieee80211_vif_is_mld(vif);
+	bool is_mld = (vif->valid_links != 0);
 	struct bss_mld_tlv *mld;
 	struct tlv *tlv;
 `

aiamadeus, 2024-07-22:
不行，5.10 不能用这么新的 mt76
那样子 7615/7915 可能没法用

dann2333, 2024-07-22:
> 不行，5.10 不能用这么新的 mt76 那样子 7615/7915 可能没法用

实际上只改了两行，我写了一个patch正在测试，一会提pr大佬看看

dann2333, 2024-07-22:
https://github.com/coolsnowwolf/lede/pull/12353

commit 150b7452 (2024-07-22): mt76: fixes build for older kernel

Fixes: #12343
closed 2024-07-22 by aiamadeus