coolsnowwolf/lede #12345: bpi-r3编译docker错误

dd-bpir3, 2024-07-20:
### 详细叙述

bpi-r3编译docker错误去掉docker插件可以编译成功

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

bpi-r3

### 详细日志

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency iptables-mod-fullconenat for firewall
 * pkg_hash_fetch_best_installation_candidate: Packages for firewall found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	iptables-mod-fullconenat
 * opkg_install_cmd: Cannot install package default-settings.
 * check_conflicts_for: The following packages conflict with firewall:
 * check_conflicts_for: 	firewall4 * 
 * opkg_install_cmd: Cannot install package firewall.
 * check_data_file_clashes: Package luci-app-docker wants to install file /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/config/dockerd
	But that file is already provided by package  * dockerd
 * check_data_file_clashes: Package luci-app-docker wants to install file /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/init.d/dockerd
	But that file is already provided by package  * dockerd
 * opkg_install_cmd: Cannot install package luci-app-docker.
 * check_data_file_clashes: Package luci-app-docker wants to install file /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/config/dockerd
	But that file is already provided by package  * dockerd
 * check_data_file_clashes: Package luci-app-docker wants to install file /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/init.d/dockerd
	But that file is already provided by package  * dockerd
 * opkg_install_cmd: Cannot install package luci-i18n-docker-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/yph/lede'
make[1]: *** [package/Makefile:111: /home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/yph/lede'
make: *** [/home/yph/lede/include/toplevel.mk:231: world] Error 2

yanxin152133, 2024-07-29:
make clean 之后呢

dd-bpir3, 2024-07-29:
> * luci-app-docker和[luci-app-dockerman冲突了，之前可以一起编译。
make dirclean
rm -rf ./tmp && rm -rf .config
编译中

coolsnowwolf, 2026-09-05:
当前代码已不复现日志中的两个冲突：`luci-app-docker` 目前安装的是 `/etc/config/docker`、`/etc/init.d/docker`，而 `dockerd` 包安装的是 `/etc/config/dockerd`、`/etc/init.d/dockerd`，不再争用同一路径；旧日志对应的 packages/LuCI 版本已过时。`iptables-mod-fullconenat` 依赖则属于旧 firewall3/iptables 配置，当前 BPi-R3 使用 firewall4/nftables。Issue 已过时，关闭。

closed 2026-09-05 by coolsnowwolf