coolsnowwolf/lede #12346: H68K外接硬盘导致系统卡死

YHJMM-303, 2024-07-20:
### 详细叙述

H68K连接sata转usb的硬盘盒，插上后不到几秒，网口停止通信，整个系统卡死

外接硬盘的情况下，机器断电后无法启动，断开硬盘后断电正常重启
使用其它usb储存设备运行正常，如 U盘或读卡器
无法捕捉到接上硬盘后的系统日志和内核日志（自定义脚本也无法捕捉到）

只连接硬盘盒（硬盘盒不插硬盘），运行正常，磁盘管理能识别到
已确认硬盘未损坏
尝试换大约一个多月前编译的固件，运行正常

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

H68K-PRO  2023

### 详细日志

config文件
尝试编译进所有 kmod-ata-的包，无效
[config.txt](https://github.com/user-attachments/files/16319896/config.txt)

aiamadeus, 2024-07-20:
你 hdmi 接个显示器康康日志

YHJMM-303, 2024-07-20:
> 你 hdmi 接个显示器康康日志

hdmi无信号

aiamadeus, 2024-07-20:
选上 kmod-drm-rockchip

YHJMM-303, 2024-07-20:
> 选上 kmod-drm-rockchip

不行，还是没有视频输出
把video support里的都选上依旧没有视频输出

aiamadeus, 2024-07-20:
那不插硬盘时候的日志呢

YHJMM-303, 2024-07-20:
[Uploading kernel.log…]()

YHJMM-303, 2024-07-20:
[kernel.log](https://github.com/user-attachments/files/16321352/kernel.log)

YHJMM-303, 2024-07-20:
有了新发现，插usb2.0的接口没有此问题

aiamadeus, 2024-07-20:
```
[    2.889896] rockchip-drm display-subsystem: bound fe040000.vop (ops vop2_component_ops [rockchipdrm])
[    2.894642] dwhdmi-rockchip fe0a0000.hdmi: Detected HDMI TX controller v2.11a with HDCP (DWC HDMI 2.0 TX PHY)
[    2.895866] dwhdmi-rockchip fe0a0000.hdmi: registered DesignWare HDMI I2C bus driver
[    2.897018] rockchip-drm display-subsystem: bound fe0a0000.hdmi (ops vop_component_ops [rockchipdrm])
[    2.898126] [drm] Initialized rockchip 1.0.0 20140818 for display-subsystem on minor 0
[    2.898893] rockchip-drm display-subsystem: [drm] Cannot find any crtc or sizes
[    2.899603] rockchip-drm display-subsystem: [drm] Cannot find any crtc or sizes
[    2.900268] rockchip-drm display-subsystem: [drm] Cannot find any crtc or sizes
```
有驱动啊，不插硬盘接 HDMI 没输出？

YHJMM-303, 2024-07-20:
> ```
> [    2.889896] rockchip-drm display-subsystem: bound fe040000.vop (ops vop2_component_ops [rockchipdrm])
> [    2.894642] dwhdmi-rockchip fe0a0000.hdmi: Detected HDMI TX controller v2.11a with [...]

确认没有

YHJMM-303, 2024-07-20:
> ```
> [    2.889896] rockchip-drm display-subsystem: bound fe040000.vop (ops vop2_component_ops [rockchipdrm])
> [    2.894642] dwhdmi-rockchip fe0a0000.hdmi: Detected HDMI TX controller v2.11a with [...]

重启会闪一下黑屏，紧接着继续无信号

ufk119, 2024-07-21:
R5S也有这个问题，在6.1.97版本没问题，98版本就有这个问题了。 具体是带某种功能的硬盘盒一插上必死机，只有拔电重启能恢复。我接一般的usb3的移动硬盘盒没问题，但是接一个带自动休眠的功能多点的硬盘盒就完蛋。

YHJMM-303, 2024-07-22:
> R5S也有这个问题，在6.1.97版本没问题，98版本就有这个问题了。 具体是带某种功能的硬盘盒一插上必死机，只有拔电重启能恢复。我接一般的usb3的移动硬盘盒没问题，但是接一个带自动休眠的功能多点的硬盘盒就完蛋。

试了下，换成6.6的内核就正常了，hdmi也有输出

aiamadeus, 2024-07-23:
6.1.100 已经修复了

closed 2024-07-24 by YHJMM-303

ufk119, 2024-07-25:
换6.1.100 正常。2024年7月24日编译的。