coolsnowwolf/lede #12349: x86_64增加Utilities -->apk编译时出错

iHub-2020, 2024-07-21:
### 详细叙述

主要目的是为了安装openjdk，于是在编译的时候增加了Utilities -->apk，于是出现了报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

make[3]: Entering directory '/home/openwrt/lede/feeds/packages/utils/apk'
rm -f /home/openwrt/lede/build_dir/target-x86_64_musl/apk-tools-v2.12.4/.built
touch /home/openwrt/lede/build_dir/target-x86_64_musl/apk-tools-v2.12.4/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/openwrt/lede/build_dir/target-x86_64_musl/apk-tools-v2.12.4=apk-tools-v2.12.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/openwrt/lede/build_dir/target-x86_64_musl/apk-tools-v2.12.4=apk-tools-v2.12.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/openwrt/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro " make  -C /home/openwrt/lede/build_dir/target-x86_64_musl/apk-tools-v2.12.4/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/openwrt/lede/build_dir/target-x86_64_musl/apk-tools-v2.12.4=apk-tools-v2.12.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" LUA=no ;
make[4]: Entering directory '/home/openwrt/lede/build_dir/target-x86_64_musl/apk-tools-v2.12.4'
  GENHELP src/help.h
ln -sf libapk.so.3.12.0 src/libapk.so
  CC      src/database.o
src/database.c: In function 'find_mountpoint':
src/database.c:1371:23: error: storage size of 'st' isn't known
 1371 |         struct stat64 st;
      |                       ^~
src/database.c:1376:13: error: implicit declaration of function 'fstatat64'; did you mean 'fstatat'? [-Werror=implicit-function-declaration]
 1376 |         if (fstatat64(atfd, rel_path, &st, 0) != 0)
      |             ^~~~~~~~~
      |             fstatat
src/database.c:1371:23: error: unused variable 'st' [-Werror=unused-variable]
 1371 |         struct stat64 st;
      |                       ^~
cc1: all warnings being treated as errors
make[5]: *** [Make.rules:221: src/database.o] Error 1
make[4]: *** [Make.rules:337: src/] Error 2
make[4]: Leaving directory '/home/openwrt/lede/build_dir/target-x86_64_musl/apk-tools-v2.12.4'
make[3]: *** [Makefile:76: /home/openwrt/lede/build_dir/target-x86_64_musl/apk-tools-v2.12.4/.built] Error 2
make[3]: Leaving directory '/home/openwrt/lede/feeds/packages/utils/apk'
time: package/feeds/packages/apk/compile#0.20#0.04#0.24
    ERROR: package/feeds/packages/apk failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/apk/compile] Error 1
make[2]: Leaving directory '/home/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
make: *** [/home/openwrt/lede/include/toplevel.mk:231: world] Error 2
openwrt@reyanmatic:~/lede$

yanxin152133, 2024-07-24:
https://blog.csdn.net/x356982611/article/details/71079937
试试这个

bzi-bit, 2024-07-25:
直接把c库从musl调成glibc，下载安装正常版本的就好了