coolsnowwolf/lede #12350: 树莓派5 编译gobinet出错

qingshutie, 2024-07-21:
### 详细叙述

树莓派5，编译时添加移远5g模组支持，gobinet-1.63编译出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

树莓派5

### 详细日志

make[4]: Entering directory '/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/linux-6.1.98'
  CC [M]  /home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.o
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c: In function 'bridge_arp_reply':
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c:309:13: error: implicit declaration of function 'netif_rx_ni'; did you mean 'netif_rx'? [-Werror=implicit-function-declaration]
  309 |             netif_rx_ni(reply);
      |             ^~~~~~~~~~~
      |             netif_rx
In file included from ./include/linux/string.h:293,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
                 from ./include/linux/smp.h:13,
                 from ./arch/arm64/include/asm/arch_timer.h:18,
                 from ./arch/arm64/include/asm/timex.h:8,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c:55:
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c: In function 'qmap_register_device':
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c:947:21: warning: passing argument 1 of '__builtin_memcpy' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  947 |     memcpy (qmap_net->dev_addr, real_dev->dev_addr, ETH_ALEN);
      |             ~~~~~~~~^~~~~~~~~~
./include/linux/fortify-string.h:469:27: note: in definition of macro '__fortify_memcpy_chk'
  469 |         __underlying_##op(p, q, __fortify_size);                        \
      |                           ^
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c:947:5: note: in expansion of macro 'memcpy'
  947 |     memcpy (qmap_net->dev_addr, real_dev->dev_addr, ETH_ALEN);
      |     ^~~~~~
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c:947:21: note: expected 'void *' but argument is of type 'const unsigned char *'
  947 |     memcpy (qmap_net->dev_addr, real_dev->dev_addr, ETH_ALEN);
      |             ~~~~~~~~^~~~~~~~~~
./include/linux/fortify-string.h:469:27: note: in definition of macro '__fortify_memcpy_chk'
  469 |         __underlying_##op(p, q, __fortify_size);                        \
      |                           ^
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c:947:5: note: in expansion of macro 'memcpy'
  947 |     memcpy (qmap_net->dev_addr, real_dev->dev_addr, ETH_ALEN);
      |     ^~~~~~
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c: In function 'GobiNetDriverBind':
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c:1542:32: error: assignment of read-only location '*pDev->net->dev_addr'
 1542 |         pDev->net->dev_addr[0] |= 0x02; /* set local assignment bit */
      |                                ^~
/home/qstel/lede/build_dir/target-aarch64_cortex-a76_musl/linux-bcm27xx_bcm2712/gobinet-1.6.3/GobiUSBNet.c:1543:32: error: assignment of read-only location '*pDev->net->dev_addr'
 1543 |         pDev->net->dev_addr[0] &= 0xbf; /* clear "IP" bit */
      |                                ^~

yanxin152133, 2024-07-24:
https://github.com/coolsnowwolf/lede/issues/12229

qingshutie, 2024-07-25:
> #12229

感谢🙏，我没找到这个lssuse

coolsnowwolf, 2026-09-05:
该编译错误与已关闭的 #12229 完全同类：Quectel GobiNet 1.6.x 直接写入内核新版本中只读的 `net_device->dev_addr`，并调用已移除/未导出的 API。#12229 的后续评论已提供对应内核版本的驱动源码和 backport patch，并确认可以解决；当前 issue 没有新的独立 LEDE 回归或日志。按重复问题处理，关闭为 completed。

closed 2026-09-05 by coolsnowwolf