coolsnowwolf/lede #12352: 请问这个报错怎么处理

free-diy, 2024-07-22:
### 详细叙述

下面的是报错的日志

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

  g++ -o /home/lee/lede/build_dir/hostpkg/node-v18.15.0/out/Release/obj.target/libnode/gen/node_javascript.o /home/lee/lede/build_dir/hostpkg/node-v18.15.0/out/Release/obj/gen/node_javascript.cc '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DNODE_OPENSSL_HAS_QUIC' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DNODE_ARCH="x64"' '-DNODE_PLATFORM="linux"' '-DNODE_WANT_INTERNALS=1' '-DV8_DEPRECATION_WARNINGS=1' '-DNODE_OPENSSL_SYSTEM_CERT_PATH=""' '-DNODE_USE_NODE_CODE_CACHE=1' '-DHAVE_INSPECTOR=1' '-DNODE_ENABLE_LARGE_CODE_PAGES=1' '-D__POSIX__' '-DNODE_USE_V8_PLATFORM=1' '-DNODE_V8_SHARED_RO_HEAP' '-DNODE_HAVE_I18N_SUPPORT=1' '-DNODE_HAVE_SMALL_ICU=1' '-DHAVE_OPENSSL=1' '-DOPENSSL_API_COMPAT=0x10100000L' '-DBASE64_STATIC_DEFINE' '-DUCONFIG_NO_SERVICE=1' '-DU_ENABLE_DYLOAD=0' '-DU_STATIC_IMPLEMENTATION=1' '-DU_HAVE_STD_STRING=1' '-DUCONFIG_NO_BREAK_ITERATION=0' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_POSIX_C_SOURCE=200112' '-DNGHTTP2_STATICLIB' '-DNDEBUG' '-DOPENSSL_USE_NODELETE' '-DL_ENDIAN' '-DOPENSSL_BUILDING_OPENSSL' '-DAES_ASM' '-DBSAES_ASM' '-DCMLL_ASM' '-DECP_NISTZ256_ASM' '-DGHASH_ASM' '-DKECCAK1600_ASM' '-DMD5_ASM' '-DOPENSSL_BN_ASM_GF2m' '-DOPENSSL_BN_ASM_MONT' '-DOPENSSL_BN_ASM_MONT5' '-DOPENSSL_CPUID_OBJ' '-DOPENSSL_IA32_SSE2' '-DPADLOCK_ASM' '-DPOLY1305_ASM' '-DSHA1_ASM' '-DSHA256_ASM' '-DSHA512_ASM' '-DVPAES_ASM' '-DWHIRLPOOL_ASM' '-DX25519_ASM' '-DOPENSSL_PIC' '-DNGTCP2_STATICLIB' '-DNGHTTP3_STATICLIB' -I../src -I/home/lee/lede/build_dir/hostpkg/node-v18.15.0/out/Release/obj/gen -I/home/lee/lede/build_dir/hostpkg/node-v18.15.0/out/Release/obj/gen/include -I/home/lee/lede/build_dir/hostpkg/node-v18.15.0/out/Release/obj/gen/src -I../deps/base64/base64/include -I../deps/googletest/include -I../deps/histogram/src -I../deps/uvwasi/include -I../deps/simdutf -I../deps/v8/include -I../deps/icu-small/source/i18n -I../deps/icu-small/source/common -I../deps/zlib -I../deps/llhttp/include -I../deps/cares/include -I../deps/uv/include -I../deps/nghttp2/lib/includes -I../deps/brotli/c/include -I../deps/openssl/openssl/include -I../deps/openssl/openssl/crypto/include -I../deps/openssl/config/archs/linux-x86_64/asm/include -I../deps/openssl/config/archs/linux-x86_64/asm -I../deps/ngtcp2 -I../deps/ngtcp2/ngtcp2/lib/includes -I../deps/ngtcp2/ngtcp2/crypto/includes -I../deps/ngtcp2/ngtcp2/crypto -I../deps/ngtcp2/nghttp3/lib/includes  -Wall -Wextra -Wno-unused-parameter -pthread -Wall -Wextra -Wno-unused-parameter -m64 -Werror=unused-result -O3 -fno-omit-frame-pointer -fno-rtti -fno-exceptions -std=gnu++17 -MMD -MF /home/lee/lede/build_dir/hostpkg/node-v18.15.0/out/Release/.deps//home/lee/lede/build_dir/hostpkg/node-v18.15.0/out/Release/obj.target/libnode/gen/node_javascript.o.d.raw -I/home/lee/lede/staging_dir/host/include -I/home/lee/lede/staging_dir/hostpkg/include -I/home/lee/lede/staging_dir/target-x86_64_musl/host/include  -c
/home/lee/lede/build_dir/hostpkg/node-v18.15.0/out/Release/obj/gen/node_javascript.cc:137790:1: fatal error: error writing to /home/lee/lede/tmp/ccEtcmYc.s: No space left on device
137790 | }  // namespace node
       | ^
compilation terminated.
make[5]: *** [libnode.target.mk:482: /home/lee/lede/build_dir/hostpkg/node-v18.15.0/out/Release/obj.target/libnode/gen/node_javascript.o] Error 1
rm 1a65a0c8d79e35c0b1366219e669af0717973fec.intermediate
make[4]: *** [Makefile:134: node] Error 2
make[4]: Leaving directory '/home/lee/lede/build_dir/hostpkg/node-v18.15.0'
make[3]: *** [Makefile:233: /home/lee/lede/build_dir/hostpkg/node-v18.15.0/.built] Error 2
make[3]: Leaving directory '/home/lee/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/host-compile#705.74#97.40#798.73
    ERROR: package/feeds/packages/node [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/node/host/compile] Error 1
make[2]: Leaving directory '/home/lee/lede'
make[1]: *** [package/Makefile:110: /home/lee/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lee/lede'
make: *** [/home/lee/lede/include/toplevel.mk:231: world] Error 2

dann2333, 2024-07-22:
你编译的设备没有存储空间了

free-diy, 2024-07-22:
> 你编译的设备没有存储空间了

我留了100G还不够啊

dann2333, 2024-07-22:
考虑df -hT检查一下当前挂载点下还有没有剩余空间

bob-zebedy, 2024-07-23:
> > 你编译的设备没有存储空间了
> 
> 我留了100G还不够啊

带 node 要的空间很大

closed 2024-07-23 by free-diy