coolsnowwolf/lede #12356: 源码编译mediatek失败

audyfan, 2024-07-23:
### 详细叙述

拉取最新源码，5.15内核，一直编译失败。网络问题已排除

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

H3C NX30 Pro

### 详细日志

  CC      lib/rhashtable.o
  CC      drivers/mtd/nand/mtk_bmt.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_inbwctrl.o
  CC      lib/once.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_interrupt.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_led.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_lut.o
  CC      drivers/mtd/spi-nor/sysfs.o
  CC      drivers/mtd/spi-nor/atmel.o
  CC      drivers/mtd/nand/mtk_bmt_v2.o
  CC      drivers/mtd/spi-nor/catalyst.o
  AR      net/ethtool/built-in.a
  CC [M]  net/netfilter/ipset/ip_set_core.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_meter.o
  CC      lib/refcount.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_mib.o
drivers/mtd/nand/mtk_bmt_v2.c: In function 'mtk_bmt_init_v2':
drivers/mtd/nand/mtk_bmt_v2.c:446:18: error: conflicting types for 'block'; have 'u16' {aka 'short unsigned int'}
  446 |         for (u16 block = bmtd.total_blks - 1; !bmtd.bbt && block >= bmtd.pool_lba; block--) {
      |                  ^~~~~
drivers/mtd/nand/mtk_bmt_v2.c:395:20: note: previous declaration of 'block' with type 'u32' {aka 'unsigned int'}
  395 |         u32 bufsz, block;
      |                    ^~~~~
drivers/mtd/nand/mtk_bmt_v2.c:446:9: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
  446 |         for (u16 block = bmtd.total_blks - 1; !bmtd.bbt && block >= bmtd.pool_lba; block--) {
      |         ^~~
drivers/mtd/nand/mtk_bmt_v2.c:446:9: note: use option '-std=c99', '-std=gnu99', '-std=c11' or '-std=gnu11' to compile your code
make[8]: *** [scripts/Makefile.build:289: drivers/mtd/nand/mtk_bmt_v2.o] Error 1
make[7]: *** [scripts/Makefile.build:552: drivers/mtd/nand] Error 2
make[7]: *** Waiting for unfinished jobs....
  CC      lib/usercopy.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_mirror.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_misc.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_oam.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_phy.o
  CC      fs/binfmt_script.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_port.o
  CC      net/netfilter/core.o
  CC      net/netfilter/nf_log.o
  CC      drivers/mtd/spi-nor/eon.o
  AR      net/ipv4/netfilter/built-in.a
  CC [M]  net/ipv4/netfilter/nf_defrag_ipv4.o
  CC      lib/errseq.o
  CC      lib/bucket_locks.o
  CC      fs/binfmt_elf.o
  CC      drivers/mtd/spi-nor/esmt.o
  CC      lib/generic-radix-tree.o
  CC [M]  net/ipv4/netfilter/nf_tproxy_ipv4.o
  CC      lib/string_helpers.o
  CC      drivers/mtd/spi-nor/everspin.o
  CC [M]  net/ipv4/netfilter/nf_reject_ipv4.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_portIsolation.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_qos.o
  CC      drivers/net/phy/phy-c45.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_rldp.o
  CC      drivers/net/phy/rtk/rtl8367c/rtl8367c_asicdrv_rma.o
[... 492 lines cut ...]
  CC [M]  net/netfilter/xt_time.o
  LD [M]  net/netfilter/nf_conntrack.o
  CC      net/bridge/br_vlan_options.o
  LD [M]  net/netfilter/nf_conntrack_h323.o
  CC      net/bridge/br_switchdev.o
  LD [M]  net/netfilter/nf_nat.o
  LD [M]  net/netfilter/nf_flow_table.o
  CC      net/ipv6/mcast_snoop.o
  CC [M]  net/ipv6/sit.o
  AR      net/netfilter/built-in.a
  AR      net/ipv6/built-in.a
  AR      net/bridge/built-in.a
  AR      net/built-in.a
make[5]: Leaving directory '/home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.162'
make[4]: *** [Makefile:21: /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-5.15.162/.modules] Error 2
make[4]: Leaving directory '/home/audy/lede/target/linux/mediatek'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/audy/lede/target/linux'
time: target/linux/compile#1039.81#131.72#172.43
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/audy/lede'
make[1]: *** [target/Makefile:23: /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/audy/lede'
make: *** [/home/audy/lede/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2024-07-23:
5.15 这个patch 将会被 Drop，至少从 6.1 开始吧

audyfan, 2024-07-23:
谢谢大佬回复。5.15，6.1，6.6，这几个内核也都试过了，也都是编译失败报错

cp -fpR /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/libsepol-3.3/ipkg-install/usr/include/* /home/audy/lede/tmp/stage-libsepol/usr/include/
install -d -m0755 /home/audy/lede/tmp/stage-libsepol/usr/lib
cp -fpR /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/libsepol-3.3/ipkg-install/usr/lib/* /home/audy/lede/tmp/stage-libsepol/usr/lib/
install -d -m0755 /home/audy/lede/tmp/stage-libsepol/usr/lib/pkgconfig
install -m0644 /home/audy/lede/build_dir/target-aarch64_cortex-a53_musl/libsepol-3.3/ipkg-install/usr/lib/pkgconfig/libsepol.pc /home/audy/lede/tmp/stage-libsepol/usr/lib/pkgconfig/
/home/audy/lede/staging_dir/host/bin/sed -i -e 's,/usr/include,${prefix}/include,g' /home/audy/lede/tmp/stage-libsepol/usr/lib/pkgconfig/libsepol.pc
/home/audy/lede/staging_dir/host/bin/sed -i -e 's,/usr/lib,${exec_prefix}/lib,g' /home/audy/lede/tmp/stage-libsepol/usr/lib/pkgconfig/libsepol.pc
find /home/audy/lede/tmp/stage-libsepol -name '*.la' | xargs -r rm -f;
if [ -f /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/packages/libsepol.list ]; then /home/audy/lede/scripts/clean-package.sh "/home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/packages/libsepol.list" "/home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl"; fi
if [ -d /home/audy/lede/tmp/stage-libsepol ]; then (cd /home/audy/lede/tmp/stage-libsepol; find ./ > /home/audy/lede/tmp/stage-libsepol.files);         SHELL= flock /home/audy/lede/tmp/.staging-dir.flock -c ' mv /home/audy/lede/tmp/stage-libsepol.files /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/packages/libsepol.list && cp -fpR /home/audy/lede/tmp/stage-libsepol/* /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/; '; fi
rm -rf /home/audy/lede/tmp/stage-libsepol
touch /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.libsepol_installed
make[3]: Leaving directory '/home/audy/lede/package/libs/libsepol'
time: package/libs/libsepol/compile#29.94#2.19#31.88
make[2]: Leaving directory '/home/audy/lede'
make[1]: *** [package/Makefile:110: /home/audy/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/audy/lede'
make: *** [/home/audy/lede/include/toplevel.mk:231: world] Error 2

commit ff2689bd (2024-07-24): mediatek: fixes kernel 5.15 build

Fixes: #12356
closed 2024-07-24 by aiamadeus