coolsnowwolf/lede #12360: 云编译K3源码编译错误: drivers/net/phy/b53/b53_common.c:512:25: error: too many arguments to function 'of_get_phy_mode'

ramblerSY, 2024-07-25:
### 详细叙述

类型的问题如下:https://github.com/coolsnowwolf/lede/issues/12304
r7800出现交换芯片驱动的错误?影响了k3?
错误如下:
drivers/net/phy/b53/b53_common.c: In function 'b53_configure_ports_of':
drivers/net/phy/b53/b53_common.c:512:25: error: too many arguments to function 'of_get_phy_mode'
  512 |                         of_get_phy_mode(pn, &mode);
      |                         ^~~~~~~~~~~~~~~
In file included from drivers/net/phy/b53/b53_common.c:29:
./include/linux/of_net.h:13:12: note: declared here
   13 | extern int of_get_phy_mode(struct device_node *np);
      |            ^~~~~~~~~~~~~~~

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

K3

### 详细日志

Run cd openwrt
4 thread compile
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
 make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/patch compile
 make[3] -C tools/libdeflate compile
 make[3] -C tools/tar compile
 make[3] -C tools/m4 compile
 make[3] -C tools/autoconf-archive compile
 make[3] -C tools/ninja compile
 make[3] -C tools/expat compile
 make[3] -C tools/zlib compile
 make[3] -C tools/cpio compile
 make[3] -C tools/lzma compile
 make[3] -C tools/make-ext4fs compile
 make[3] -C tools/mtools compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/squashfskit4 compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/zip compile
 make[3] -C tools/meson compile
 make[3] -C tools/pkgconf compile
 make[3] -C tools/zstd compile
 make[3] -C tools/libressl compile
 make[3] -C tools/autoconf compile
 make[3] -C tools/elfutils compile
 make[3] -C tools/automake compile
 make[3] -C tools/missing-macros compile
 make[3] -C tools/libtool compile
 make[3] -C tools/dosfstools compile
 make[3] -C tools/cmake compile
 make[3] -C tools/flex compile
 make[3] -C tools/e2fsprogs compile
 make[3] -C tools/fakeroot compile
 make[3] -C tools/gengetopt compile
 make[3] -C tools/mklibs compile
 make[3] -C tools/mtd-utils compile
 make[3] -C tools/patchelf compile
 make[3] -C tools/gmp compile
[... 73 lines cut ...]
drivers/net/phy/b53/b53_common.c: In function 'b53_configure_ports_of':
drivers/net/phy/b53/b53_common.c:512:[25](https://github.com/ramblerSY/P3TERX-Actions-OpenWrt/actions/runs/10089503812/job/27897076657#step:12:26): error: too many arguments to function 'of_get_phy_mode'
  512 |                         of_get_phy_mode(pn, &mode);
      |                         ^~~~~~~~~~~~~~~
In file included from drivers/net/phy/b53/b53_common.c:29:
./include/linux/of_net.h:13:12: note: declared here
   13 | extern int of_get_phy_mode(struct device_node *np);
      |            ^~~~~~~~~~~~~~~
make[9]: *** [scripts/Makefile.build:[26](https://github.com/ramblerSY/P3TERX-Actions-OpenWrt/actions/runs/10089503812/job/27897076657#step:12:27)2: drivers/net/phy/b53/b53_common.o] Error 1
make[8]: *** [scripts/Makefile.build:497: drivers/net/phy/b53] Error 2
make[7]: *** [scripts/Makefile.build:497: drivers/net/phy] Error 2
make[6]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make[5]: *** [Makefile:1750: drivers] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.[27](https://github.com/ramblerSY/P3TERX-Actions-OpenWrt/actions/runs/10089503812/job/27897076657#step:12:28)8'
make[4]: *** [Makefile:32: /workdir/openwrt/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.278/.modules] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/bcm53xx'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#5.15#2.99#7.98
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:[30](https://github.com/ramblerSY/P3TERX-Actions-OpenWrt/actions/runs/10089503812/job/27897076657#step:12:31): target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:2[31](https://github.com/ramblerSY/P3TERX-Actions-OpenWrt/actions/runs/10089503812/job/27897076657#step:12:32): world] Error 2

ramblerSY, 2024-07-25:
看了相关的修改 https://github.com/coolsnowwolf/lede/blob/master/target/linux/generic/files/drivers/net/phy/ar8216.c
#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 5, 0)
	of_get_phy_mode(priv->pdev->of_node, &phy_if_mode);
#else
	phy_if_mode = of_get_phy_mode(priv->pdev->of_node);
#endif
k3好像用到的内核版本是linux-5.4.278， 没找到对应k3在哪里修改

coolsnowwolf, 2024-07-26:
已修复

closed 2024-07-26 by coolsnowwolf