coolsnowwolf/lede #12362: xr_usb_serial_common报错

wanderer-fly, 2024-07-26:
### 详细叙述

Latest version arm64 Raspberry Pi 4B build failed

1. USB设备支持我全选

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Raspberry Pi 4B

### 详细日志

'''
make[4]: Entering directory '/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.1.100'
  CC [M]  /home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.o
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c:2019:33: error: initialization of 'unsigned int (*)(struct tty_struct *)' from incompatible pointer type 'int (*)(struct tty_struct *)' [-Werror=incompatible-pointer-types]
 2019 |         .write_room =           xr_usb_serial_tty_write_room,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c:2019:33: note: (near initialization for 'xr_usb_serial_ops.write_room')
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c:2026:33: error: initialization of 'unsigned int (*)(struct tty_struct *)' from incompatible pointer type 'int (*)(struct tty_struct *)' [-Werror=incompatible-pointer-types]
 2026 |         .chars_in_buffer =      xr_usb_serial_tty_chars_in_buffer,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c:2026:33: note: (near initialization for 'xr_usb_serial_ops.chars_in_buffer')
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c:2028:33: error: initialization of 'void (*)(struct tty_struct *, const struct ktermios *)' from incompatible pointer type 'void (*)(struct tty_struct *, struct ktermios *)' [-Werror=incompatible-pointer-types]
 2028 |         .set_termios =          xr_usb_serial_tty_set_termios,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c:2028:33: note: (near initialization for 'xr_usb_serial_ops.set_termios')
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c: In function 'xr_usb_serial_init':
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c:2040:36: error: implicit declaration of function 'alloc_tty_driver' [-Werror=implicit-function-declaration]
 2040 |         xr_usb_serial_tty_driver = alloc_tty_driver(XR_USB_SERIAL_TTY_MINORS);
      |                                    ^~~~~~~~~~~~~~~~
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c:2040:34: warning: assignment to 'struct tty_driver *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
 2040 |         xr_usb_serial_tty_driver = alloc_tty_driver(XR_USB_SERIAL_TTY_MINORS);
      |                                  ^
/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.c:2057:17: error: implicit declaration of function 'put_tty_driver' [-Werror=implicit-function-declaration]
 2057 |                 put_tty_driver(xr_usb_serial_tty_driver);
      |                 ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/xr_usb_serial_common.o] Error 1
make[4]: *** [Makefile:2009: /home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial] Error 2
make[4]: Leaving directory '/home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.1.100'
make[3]: *** [Makefile:35: /home/wanderer/lede/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xr-usb-serial/.built] Error 2
make[3]: Leaving directory '/home/wanderer/lede/package/kernel/xr-usb-serial'
time: package/kernel/xr-usb-serial/compile#0.81#0.04#0.93
    ERROR: package/kernel/xr-usb-serial failed to build.
make[2]: *** [package/Makefile:116: package/kernel/xr-usb-serial/compile] Error 1
make[2]: Leaving directory '/home/wanderer/lede'
make[1]: *** [package/Makefile:110: /home/wanderer/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wanderer/lede'
make: *** [/home/wanderer/lede/include/toplevel.mk:231: world] Error 2
'''

wanderer-fly, 2024-07-26:
解决了，我把 `xr_usb_serial` 勾掉了

closed 2024-07-28 by aiamadeus

Pigit2, 2024-08-20:
ITs my fault, I think is a dependency bug. Turn off the xr_usb_serial is helpful.

wanderer-fly, 2024-10-28:
> ITs my fault, I think is a dependency bug. Turn off the xr_usb_serial is helpful.

Thank you ❤️😘 Love you