coolsnowwolf/lede #12365: 编译出来的img转esxi磁盘有问题

free-diy, 2024-07-27:
### 详细叙述

不管是云编译还是本地编译，编译出来的文件转esxi磁盘会报错，但不影响使用，能正常使用，在esxi里面看磁盘就不正常，不管是esxi7还是esxi8都有这样的错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

[image]

miaoermua, 2024-07-27:
用 starwind-converter 转

free-diy, 2024-07-28:
> 用 starwind-converter 转

就是用starwind-converter 转的，最新版

miaoermua, 2024-07-28:
> > 用 starwind-converter 转
> 
> 就是用starwind-converter 转的，最新版

你的意思是指大于原始磁盘大小这个错误吗，那我理解错了，这里测试 VMware 的只有 Workstation 大于一点也可以，我这里暂时没有解决方法

yanxin152133, 2024-07-29:
转的文件是两个还是一个？

free-diy, 2024-07-29:
> 转的文件是两个还是一个？

二个

wushenghn, 2024-08-01:
用vmkfstools命令转换一下vmdk文件就可以了 vmdk上传到esxi后进入shell
vmkfstools -i x.vmdk -d thin y.vmdk 创建虚拟机用转换后的vmdk文件就不会有这个错误提示了

free-diy, 2024-08-02:
> 用vmkfstools命令转换一下vmdk文件就可以了 vmdk上传到esxi后进入shell vmkfstools -i x.vmdk -d thin y.vmdk 创建虚拟机用转换后的vmdk文件就不会有这个错误提示了

这不是解决问题的方法，太复杂了。

coolsnowwolf, 2026-09-05:
该 issue 没有给出具体转换命令、输入/输出镜像格式、错误文本或可复现文件，且编译生成的 x86 镜像本身可以正常启动使用。评论已提供 ESXi 侧 `vmkfstools -i x.vmdk -d thin y.vmdk` 的标准转换方式；这是 VMware/StarWind 的磁盘格式兼容性问题，不是 LEDE 镜像生成代码的可定位故障。关闭为 not planned。

closed 2026-09-05 by coolsnowwolf