coolsnowwolf/lede #12367: R4S 切换5.15内核编译错误

liuyanantj, 2024-07-29:
### 详细叙述

lede/target/linux/rockchip/Makefile  把内核修改成5.15  然后编译  最后提示错误   日志如下
请教下，怎么回退5.15的内核啊

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R4S

### 详细日志

Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:181.1-5 Label or path gpu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:186.1-6 Label or path hdmi not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:192.1-9 Label or path hdmi_in not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:198.1-10 Label or path hdmi_out not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:204.1-12 Label or path hdmi_sound not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:450.1-10 Label or path i2s0_8ch not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:645.1-5 Label or path vop not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:651.1-9 Label or path vop_mmu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:655.1-5 Label or path vp0 not found
FATAL ERROR: Syntax error parsing input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-opc-h66k.dtb] Error 1
make[7]: *** Waiting for unfinished jobs....
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts:492.3-13: Warning (reg_format): /mmc@fe2c0000/sdio_wifi@1:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb: Warning (pci_device_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb: Warning (pci_device_bus_num): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts:491.14-496.4: Warning (avoid_default_addr_size): /mmc@fe2c0000/sdio_wifi@1: Relying on default #address-cells value
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts:491.14-496.4: Warning (avoid_default_addr_size): /mmc@fe2c0000/sdio_wifi@1: Relying on default #size-cells value
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:181.1-5 Label or path gpu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:186.1-6 Label or path hdmi not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:192.1-9 Label or path hdmi_in not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:198.1-10 Label or path hdmi_out not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:204.1-12 Label or path hdmi_sound not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:181.1-5 Label or path gpu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:186.1-6 Label or path hdmi not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:192.1-9 Label or path hdmi_in not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:198.1-10 Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:450.1-10 Label or path hdmi_out not foundLabel or path i2s0_8ch not found

Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:204.1-12 Label or path hdmi_sound not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:450.1-10 Label or path i2s0_8ch not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:645.1-5 Label or path vop not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:651.1-9 Label or path vop_mmu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:655.1-5 Label or path vp0 not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:645.1-5 FATAL ERROR: Label or path vop not foundSyntax error parsing input tree

Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:651.1-9 Label or path vop_mmu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi:655.1-5 Label or path vp0 not found
FATAL ERROR: Syntax error parsing input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-opc-h68k.dtb] Error 1
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-opc-h69k.dtb] Error 1
Error: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts:262.1-5 Label or path gpu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts:267.1-6 Label or path hdmi not found
Error: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts:275.1-9 Label or path hdmi_in not found
Error: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts:281.1-10 Label or path hdmi_out not found
Error: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts:287.1-12 Label or path hdmi_sound not found
Error: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts:556.1-10 Label or path i2s0_8ch not found
Error: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts:775.1-5 Label or path vop not found
Error: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts:781.1-9 Label or path vop_mmu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts:785.1-5 Label or path vp0 not found
FATAL ERROR: Syntax error parsing input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dtb] Error 1
Error: arch/arm64/boot/dts/rockchip/rk3568-fastrhino.dtsi:121.1-19 Label or path display_subsystem not found
Error: arch/arm64/boot/dts/rockchip/rk3568-fastrhino.dtsi:125.1-5 Label or path gpu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-fastrhino.dtsi:497.1-5 Label or path vop not found
Error: arch/arm64/boot/dts/rockchip/rk3568-fastrhino.dtsi:503.1-9 Label or path vop_mmu not found
FATAL ERROR: Syntax error parsing input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-r68s.dtb] Error 1
Error: arch/arm64/boot/dts/rockchip/rk3568-fastrhino.dtsi:121.1-19 Label or path display_subsystem not found
Error: arch/arm64/boot/dts/rockchip/rk3568-fastrhino.dtsi:125.1-5 Label or path gpu not found
Error: arch/arm64/boot/dts/rockchip/rk3568-fastrhino.dtsi:497.1-5 Label or path vop not found
Error: arch/arm64/boot/dts/rockchip/rk3568-fastrhino.dtsi:503.1-9 Label or path vop_mmu not found
FATAL ERROR: Syntax error parsing input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-r66s.dtb] Error 1
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dts:517.3-13: Warning (reg_format): /usb@fd000000/device@2:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dtb: Warning (pci_device_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dtb: Warning (pci_device_bus_num): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dts:515.20-519.4: Warning (avoid_default_addr_size): /usb@fd000000/device@2: Relying on default #address-cells value
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dts:515.20-519.4: Warning (avoid_default_addr_size): /usb@fd000000/device@2: Relying on default #size-cells value
make[6]: *** [scripts/Makefile.build:552: arch/arm64/boot/dts/rockchip] Error 2
make[5]: *** [Makefile:1449: dtbs] Error 2
make[5]: *** Waiting for unfinished jobs....
make[5]: Leaving directory '/home/tc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.162'
make[4]: *** [Makefile:26: /home/tc/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.162/.modules] Error 2
make[4]: Leaving directory '/home/tc/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/tc/lede/target/linux'
time: target/linux/compile#9.21#10.45#12.92
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/tc/lede'
make[1]: *** [target/Makefile:23: /home/tc/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/tc/lede'
make: *** [/home/tc/lede/include/toplevel.mk:231：world] 错误 2

yanxin152133, 2024-07-30:
https://github.com/coolsnowwolf/lede/discussions/9993

yanxin152133, 2024-07-30:
https://github.com/coolsnowwolf/lede/issues/1415

dragon888888888888, 2024-07-30:
rk3568-hinlink-opc.dtsi
回退到上一个版本，
它共更新了rk3568-fastrhino.dtsi、rk3568-rock-3a.dts、rk3568-hinlink-opc.dtsi这3个文件更新影响的是5.X内核，使用6.X内核不爱影响
所以有2种办法，更换6.X内核或回退3个文件

liuyanantj, 2024-07-31:
> rk3568-hinlink-opc.dtsi 回退到上一个版本， 它共更新了rk3568-fastrhino.dtsi、rk3568-rock-3a.dts、rk3568-hinlink-opc.dtsi这3个文件更新影响的是5.X内核，使用6.X内核不爱影响 所以有2种办法，更换6.X内核或回退3个文件

个人小白  这个回退要咋弄  我翻看了更改的记录  有好多行  要是一行一行改  怕改错了 不知道有没有更简便一些的方法

xyz201704, 2024-08-22:
建议回退文件， 把之前的下载下来，替换掉（我还没试）
不建议上6.x版本，没啥插件（全要自己编译。麻烦） 
@liuyanantj

xyz201704, 2024-08-22:
> rk3568-hinlink-opc.dtsi 回退到上一个版本， 它共更新了rk3568-fastrhino.dtsi、rk3568-rock-3a.dts、rk3568-hinlink-opc.dtsi这3个文件更新影响的是5.X内核，使用6.X内核不爱影响 所以有2种办法，更换6.X内核或回退3个文件

上一个版本指的那一个？5.14吗？

dragon888888888888, 2024-08-22:
> rk3568-hinlink-opc.dtsi

https://github.com/coolsnowwolf/lede/commit/a87cb8cd9696a683e71112b8231c8d26a5143ad2

5.15也是有版本的

dragon888888888888, 2024-08-22:
不会就下载这个
https://github.com/coolsnowwolf/lede/blob/d3d38a20cba66c2f0de6f6a180245e0f0f36f615/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi
上个版本，替代现在的版本
还不会就换6.1/6.6内核

xyz201704, 2024-08-22:
> 不会就下载这个 https://github.com/coolsnowwolf/lede/blob/d3d38a20cba66c2f0de6f6a180245e0f0f36f615/target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3568-hinlink-opc.dtsi 上个版本，替代现在的版本 还不会就换6.1/6.6内核

谢谢大佬

coolsnowwolf, 2026-09-05:
已修复并推送 commit `f4801ba9e88`。

原因是 5.15 内核缺少新版 RK3568 多媒体 DTS 节点标签，而同步后的 `rk3568-hinlink-opc.dtsi`、`rk3568-fastrhino.dtsi` 和 ROCK 3A DTS 无条件引用这些标签。现在为 5.15 提供兼容 DTS 文件，并在 5.15 的 ROCK 3A 补丁中恢复 `DTS_NO_LEGACY` 条件保护，避免 `gpu/hdmi/vop/i2s0_8ch` 等引用在 5.15 下触发 “Label or path not found”。6.12/6.18 仍使用当前新版 DTS。

已完成补丁应用的 dry-run 校验；当前 checkout 已推送并与 `origin/master` 同步。

closed 2026-09-05 by coolsnowwolf