coolsnowwolf/lede #12371: 编译错误提示 [package/Makefile:70: package/install] Error 255

dd-bpir3, 2024-07-30:
### 详细叙述

每次编译到package/install过不去提示错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

bpi-r3

### 详细日志

* pkg_hash_check_unresolved: cannot find dependency iptables-mod-fullconenat for firewall
 * pkg_hash_fetch_best_installation_candidate: Packages for firewall found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	iptables-mod-fullconenat
 * opkg_install_cmd: Cannot install package default-settings.
 * check_conflicts_for: The following packages conflict with firewall:
 * check_conflicts_for: 	firewall4 * 
 * opkg_install_cmd: Cannot install package firewall.
 * check_data_file_clashes: Package vsftpd wants to install file /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/init.d/vsftpd
	But that file is already provided by package  * vsftpd-alt
 * check_data_file_clashes: Package vsftpd wants to install file /home/yph/lede/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/usr/sbin/vsftpd
	But that file is already provided by package  * vsftpd-alt
 * opkg_install_cmd: Cannot install package vsftpd.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/yph/lede'
make[1]: *** [package/Makefile:111: /home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/yph/lede'
make: *** [/home/yph/lede/include/toplevel.mk:231: world] Error 2

yanxin152133, 2024-07-31:
vsftpd-alt和vsftpd 冲突了

yanxin152133, 2024-07-31:
https://github.com/coolsnowwolf/lede/issues/12339

dd-bpir3, 2024-07-31:
> vsftpd-alt和vsftpd 冲突了

> vsftpd-alt和vsftpd 冲突了

vsftpd-alt和vsftpd是我切换23.05造成的，切换回去还是提示上面错误。

yanxin152133, 2024-07-31:
> > vsftpd-alt和vsftpd 冲突了
> 
> > vsftpd-alt和vsftpd 冲突了
> 
> vsftpd-alt和vsftpd是我切换23.05造成的，切换回去还是提示上面错误。

万能大法试试

dd-bpir3, 2024-07-31:
> > > vsftpd-alt和vsftpd 冲突了
> > 
> > 
> > > vsftpd-alt和vsftpd 冲突了
> > 
> > 
> > vsftpd-alt和vsftpd是我切换23.05造成的，切换回去还是提示上面错误。
> 
> 万能大法试试
没好使

dd-bpir3, 2024-08-17:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency iptables-mod-fullconenat for firewall
 * pkg_hash_fetch_best_installation_candidate: Packages for firewall found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 *      iptables-mod-fullconenat
 * opkg_install_cmd: Cannot install package default-settings.
 * check_conflicts_for: The following packages conflict with firewall:
 * check_conflicts_for:         firewall4 * 
 * opkg_install_cmd: Cannot install package firewall.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/yph/lede'
make[1]: *** [package/Makefile:111: /home/yph/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/yph/lede'
make: *** [/home/yph/lede/include/toplevel.mk:231: world] Error 2

closed 2024-11-24 by aiamadeus