coolsnowwolf/lede #12372: 端口转发后无法在路由器本机访问

Co1lin, 2024-07-31:
### 详细叙述

Router (10.0.0.1) -> Ubuntu (10.0.0.122, 在 54321 监听 HTTP)
Router 上设置端口转发从 wan 到 `10.0.0.122:54321` 之后，
- 外网设备可以通过 `公网IP:54321` 访问到 Ubuntu 上的 HTTP 服务
- Ubuntu 自己可以通过 `公网IP:54321` 访问到自己上的 HTTP 服务
- 外网设备通过 IPSec VPN 加入内网获得内网 IP 后可以通过 `公网IP:54321` 访问到 Ubuntu 上的 HTTP 服务
- **但是** Router 无法通过 `公网IP:54321` 访问到 Ubuntu 上的 HTTP 服务，只能通过内网地址 `10.0.0.122:54321` 访问。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64, 自编译 OpenWrt R22.11.11

### 详细日志

N/A

wushenghn, 2024-08-01:
访问公网地址的时候做一个源nat，随便转换成一个和内外网不一个网段的地址就可以了

Co1lin, 2024-08-01:
@wushenghn 感谢回复！请问可以提供一个具体的命令示例吗？

Co1lin, 2024-08-03:
@wushenghn 尝试做了一个 `-A zone_wan_postrouting -d 公网IP/32 -m comment --comment "!fw3: snat" -j SNAT --to-source 8.8.4.4` ，但还是 connection refused 。

13922740870, 2025-07-06:
我也是这个问题，路由器自身无法访问转发端口的服务地址，外网和内网其他机器都可以访问，内网部署的opkg源路由器访问不了。

13922740870, 2025-07-06:
@Co1lin 你的这个问题解决了吗？

coolsnowwolf, 2026-09-05:
该现象是 Linux/NAT 处理路径的预期行为：路由器自身发往公网地址的连接走本机 `OUTPUT`，不会经过端口转发规则所在的 `PREROUTING` 链，因此不会触发普通的 NAT reflection（hairpin NAT）。iptables-save 中已有 LAN 反射规则，只适用于 LAN 主机发起的流量；给 WAN postrouting 添加 SNAT 也无法在 DNAT 之前改变本机 OUTPUT 路径。

建议在路由器上使用内网地址访问，或通过分 split-DNS/本地 hosts 将域名解析到 `10.0.0.122`；不要用公网 IP 回环测试端口转发。该行为不是 LEDE 包缺陷，关闭 issue。

closed 2026-09-05 by coolsnowwolf