coolsnowwolf/lede #12373: RTL8822CE无线网卡无法发射信号

wzwtt, 2024-07-31:
### 详细叙述

x86中安装了一张RTL8822CE无线网卡，固件里已勾选编译并安装kmod-rtw88和rtl8822ce-firmware，其工作在AP模式，但在无线概况中显示其信号为0dBm，并且手机也接收不到它的信号。详细信息如下图。
[image]
[image]
其可以扫描到其他的WiFi。
[image]
需要怎么做才能使它能够正常工作，发射WiFi信号？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86，RTL8822CE无线网卡

### 详细日志

```
root@OpenWrt:~# dmesg | grep rtw
[   14.742954] rtw_8822ce 0000:00:10.0: Firmware version 9.9.15, H2C version 15
[   14.743942] rtw_8822ce 0000:00:10.0: Firmware version 9.9.4, H2C version 15
```

banana121, 2024-08-12:
接口配置-》无线安全设置是怎么设置的?安装wpad了么

coolsnowwolf, 2026-09-05:
当前 LEDE 已提供 `kmod-rtw88-8822ce` 及 `rtl8822ce-firmware`，原日志中的 `Firmware version 9.9.x` 表明 PCIe 驱动和固件均已加载。仅凭 LuCI 显示 0 dBm 和手机搜不到 SSID，无法判断是 wpad/hostapd 配置、国家码/监管域、信道、天线连接还是驱动问题。

请补充 `iw dev`、`iw phy`、无线配置（隐藏密码）、国家码/信道、hostapd 日志及 `dmesg` 中完整 rtw88 输出；在缺少这些信息前不制作猜测性驱动补丁，issue 保持开放。

closed 2026-09-05 by coolsnowwolf