coolsnowwolf/lede #12374: 编译的x86 OpenWrt系统不停重启的问题

harry10086, 2024-08-01:
### 详细叙述

最近两三周我一直在折腾一个问题：只要在软路由下游插上路由器或交换机，软路由系统就不停的重启。

开始我以为是路由器或交换机问题，换了不同品牌的设备还是不行。
后来我以为是x86的设备问题，我就用树莓派4b编译了一套固件，连上后也是一样的现象。
我想是不是网线问题，换了Wi-Fi连接设备也一样。
如果下游连接电脑选择自动获取IP，OpenWrt显示的就是DHCP 没有分配租约，电脑可以连接软路由，但无法连接互联网，但实际上软路由是可以ping通的。如果我指定了电脑设备的IP，那OpenWrt系统就不停的重启。
如果我软路由的lan只连接单台电脑，那就没有任何问题，所有功能都正常。
因为我五月份编译的固件没有这种现象，然后我就挨个关闭插件，但排除来排除去，也没有找到问题所在。

请大家指教。谢谢！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

无

smallprogram, 2024-08-01:
+1,由来已久，痛苦不已

yanxin152133, 2024-08-01:
https://github.com/coolsnowwolf/lede/issues/12336

coolsnowwolf, 2024-08-01:
明显回环了

smallprogram, 2024-08-01:
@harry10086 试试开启STP协议。

harry10086, 2024-08-01:
> #12336

感觉和这个不太一样，我这里直接就是无法正常工作

harry10086, 2024-08-01:
> 明显回环了

感觉是，但不知道是哪里设置不对，我再试试吧

harry10086, 2024-08-01:
> @harry10086 试试开启STP协议。

路由器没找到这个协议的设置，但交换机我搜了下说都是默认打开stp的，我的交换机都是没有设置的

估计还是软路由里面哪个设置导致了回环

smallprogram, 2024-08-01:
> > @harry10086 试试开启STP协议。
> 
> 
> 
> 路由器没找到这个协议的设置，但交换机我搜了下说都是默认打开stp的，我的交换机都是没有设置的
> 
> 
> 
> 估计还是软路由里面哪个设置导致了回环

在接口 br-lan里。但是我试了试并没有什么卵用

wi11iamzhao, 2024-08-01:
需要日志才能分析具体原因。
目前三台软路由，一台接着10G非网管交换机+AP模式的无线路由器，一台接着2.5G网管交换机+路由模式的无线路由器，一台接着AC+AP都没有遇到相关问题。测试过的系统版本包括R24.4.4、R24.7.7、R24.8.1

NeedforGit, 2024-08-02:
大雕最近换6.6内核了，再用6.6内核重新编译试试。

harry10086, 2024-08-05:
感谢各位的解答，我再试试

smallprogram, 2024-08-05:
稳定两天了
[image]

harry10086, 2024-08-08:
虽然现在编译的内核和之前不一样了，但我觉得还是 Turbo ACC 插件的问题，我改成了 Shortcut-FE 后正常了，无论软路由下游加路由器还是交换机系统都不会重启了。
感谢各位。

closed 2024-08-08 by harry10086