coolsnowwolf/lede #12379: RX中的dropped值一直增加

lovexiaofan123, 2024-08-03:
### 详细叙述

ifconfig看了网卡信息，无意发现RX中的dropped值很大，而且一直在增加，不清楚是什么情况造成的，我刷了lede的最新的内核6.6也是一样，防火墙默认配置，插件编译也是lede默认选项，没动过。附接口图。我之前刷过爱快，爱快这个问题不存在，所以可以排除硬件问题。
ipv6我是关闭的，eth3没插网线，用不上 所以没信息。
br-lan    Link encap:Ethernet  HWaddr 60:BE:B4:08:F2:AF
          inet addr:192.168.168.1  Bcast:192.168.168.255  Mask:255.255.255.0
          inet6 addr: fe80::62be:b4ff:fe08:f2af/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3085217 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7228570 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1353469635 (1.2 GiB)  TX bytes:13941824899 (12.9 GiB)

br-lan-ifb Link encap:Ethernet  HWaddr 22:81:C1:A6:83:C3
          inet6 addr: fe80::2081:c1ff:fea6:83c3/64 Scope:Link
          UP BROADCAST RUNNING NOARP  MTU:1500  Metric:1
          RX packets:2880765 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2880765 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:32
          RX bytes:1056736620 (1007.7 MiB)  TX bytes:1056736620 (1007.7 MiB)

eth0      Link encap:Ethernet  HWaddr 60:BE:B4:08:F2:AE
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:11263074 errors:0 dropped:19996 overruns:0 frame:0
          TX packets:3319741 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:14536907116 (13.5 GiB)  TX bytes:1510516867 (1.4 GiB)
          Memory:80200000-802fffff

eth1      Link encap:Ethernet  HWaddr 60:BE:B4:08:F2:AF
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2681802 errors:0 dropped:3943 overruns:0 frame:0
          TX packets:9136335 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:943732415 (900.0 MiB)  TX bytes:12177582563 (11.3 GiB)
          Memory:80000000-800fffff

eth2      Link encap:Ethernet  HWaddr 60:BE:B4:08:F2:B0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:569285 errors:0 dropped:3953 overruns:0 frame:0
          TX packets:1337915 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:459801358 (438.5 MiB)  TX bytes:1612433264 (1.5 GiB)
          Memory:7fe00000-7fefffff

eth3      Link encap:Ethernet  HWaddr 60:BE:B4:08:F2:B1
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Memory:7fc00000-7fcfffff

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:708680 errors:0 dropped:0 overruns:0 frame:0
          TX packets:708680 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1679490786 (1.5 GiB)  TX bytes:1679490786 (1.5 GiB)

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

N5105

### 详细日志

编译没任何问题

aiamadeus, 2024-08-03:
你没说是啥网卡啊。。大概率驱动问题

lovexiaofan123, 2024-08-03:
> 你没说是啥网卡啊。。大概率驱动问题

i226

lovexiaofan123, 2024-08-03:
内核日志

[    8.532991] xt_time: kernel timezone is -0000
[    8.537939] usbcore: registered new interface driver aqc111
[    8.544257] usbcore: registered new interface driver ax88179_178a
[    8.558472] bnx2x: QLogic 5771x/578xx 10/20-Gigabit Ethernet Driver bnx2x 1.713.36-0 (2014/02/10)
[    8.574589] PPP generic driver version 2.4.2
[    8.579441] PPP MPPE Compression module registered
[    8.584842] NET: Registered protocol family 24
[    8.591783] kmodloader: done loading kernel modules from /etc/modules.d/*
[   10.102915] br-lan: port 1(eth1) entered blocking state
[   10.108680] br-lan: port 1(eth1) entered disabled state
[   10.114489] device eth1 entered promiscuous mode
[   10.190059] br-lan: port 2(eth2) entered blocking state
[   10.195847] br-lan: port 2(eth2) entered disabled state
[   10.201659] device eth2 entered promiscuous mode
[   10.276662] br-lan: port 3(eth3) entered blocking state
[   10.282417] br-lan: port 3(eth3) entered disabled state
[   10.288217] device eth3 entered promiscuous mode
[   10.375897] HTB: quantum of class 10001 is big. Consider r2q change.
[   10.418258] HTB: quantum of class 10001 is big. Consider r2q change.
[   10.426226] HTB: quantum of class 10011 is big. Consider r2q change.
[   10.434797] HTB: quantum of class 10001 is big. Consider r2q change.
[   10.467140] HTB: quantum of class 10013 is big. Consider r2q change.
[   10.476666] HTB: quantum of class 10014 is big. Consider r2q change.
[   10.484382] HTB: quantum of class 10015 is big. Consider r2q change.
[   10.496671] HTB: quantum of class 10018 is big. Consider r2q change.
[   10.509622] HTB: quantum of class 10110 is big. Consider r2q change.

lovexiaofan123, 2024-08-03:
又仔细的翻了翻日志，似乎有东西没加载
[    1.509585] ACPI: Enabled 6 GPEs in block 00 to 7F
[    6.934310] block: attempting to load /tmp/overlay/upper/etc/config/fstab
[    6.941802] block: unable to load configuration (fstab: Entry not found)
[    6.949183] block: attempting to load /tmp/overlay/etc/config/fstab
[    6.956095] block: unable to load configuration (fstab: Entry not found)
[    6.963474] block: attempting to load /etc/config/fstab
[    6.969247] block: unable to load configuration (fstab: Entry not found)
[    6.976634] block: no usable configuration
[    6.981156] block: attempting to load /etc/config/fstab
[    6.986918] block: unable to load configuration (fstab: Entry not found)
[    6.994296] block: no usable configuration
[    7.115142] block: attempting to load /tmp/overlay/upper/etc/config/fstab
[   10.102915] br-lan: port 1(eth1) entered blocking state
[   10.190059] br-lan: port 2(eth2) entered blocking state
[   10.276662] br-lan: port 3(eth3) entered blocking state
[   12.858839] br-lan: port 2(eth2) entered blocking state
[   13.206759] br-lan: port 1(eth1) entered blocking state
[   21.022851] br-lan: port 2(eth2) entered blocking state
[   21.701259] br-lan: port 1(eth1) entered blocking state

HOT-POT1, 2024-08-06:
有没有单口复用？比如单口做lan和wan，就会一直有dropped增加

lovexiaofan123, 2024-08-08:
> 有没有单口复用？比如单口做lan和wan，就会一直有dropped增加
没有，所有接口网线直连设备，单路复用从来没用过，也用不上

coolsnowwolf, 2026-09-05:
该 issue 与 I226-V 网卡问题重复。当前 LEDE 的 x86 5.10/5.15/6.1/6.6/6.12 分支均包含 `996-intel-igc-i225-i226-disable-eee.patch`，用于规避 I225/I226 链路掉包问题。

原日志只有 `ifconfig` 的 RX dropped 累计值，没有 `ethtool -S`、`dmesg` 错误或链路协商信息；`ifconfig` 的 dropped 也可能来自接收队列/桥接层丢弃，不能单独归因于驱动。当前已有 workaround，无需新增补丁，关闭 issue。若仍异常，请提供 `ethtool -S ethX` 和完整内核日志。

closed 2026-09-05 by coolsnowwolf