coolsnowwolf/lede #12382: 6.6核心x86系统固件在J4205系统无法启动，同样的固件用在虚拟机上则正常

yujieli001, 2024-08-04:
### 详细叙述

用最新的6.6核心自己编译x86固件，通过linux的dd命令写入ssd硬盘或用web界面从6.1核心升级到6.6，启动后停留在最开始“OPENWRT”字样就死机，无法启动。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Intel(R) Pentium(R) CPU J4205 @ 1.50GHz : 4C4T (CpuMark : 45125.240169 Scores)

### 详细日志

无法采集日志

bi4tms, 2024-08-04:
看看你的启动方式

yujieli001, 2024-08-05:
@bi4tms 6.6的启动方式有改变？

yujieli001, 2024-08-05:
经过仔细测试，UEFI和普通启动都不行。况且没有去改变原来可以启动的6.1的启动方式。

yujieli001, 2024-08-19:
昨天用最新的源码编译，还是一样的情况，大家有没有这种情况？

spart9527, 2024-08-25:
> 昨天用最新的源码编译，还是一样的情况，大家有没有这种情况？

24/8/25:j4205碰到了跟你完全一样的情况，直接无法启动，折腾好久最后换回6.1的老版本了

yujieli001, 2024-08-30:
看来固件缺省改变了一些东西，造成无法兼容原来的硬件的问题。

zorrose82, 2024-09-08:
一样碰到了，正在头疼……

zorrose82, 2024-09-18:
不死心用最新代码编译，再次无法开机，老老实实刷6.1的了……

zorrose82, 2024-09-20:
[image]

启动卡在这里
6.1 则很顺利启动
下一段是  pci cls 0 byte，default 64

yujieli001, 2024-10-21:
就不管老的硬件了吗？

zorrose82, 2024-10-22:
> 就不管老的硬件了吗？

不是cpu的锅，感觉是板子的pci有问题，相同设备我装了exsi，再装新固件就能用了……
你也照着折腾一下把

zzy19868, 2024-10-29:
一样的硬件一样的问题，我还以为是编译哪个地方有问题，看来不用折腾了。

zorrose82, 2024-11-04:
> 一样的硬件一样的问题，我还以为是编译哪个地方有问题，看来不用折腾了。 J4205相当好用，夏天室外温度38度，放在弱电箱被动散热不到50度。稳的不像话。

折腾啊，装exsi继续用

openwrtledediy, 2025-01-27:
是的，两台不同品牌的N4200，都无法启动，换成最新的istoreos可以启动了

coolsnowwolf, 2026-09-05:
该问题在多个 J4205/N4200 设备上被报告，但现有内容只有“Booting 'openwrt'”或截图，没有串口全量日志。日志中常见的 `PCI: CLS 0 bytes, default 64` 本身是 PCI 通用提示，不能证明是挂起根因；当前也没有足够证据判断是 PCI、显卡、ACPI、UEFI 还是启动介质。

请补充可复现构建的 commit、6.1/6.6 `.config` 差异、GRUB/UEFI 启动参数及串口/早期 printk 日志（建议加入 `earlyprintk=efi,keep loglevel=8`）。在缺乏这些证据前不制作针对 J4205 的猜测性内核补丁，issue 保持开放。

closed 2026-09-05 by coolsnowwolf