coolsnowwolf/lede #12383: 使用Ubuntu 24.04 LTS 编译时出错 求大神解答

Chaychen-TM, 2024-08-05:
### 详细叙述

使用Ubuntu 24.04 LTS 编译时出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

vmware虚拟机

### 详细日志

ependency tinfo skipped: feature tinfo disabled
Found pkg-config: /home/chay/桌面/lede/staging_dir/host/bin/pkg-config (1.9.3)
Traceback (most recent call last):
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/mesonmain.py", line 148, in run
    return options.run_func(options)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/msetup.py", line 294, in run
    app.generate()
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/msetup.py", line 185, in generate
    self._generate(env)
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/msetup.py", line 229, in _generate
    intr.run()
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/interpreter.py", line 2554, in run
    super().run()
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 148, in run
    self.evaluate_codeblock(self.ast, start=1)
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 173, in evaluate_codeblock
    raise e
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 166, in evaluate_codeblock
    self.evaluate_statement(cur)
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 181, in evaluate_statement
    self.assignment(cur)
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 566, in assignment
    value = self.evaluate_statement(node.value)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 179, in evaluate_statement
    return self.function_call(cur)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/interpreterbase.py", line 454, in function_call
    res = func(node, func_args, kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 768, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 768, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 768, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  [Previous line repeated 5 more times]
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 96, in wrapped
    ret = f(*wrapped_args, **wrapped_kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 115, in wrapped
    return f(*wrapped_args, **wrapped_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreterbase/decorators.py", line 276, in wrapper
    return f(*nargs, **wrapped_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/interpreter.py", line 1577, in func_dependency
    d = df.lookup(kwargs)
        ^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/dependencyfallbacks.py", line 347, in lookup
    dep = func(kwargs, func_args, func_kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/interpreter/dependencyfallbacks.py", line 90, in _do_dependency
    dep = dependencies.find_external_dependency(name, self.environment, kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/detect.py", line 114, in find_external_dependency
    d = c()
        ^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 96, in __init__
    self._set_libs()
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 372, in _set_libs
    ret, out, err = self._call_pkgbin(libcmd, env=env)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 157, in _call_pkgbin
    cache[(self.pkgbin, targs, fenv)] = self._call_pkgbin_real(args, env)
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/dependencies/pkgconfig.py", line 115, in _call_pkgbin_real
    p, out, err = Popen_safe(cmd, env=env)
                  ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/chay/桌面/lede/staging_dir/host/bin/meson.py/mesonbuild/mesonlib/universal.py", line 1397, in Popen_safe
    o, e = p.communicate(write)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 1209, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 2153, in _communicate
    stdout = self._translate_newlines(stdout,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/subprocess.py", line 1086, in _translate_newlines
    data = data.decode(encoding, errors)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe6 in position 14: invalid continuation byte

../../../build_dir/target-x86_64_musl/util-linux-2.38.1/meson.build:214:0: ERROR: Unhandled python exception

    This is a Meson bug and should be reported!
make[3]: *** [Makefile:878: /home/chay/桌面/lede/build_dir/target-x86_64_musl/util-linux-2.38.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[3]: Leaving directory '/home/chay/桌面/lede/package/utils/util-linux'
time: package/utils/util-linux/compile#12.44#3.41#15.77
    ERROR: package/utils/util-linux failed to build.
make[2]: *** [package/Makefile:116: package/utils/util-linux/compile] Error 1
make[2]: Leaving directory '/home/chay/桌面/lede'
make[1]: *** [package/Makefile:110: /home/chay/桌面/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/chay/桌面/lede'
make: *** [/home/chay/桌面/lede/include/toplevel.mk:231：world] 错误 2

jeasonliang, 2024-08-05:
我也是 /stamp/.package_compile] Error 2
toplevel.mk:231：world] 错误 2

lanruihua026, 2024-08-05:
建议使用Ubuntu 22.04lts编译，新版本有些插件不支持，云编译也是用的Ubuntu 22.04

seasun2k, 2024-08-06:
架梯子，重新menuconfig选依赖，我就是24，正常编译

AlittleJerry-maker, 2024-08-26:
目录尽量别有中文 ，OpenWRT官方有关于Ubuntu 24.04 版本所需的依赖库的说明：
https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem
Modernized set for Ubuntu 24.04 that has Python 3.12 without python3-distutils: (OpenWrt main/master in Apr 2024)

```
sudo apt update
sudo apt install build-essential clang flex bison g++ gawk \
gcc-multilib g++-multilib gettext git libncurses5-dev libssl-dev \
python3-setuptools rsync swig unzip zlib1g-dev file wget
```

coolsnowwolf, 2026-09-05:
日志的首个实际失败是 Meson/Python 3.12 的 `UnicodeDecodeError`，发生在读取 `pkg-config` 输出时；构建路径 `/home/chay/桌面/lede` 含中文目录名，导致本地化输出无法按 UTF-8 解码。当前仓库的 util-linux 已升级到 2.42.2，Ubuntu 24.04 也在支持范围内。

请将源码放在纯 ASCII 路径，并按 README 安装完整依赖后重新构建。该问题不是目标包源码错误，当前无需修改，关闭 issue。

closed 2026-09-05 by coolsnowwolf