coolsnowwolf/lede #12388: 京东云亚瑟  JDCloud AX1800pro 编译成功了，但是USB3.0无法使用。

yui1987, 2024-08-06:
### 详细叙述

按照教程，云编译了京东云亚瑟  JDCloud AX1800pro 的lede固件，成功启动了，其他功能都正常，插U盘无反应。有大佬知道原因吗？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

 JDCloud AX1800pro

### 详细日志

无

aiamadeus, 2024-08-06:
内核日志都不给？lsusb 的输出呢

d975xbx, 2024-08-07:
挂载的问题，你要勾选自动挂载的

yui1987, 2024-08-07:
开启自动挂载了，保存应用没有效果。

yui1987, 2024-08-07:
内核日志不太懂。lsusb   结果not found。  
root@OpenWrt:~# lsusb
-ash: lsusb: not found

seed.config就是默认设置。
CONFIG_TARGET_qualcommax=y
CONFIG_TARGET_qualcommax_ipq60xx=y
CONFIG_TARGET_qualcommax_ipq60xx_DEVICE_jdcloud_re-ss-01=y

yui1987, 2024-08-07:
固件里包含的usb软件包

 kmod-usb-core | 6.1.100-1
 kmod-usb-dwc3 | 6.1.100-1
 kmod-usb-dwc3-qcom | 6.1.100-1
 kmod-usb-xhci-hcd | 6.1.100-1
 kmod-usb3 | 6.1.100-1

yui1987, 2024-08-08:
感谢楼上大佬，编译了一个简单的固件验证过了，确实是这个问题。非常感谢！！！开源社区因为有你更精彩。

commit 3341f523 (2024-08-08): update default packages for filogic and ipq60xx, x86

Fixes: #12388
closed 2024-08-08 by aiamadeus